MiczFlor/RPi-Jukebox-RFID #813: no more spotify working on both of my boxes with 1.1.8-rc2 & stretch

sonicdee, 2020-02-26:
## Bug
I can't play Spotify stuff at all, since a couple of days,
but local media mp3 is working still

### What I did
tried to use new keys for mopidy,

### What happened
nothing, 
webinterface and everything works fine, but no audio from Sportify...
I think there is any kind of library or connection to spotify not supported anymore!?

Is there any chance to not set up a new system !?

Raspbian GNU/Linux 9 (stretch)
Phomiebox 1.1.8-rc2

sonicdee, 2020-02-26:
sudo nano /var/log/mopidy/mopidy.log
there are lines with:
ERROR [520:SpotifyEventLoop] spotify.session: Spotify connection error: <ErrorType.UNABLE_TO_CONTACT_SERVER: 8>

s-martin, 2020-02-26:
Could you try to play something from Spotify and afterwards post the results of `tail -n 500 /var/log/syslog` and `tail -n 500 /var/log/mopidy/mopidy.log`?

sonicdee, 2020-02-26:
2020-02-26 10:19:58,030 INFO [530:MainThread] mopidy.config: Loading config from builtin defaults
2020-02-26 10:19:58,045 INFO [530:MainThread] mopidy.config: Loading config from /etc/mopidy/mopidy.conf
2020-02-26 10:19:58,052 INFO [530:MainThread] mopidy.config: Loading config from command line options
2020-02-26 10:19:58,218 INFO [530:MainThread] mopidy.__main__: Enabled extensions: iris, mpd, http, stream, spotify, m3u, softwaremixer, file, local-images, local
2020-02-26 10:19:58,219 INFO [530:MainThread] mopidy.__main__: Disabled extensions: none
2020-02-26 10:19:58,761 INFO [530:MainThread] mopidy.commands: Starting Mopidy mixer: SoftwareMixer
2020-02-26 10:19:58,767 INFO [530:MainThread] mopidy.commands: Mixer volume set to 30
2020-02-26 10:19:58,769 INFO [530:MainThread] mopidy.commands: Starting Mopidy audio
2020-02-26 10:19:58,775 INFO [530:MainThread] mopidy.commands: Starting Mopidy backends: SpotifyBackend, StreamBackend, M3UBackend, FileBackend, LocalBackend
2020-02-26 10:19:58,991 INFO [530:MainThread] mopidy.local.library: Loaded 79 local tracks using json
2020-02-26 10:19:59,023 INFO [530:Audio-2] mopidy.audio.actor: Audio output set to "alsasink"
2020-02-26 10:20:00,063 INFO [530:SpotifyBackend-3] mopidy_spotify.web: Logged into Spotify Web API as s2x3dnn2eedc78ckw8bar3sst
2020-02-26 10:20:00,065 INFO [530:SpotifyBackend-3] mopidy_spotify.backend: Started spotify backend
2020-02-26 10:20:00,074 INFO [530:MainThread] mopidy.commands: Starting Mopidy core
2020-02-26 10:20:00,203 INFO [530:Thread-10] mopidy_spotify.web: Loaded 1 playlists
2020-02-26 10:20:00,209 INFO [530:Thread-10] mopidy_spotify.utils: Refresh Playlists took 143ms
2020-02-26 10:20:00,562 INFO [530:MainThread] mopidy.commands: Starting Mopidy frontends: IrisFrontend, MpdFrontend, HttpFrontend
2020-02-26 10:20:00,566 INFO [530:IrisFrontend-12] mopidy_iris.core: Starting Iris 3.30.0
2020-02-26 10:20:00,571 INFO [530:MainThread] mopidy.mpd.actor: MPD server running at [::ffff:0.0.0.0]:6600
2020-02-26 10:20:00,581 INFO [530:HttpFrontend-15] mopidy.http.actor: HTTP server running at [::ffff:0.0.0.0]:6680
2020-02-26 10:20:00,582 INFO [530:MainThread] mopidy.commands: Starting GLib mainloop
2020-02-26 10:39:22,389 ERROR [530:SpotifyEventLoop] spotify.session: Spotify login error: <ErrorType.UNABLE_TO_CONTACT_SERVER: 8>
2020-02-26 10:39:22,395 INFO [530:SpotifyEventLoop] spotify.session: Spotify logged out
2020-02-26 12:39:05,507 INFO [530:MpdSession-16] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2020-02-26 12:39:06,538 INFO [530:MpdSession-17] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2020-02-26 12:39:07,348 INFO [530:MpdSession-18] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2020-02-26 12:39:26,922 INFO [530:MpdSession-19] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2020-02-26 12:39:27,014 INFO [530:MpdSession-20] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2020-02-26 12:39:27,503 INFO [530:SpotifyBackend-3] mopidy_spotify.lookup: Failed to lookup "spotify:album:6jbxhrzeRQ87glDoAmw42C": Session must be logged in and online to load objects: <ConnectionState.LOGGED_OUT: 0>
2020-02-26 12:39:27,534 INFO [530:MpdSession-21] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2020-02-26 12:39:28,048 INFO [530:MpdSession-22] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2020-02-26 12:39:28,105 INFO [530:MpdSession-23] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2020-02-26 12:39:29,380 INFO [530:MpdSession-24] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2020-02-26 12:39:29,432 INFO [530:MpdSession-25] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2020-02-26 12:39:29,583 INFO [530:MpdSession-26] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2020-02-26 12:39:46,326 INFO [530:MpdSession-27] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2020-02-26 12:39:46,366 INFO [530:MpdSession-28] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2020-02-26 12:39:46,391 INFO [530:MpdSession-29] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2020-02-26 12:39:46,444 INFO [530:MpdSession-30] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2020-02-26 12:39:46,525 INFO [530:MpdSession-31] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2020-02-26 12:39:51,326 INFO [530:MpdSession-32] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2020-02-26 12:39:51,332 INFO [530:MpdSession-33] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2020-02-26 12:39:51,406 INFO [530:MpdSession-34] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2020-02-26 12:39:51,423 INFO [530:MpdSession-35] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2020-02-26 12:39:51,520 INFO [530:MpdSession-36] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2020-02-26 12:39:56,875 INFO [530:MpdSession-37] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2020-02-26 12:39:56,884 INFO [530:MpdSession-38] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2020-02-26 12:39:56,963 INFO [530:MpdSession-39] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
[... cut ...]
Feb 26 12:39:26 phonieboxisa mopidy[530]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Feb 26 12:39:27 phonieboxisa mopidy[530]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Feb 26 12:39:27 phonieboxisa mopidy[530]: INFO     Failed to lookup "spotify:album:6jbxhrzeRQ87glDoAmw42C": Session must be logged in and online to load objects: <ConnectionState.LOGGED_OUT: 0>
Feb 26 12:39:27 phonieboxisa mopidy[530]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Feb 26 12:39:28 phonieboxisa mopidy[530]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600

s-martin, 2020-02-26:
just to make sure: Can you open Spotify in your browser from a device in the same WiFi network?

sonicdee, 2020-02-26:
yes I tried it on the a webplayer in chrome with this users/pwd - works

sonicdee, 2020-02-26:
I can use the spotify interface on the box like to see genre / moods / but does not play
saying "Mopidy failed to add some tracks"

sonicdee, 2020-02-26:
strange thing
I clicked "restart server" inside /iris/#/settings
which worked, but rebooting the system not !!

closed 2020-02-26 by sonicdee