MiczFlor/RPi-Jukebox-RFID #815: 1st Phoniebox: Web-App does not start MP3-File

Discoverer81, 2020-02-28:
## Web-App does not start MP3-File

### Installation of Stretch
        Installation of Phoniebox with Spotify
        Adding USB-Soundcard as "Speaker"

After uploading MP3-File (works with VLC on PC) and scanning Music-Libary, the file doen´t start when pressing play. I would expect a running time when playing.

It´s my first time working with Raspberry.
I tried different solutions from the web, in the end the system crushed totally.
So, now it´s a new installation and i hope you can help me from start.

Thanks in advance.

s-martin, 2020-02-28:
Currently there are issues with the Spotify edition, see other tickets here. 
Maybe Classic edition is an option.

If possible I suggest to use Buster and not Stretch for new installations. 

Stretch has possibly even more issues (which probably won't/can't get fixed anymore), because some third party libraries don't support it anymore, especially using Spotify edition.

Discoverer81, 2020-02-28:
Hello s-martin,

i followed your advise and installes buster.
That solved the problem with not playing within the web application.
Thanks for that.

closed 2020-02-28 by Discoverer81