MiczFlor/RPi-Jukebox-RFID #818: MQTT startet nicht

Ghostsl84, 2020-02-29:
Guten Abend,

ich habe jetzt wirklich alle Anleitungen von hier durch aber leider ohne erfolg.
Mqtt Startet nicht und ich habe die daemon_mqtt_client.py ebenfalls angepasst zu

daemon-mqtt-client.py

und die Daten darin ebenfalls:

#  SETTINGS
# ----------------------------------------------------------
DEBUG=False
mqttBaseTopic="meinephoniebox"				# MQTT base topic
mqttClientId="meinephoniebox"				# MQTT client ID
mqttHostname="netzwerkname oder muss hier die IP eingegeben werden?"		# MQTT server hostname
mqttPort=8883						# MQTT server port (typically 1883 for unencrypted, 8883 for encrypted)
mqttUsername="NAME"						# username for user/pass based authentication
mqttPassword="meingeheimesPW"						# password for user/pass based authentication
mqttCA="/home/pi/MQTT/mqtt-ca.crt"			# path to server certificate for certificate-based authentication
mqttCert="/home/pi/MQTT/mqtt-client-phoniebox.crt"	# path to client certificate for certificate-based authentication
mqttKey="/home/pi/MQTT/mqtt-client-phoniebox.key"	# path to client keyfile for certificate-based authentication
mqttConnectionTimeout=60				# in seconds; timeout for MQTT connection
refreshIntervalPlaying=5				# in seconds; how often should the status be sent to MQTT (while playing)
refreshIntervalIdle=30					# in seconds; how often should the status be sent to MQTT (when NOT playing)

vielen Dank für die Hilfe

s-martin, 2020-03-01:
Please show the output of  `tail -n 500 /var/log/syslog`

erhapp, 2020-03-22:
I can confirm the problem with MQTT.
I installed a fresh system over the weekend and MQTT does not start on boot.
If I start the script manually everything goes fine.
I did however get an error message the first time I did a manual start. Apparently "paho-mqtt" was missing on my system. This was resolved by 'pip install paho-mqtt'. Is this something that can be included in the install script?

Is there any information I can provide regarding the problem with the script not starting on startup?

erhapp, 2020-03-22:
I just noticed that the MQTT service is not loaded by default.

Solution:
1. 'pip install paho-mqtt' to install the missing MQTT Python package
mqtt

2. Enable MQTT service:

sudo cp /home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/phoniebox-mqtt-client.service.stretch-default.sample /etc/systemd/system/phoniebox-mqtt-client.service
sudo chown root:root /etc/systemd/system/phoniebox-mqtt-client.service
sudo chmod 644 /etc/systemd/system/phoniebox-mqtt-client.service
sudo systemctl enable phoniebox-mqtt-client

s-martin, 2020-03-22:
From my point of view the explanation should be added to the README.md in the components mqtt dir and a requirements.txt added.
What do you think @MiczFlor, @ZyanKLee?

closed 2020-03-30 by s-martin