MiczFlor/RPi-Jukebox-RFID #820: Phoniebox does not react on mqtt cmd

dannyv1101, 2020-02-29:
## Bug

### What I did

I set up the mqtt script like in the description. After some faults I get it running an everything showed up correctly in iobroker.

### What happened

But when I send a command from the list (for example playerpause when the player is running) to phoniebox/cmd the box will do nothing. 
(phoniebox/get/help is working also).

I am running raspbin stretch with phoniebox 2.0 rc4. 

I hope someone can help me

s-martin, 2020-03-21:
Could you post the output of `tail -n 500 /var/log/syslog`?

andreasbrett, 2020-06-27:
> But when I send a command from the list (for example playerpause when the player is running) to phoniebox/cmd the box will do nothing.

Sounds like you're sending the command incorrectly. You will have to send an **empty payload** to **phoniebox/cmd/playerpause** instead of sending **playerpause** to **phoniebox/cmd**.

Late answer but I hope it's still ongoing and helps. Let us know if it worked so we can close the issue.

closed 2020-12-18 by s-martin