MiczFlor/RPi-Jukebox-RFID #823: 🐛 | BUG SUMMARY

kuesimon, 2020-03-01:
## Bug

### What I did

I new installed the phoniebox master branch and tried to scan rfid tags and link them to spotify streams.

### What happened

There's no reaction in the phoniebox - but if look on the card id settings it is displaying the rfid number.

BUT: The rfid reader tries to login as raspberry user on the console. I observed this as i had my phoniebox connected to a screen and wondered why there where that many invalid logins.

### Software Version raspi 

2019-09-26-raspbian-buster.img

### Installscript

one line install script master branch

### RFID Reader

HXGCoLtd Keyboard

Is this a known issue? Does anybody have a hint how to fix this?
Thx

Regards,
kuesimon

s-martin, 2020-03-01:
Please post the result of `tail -n 500 /var/log/syslog`.

kuesimon, 2020-03-01:
`Mar  1 16:47:03 raspberrypi dbus-daemon[333]: [system] Successfully activated service 'org.freedesktop.systemd1'
Mar  1 16:47:03 raspberrypi systemd[1]: Started WPA supplicant.
Mar  1 16:47:04 raspberrypi cleanshutd[344]: Starting...
Mar  1 16:47:04 raspberrypi systemd[1]: Started Login Service.
Mar  1 16:47:04 raspberrypi systemd[1]: Started Avahi mDNS/DNS-SD Stack.
Mar  1 16:47:04 raspberrypi systemd[1]: Started LSB: Monitoring GPIO shutdown trigger.
Mar  1 16:47:04 raspberrypi dphys-swapfile[380]: want /var/swap=100MByte, checking existing: keeping it
Mar  1 16:47:04 raspberrypi systemd[1]: Reached target Network.
Mar  1 16:47:04 raspberrypi systemd[1]: Reached target Network is Online.
Mar  1 16:47:04 raspberrypi systemd[1]: Starting Lighttpd Daemon...
Mar  1 16:47:04 raspberrypi systemd[1]: Starting LSB: exim Mail Transport Agent...
Mar  1 16:47:04 raspberrypi systemd[1]: Condition check resulted in fast remote file copy program daemon being skipped.
Mar  1 16:47:04 raspberrypi systemd[1]: Started Phoniebox Idle Watchdog Service.
Mar  1 16:47:04 raspberrypi systemd[1]: Starting Permit User Sessions...
Mar  1 16:47:04 raspberrypi systemd[1]: Starting Mopidy music server...
Mar  1 16:47:04 raspberrypi systemd[1]: Starting Samba NMB Daemon...
Mar  1 16:47:04 raspberrypi systemd[1]: Starting /etc/rc.local Compatibility...
Mar  1 16:47:04 raspberrypi dhcpcd-run-hooks[451]: wlan0: starting wpa_supplicant
Mar  1 16:47:04 raspberrypi systemd[1]: Started Phoniebox GPIO Buttons Service.
Mar  1 16:47:04 raspberrypi kernel: [    9.150981] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
Mar  1 16:47:04 raspberrypi systemd[1]: Starting The PHP 7.3 FastCGI Process Manager...
Mar  1 16:47:04 raspberrypi systemd[1]: Started Phoniebox RFID-Reader Service.
Mar  1 16:47:04 raspberrypi systemd[1]: Starting OpenBSD Secure Shell server...
Mar  1 16:47:04 raspberrypi systemd[1]: Started dphys-swapfile - set up, mount/unmount, and delete a swap file.
Mar  1 16:47:04 raspberrypi systemd[1]: Started Permit User Sessions.
Mar  1 16:47:04 raspberrypi systemd[1]: Started /etc/rc.local Compatibility.
Mar  1 16:47:04 raspberrypi systemd[1]: Started Mopidy music server.
Mar  1 16:47:04 raspberrypi systemd[1]: Starting Terminate Plymouth Boot Screen...
Mar  1 16:47:04 raspberrypi systemd[1]: Starting Hold until boot process finishes up...
Mar  1 16:47:04 raspberrypi kernel: [    9.459840] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Mar  1 16:47:04 raspberrypi kernel: [    9.459864] brcmfmac: power management disabled
Mar  1 16:47:04 raspberrypi systemd[1]: Received SIGRTMIN+21 from PID 173 (plymouthd).
Mar  1 16:47:04 raspberrypi systemd[1]: Received SIGRTMIN+21 from PID 173 (plymouthd).
Mar  1 16:47:04 raspberrypi systemd[1]: plymouth-start.service: Succeeded.
Mar  1 16:47:04 raspberrypi systemd[1]: plymouth-quit.service: Succeeded.
Mar  1 16:47:04 raspberrypi systemd[1]: Started Terminate Plymouth Boot Screen.
Mar  1 16:47:04 raspberrypi systemd[1]: plymouth-quit-wait.service: Succeeded.
Mar  1 16:47:04 raspberrypi systemd[1]: Started Hold until boot process finishes up.
Mar  1 16:47:04 raspberrypi systemd[1]: Started Getty on tty1.
Mar  1 16:47:04 raspberrypi systemd[1]: Reached target Login Prompts.
Mar  1 16:47:05 raspberrypi raspi-config[337]: Checking if shift key is held down: No. Switching to ondemand scaling governor.
Mar  1 16:47:05 raspberrypi systemd[1]: Started LSB: Switch to ondemand cpu governor (unless shift key is pressed).
Mar  1 16:47:05 raspberrypi dhcpcd[409]: wlan0: connected to Access Point `'
Mar  1 16:47:05 raspberrypi dhcpcd[409]: eth0: waiting for carrier
Mar  1 16:47:05 raspberrypi dhcpcd[409]: wlan0: waiting for carrier
Mar  1 16:47:05 raspberrypi dhcpcd[409]: wlan0: carrier acquired
Mar  1 16:47:05 raspberrypi kernel: [   10.208946] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
Mar  1 16:47:05 raspberrypi kernel: [   10.209196] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
Mar  1 16:47:05 raspberrypi systemd[1]: Started OpenBSD Secure Shell server.
Mar  1 16:47:05 raspberrypi dhcpcd[409]: DUID 00:01:00:01:25:1e:c3:50:b8:27:eb:98:5c:47
Mar  1 16:47:05 raspberrypi dhcpcd[409]: wlan0: IAID eb:98:5c:47
Mar  1 16:47:05 raspberrypi dhcpcd[409]: wlan0: adding address fe80::a366:9e9b:be78:15ba
Mar  1 16:47:05 raspberrypi dhcpcd[409]: wlan0: carrier lost
Mar  1 16:47:05 raspberrypi avahi-daemon[355]: Joining mDNS multicast group on interface wlan0.IPv6 with address fe80::a366:9e9b:be78:15ba.
Mar  1 16:47:05 raspberrypi avahi-daemon[355]: New relevant interface wlan0.IPv6 for mDNS.
Mar  1 16:47:05 raspberrypi avahi-daemon[355]: Registering new address record for fe80::a366:9e9b:be78:15ba on wlan0.*.
Mar  1 16:47:05 raspberrypi dhcpcd[409]: wlan0: deleting address fe80::a366:9e9b:be78:15ba
Mar  1 16:47:05 raspberrypi avahi-daemon[355]: Withdrawing address record for fe80::a366:9e9b:be78:15ba on wlan0.
Mar  1 16:47:05 raspberrypi avahi-daemon[355]: Leaving mDNS multicast group on interface wlan0.IPv6 with address fe80::a366:9e9b:be78:15ba.
Mar  1 16:47:05 raspberrypi avahi-daemon[355]: Interface wlan0.IPv6 no longer relevant for mDNS.
Mar  1 16:47:05 raspberrypi systemd[1]: Started Lighttpd Daemon.
Mar  1 16:47:06 raspberrypi systemd[1]: Started The PHP 7.3 FastCGI Process Manager.
Mar  1 16:47:06 raspberrypi exim4[440]: Starting MTA: exim4.
Mar  1 16:47:06 raspberrypi systemd[1]: Started LSB: exim Mail Transport Agent.
[... cut ...]
Mar  1 16:51:45 raspberrypi python2[462]: volume: 30%   repeat: off   random: off   single: off   consume: off
Mar  1 16:51:45 raspberrypi mopidy[465]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar  1 16:51:45 raspberrypi python2[462]: volume: 30%   repeat: off   random: off   single: off   consume: off
Mar  1 16:51:45 raspberrypi mopidy[465]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar  1 16:51:45 raspberrypi python2[462]: OK MPD 0.19.0
Mar  1 16:51:45 raspberrypi python2[462]: OK
`

kuesimon, 2020-03-01:
I also made a image of the login screen where you can see the behaviour:

[image]

The odd thing is, that my spotifiy playlist is not started anymore. 
--> Then i realized this behaviour. Could there be some connection or is it "standard behaviour" as the rfid reader ist treated like a keyboard?

kuesimon, 2020-03-01:
I now found my error for the non running spotify part. 
The remaining question is: if it's an expected behaviour that the rfid is acting like keyboard.

If yes this issue can be closed.

s-martin, 2020-03-01:
Yes, the RFID reader is essentially a keyboard.

closed 2020-03-02 by kuesimon