MiczFlor/RPi-Jukebox-RFID #826: Question: Adding a microphone for singing

Discoverer81, 2020-03-02:
Hello,

i´m building the phoniebox for my son and i would like to add a microphone, so that he can also sing with it.

I have an USB-Soundcard and want to stick the cinch of the microphone in it.

Now my question, how can i add this without destroying any important codes which Phoniebox needs?

Does anybody has this option and can share his programming please?

Thanks in advance!

s-martin, 2020-03-04:
Do you want to record something from a microphone? Then you would need to add a dedicated record GPIO button and uncomment the recording functionality in `scripts/gpio-buttons.py`.

Using the IN port of the USB sound card should work, but I haven't done it myself, so I don't know, if you need to change some other settings, too.

Discoverer81, 2020-03-06:
Hello, 

i´ve checked it on alsamixer, but my micro isn´t shown there.
The mirco is dont defect, i also checked to be sure.

Even so, my target is to speack into the micro and voice come out of the boxes.
Do i need sth additional that this works?

s-martin, 2020-03-06:
Unfortunately I don't have much experience with alsa, so I can't really help you. But I think your desired functionality should be possible with alsamixer.

ZyanKLee, 2020-03-07:
I'm also not sure how this works with alsa, but it is surely just a matter of settings. A web-search for alsa and microphone to speaker should being helpful sources.

Discoverer81, 2020-03-09:
Hi,

checked on this, but found nothing.
I think i need something to set microphon as default (always on)

ZyanKLee, 2020-03-09:
> always on
😨 

at least its not sending the audio off to some proprietary cloud 😆