MiczFlor/RPi-Jukebox-RFID #827: 🚀 When upgrading - script should not ask for spotify credentials again

tommy-rk, 2020-03-03:
## Feature Description
When upgrading the system on Buster by rerunning the installation script, it should not ask again for spotify credentials if they already exist.

### How do you envision the feature to work from a users perspective?
Upgrading the system should get easier.

s-martin, 2020-05-15:
@fredg02, ist this already fixed with recent changes of the install script?

fredg02, 2020-05-15:
No, this is not fixed yet. Should not be too difficult to do though.

Groovylein, 2020-05-23:
I think we should implement this before releasing 2.0 to ensure a smooth upgrade path!

MiczFlor, 2020-05-23:
I think the best way to do this would be to alter the install script as such:

* at the end of the installation, copy `${HOME_DIR}/PhonieboxInstall.conf` to the `settings` folder
* when running the install script, it alerts anyway: there is an existing installation. Here we could add:
    * *Found PhonieboxInstall.conf script. Do you want to use this for the new install?*
    * If yes: 
        * copy `PhonieboxInstall.conf` to `${HOME_DIR}`
        * run a non-interactive install with this config file
    * If no: aks for everything

Any volunteers? :) I can do it later tonight, if noone wants to sink their teeth into the task :)

Groovylein, 2020-05-23:
I was on it and now I see your comment @MiczFlor. I'm not a fan of this proposal, but its a quick (and dirty) way.

I thought about reading from `~/.config/mopidy/mopidy.conf`. What do you and others think?

fredg02, 2020-05-23:
> I think the best way to do this would be to alter the install script as such:
> 
>    * at the end of the installation, copy `${HOME_DIR}/PhonieboxInstall.conf` to the `settings` folder

@MiczFlor Why do you want to copy `PhonieboxInstall.conf` to the settings folder and back to it later instead of just keeping it in `${HOME_DIR}`?

> I thought about reading from `~/.config/mopidy/mopidy.conf`. What do you and others think?

@Groovylein I think, using the `PhonieboxInstall.conf` is the straightforward way to re-use existing config settings in the install script. Extracting the info from `mopidy.conf` so it can be used in the install script seems to be an unnecessary extra step.

mr-gosh, 2020-11-10:
stumbled upon this during the search after a documentation for an upgrade path - is this thread dead or are there simple steps to be sure not to reconfigure everything after an upgrade?

s-martin, 2020-11-21:
Copying a 
${HOME_DIR}/PhonieboxInstall.conf file should work, but be sure to backup your music and playlists

s-martin, 2024-11-10:
Spotify implementation has been undergone various improvements.
Please check release [2.8.0](https://github.com/MiczFlor/RPi-Jukebox-RFID/releases/tag/v2.8.0).

Closing now, please reopen, if the issue persists.

closed 2024-11-10 by s-martin