MiczFlor/RPi-Jukebox-RFID #828: 🚀 when upgrading - script should detect if the installed version is the current version

tommy-rk, 2020-03-03:
## Feature Description
When upgrading with the installer script on Buster it already detects the installed version of RPi-Jukebox-RFID, but it doesn't know, if it is the current version and doesn't need to update at all.

### How do you envision the feature to work from a users perspective?
Make updates easier.

s-martin, 2020-03-03:
I agree this could be a nice enhancement.
We read already the used git commit during installation, if we save the commit number, we could compare it in future installation.

If the commit number is different, I think we can assume that the commit number coming from the repo is newer.

s-martin, 2020-03-03:
The version information is maintained here: https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/settings/version

This numbering is updated rarely, so currently the information is not very useful. The installer uses the latest commit from master branch, so the newest updates are used, but the version should reflect this or be dropped altogether.

An idea could be to store the git commit in the version file (maybe while keeping the major version), so this information is useful and can be used to determine, if the install script should run at all.

@MiczFlor, what do you think?

See also #829.

closed 2020-03-23 by MiczFlor