MiczFlor/RPi-Jukebox-RFID #829: 🐛 Upgrade from 2.0 rc7 on Buster stays at 2.0 rc7

tommy-rk, 2020-03-03:
## Bug
Upgrade from 2.0 rc7 on Buster stays at 2.0 rc7

### What I did
Re-running the installation script on Buster to get from 2.0 rc7 to 2.0 rc8. 

### What happened
After all finished successfully the version stayed at 2.0 rc7

### I expected this to happen
I expected to have 2.0 rc8 installed correctly.

## Software

### Base image and version

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian

### Branch / Release

Refresh index: 100% (309/309), done.
On branch master
Your branch is up to date with 'origin/master'.

### Installscript
cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh

Refresh index: 100% (309/309), done.
On branch master
Your branch is up to date with 'origin/master'.

## Hardware
### RaspberryPi version
Raspberry Pi 3 Model B Plus Rev 1.3pi

s-martin, 2020-03-03:
The version information is maintained here: https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/settings/version 

This numbering is updated rarely, so currently the information is not very useful. The installer uses the latest commit from master branch, so you are up to date, but I agree the version should reflect this or be dropped altogether. Maybe we should use the commit number, see #828.

The rc8 you were expecting is currently set only in `develop` not `master` branch (where it's still rc7), which makes it even more confusing.

s-martin, 2020-03-03:
I would coordinate this enhancement in #828, so we could close this ticket here, if you agree.

tommy-rk, 2020-03-03:
If this is the same problem we should merge it. (I follow the rule: Only one issue per bug report.)

s-martin, 2020-03-03:
Tracked in #828

closed 2020-03-03 by s-martin