MiczFlor/RPi-Jukebox-RFID #830: Question: Adding button which plays sound

Discoverer81, 2020-03-03:
Hello,

i would like to add a button, which plays a sound from the Phoniebox-Libary.

How / where can i change the functionality of a button (using GPIO-Script).
What do i have to change?

Thanks in advance!

s-martin, 2020-03-04:
Generally the idea of Phoniebox is that you swipe an RFID card to play a song, album, podcast, streaming service or sound.

However, if you want to use a fixed button for that, you would need to change the following:
Starting point is `scripts/gpio-buttons.py`, where GPIO numbers of used button are configured, see also https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Using-GPIO-hardware-buttons 

The actual functionality is defined in `scripts/playout_controls.sh`, so that's the place where you would need to add the funtionality to play a sound, when a button is pressed.

Discoverer81, 2020-03-04:
Hello!

I've searched now a long time and found sth. like this.

But unfortunatly it does nit work.
Can somebody find the issue?

udo nano /home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py

def def_playlist_rock():
check_call("mpc clear && mpc load Rock && mpc play", shell=True)
pl_rock = Button(27)
pl_rock.when_pressed = def_playlist_rock

sudo reboot

Thanks

s-martin, 2020-03-04:
What's the error message?

I guess you need `Button(27,pull_up=True)`. 

Can you show the whole file?

Plese check that you connected the right pin. The number 27 is the GPIO 27, the pin is 13, see here: https://www.raspberrypi-spy.co.uk/2012/06/simple-guide-to-the-rpi-gpio-header-and-pins/

Discoverer81, 2020-03-08:
Hello,

Problem was solved. 
It was a Problem in naming the playlist & a wrong format.

1st. I named it "Paw Patrol - Sounds".
The system only could read "Paw", so after changing the name in "PawPatrolSounds" it worked.

2nd: I used a wav-format ,but this format can´t be read. So i converted it into mp3 and now it works.

Thank you!

closed 2020-03-08 by Discoverer81

ZyanKLee, 2020-03-08:
great this worked out. thanks for the feedback.