MiczFlor/RPi-Jukebox-RFID #832: 🐛 | card ID does not update

cedivan0718, 2020-03-05:
## Bug

### What I did

### What happened

When trying to register an rfid card on the web interface, the rfid card reader would scan but the “last used chip ID” does not update with the rfid card number. With this I can not register any cards. Please let me know what I can try doing.

s-martin, 2020-03-06:
What do you mean by "would scan"? Did you just hear the beep or did anything else happen?

What OS are you using, do you use classic or Spotify edition (did you use the regular master install script or develop), which RFID reader are you using?

Please make sure to check the following instructions:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/CONFIGURE-stretch#identify-and-configure-your-rfid-reader

And can you swipe a card and afterwards post the output of `tail -n 500 /var/log/syslog`?

cedivan0718, 2020-03-06:
The rfid reader would beep and flash green when i scanned a card but it wouldn’t actually transfer to the web interface. I am using the September 2019 version of buster and i installed the master script . The rfid card reader that i am using is this one from amazon https://www.amazon.com/Reader-LANMU-125khz-Contactless-Proximity/dp/B07B7H6CQ2/ref=mp_s_a_1_4?keywords=rfid+card+reader&qid=1583532921&sr=8-4

cedivan0718, 2020-03-06:
The rfid reader would beep and flash green when i scanned a card but it wouldn’t actually transfer to the web interface. I am using the September 2019 version of buster and i installed the master script . The rfid card reader that i am using

s-martin, 2020-03-07:
Did you check, if the reader is registered correctly as described here https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/CONFIGURE-stretch#identify-and-configure-your-rfid-reader ?

ZyanKLee, 2020-03-07:
also a lack of permissions to write to the according files may break this:
`ls -al /home/pi/RPi-Jukebox-RFID/settings/*`
user `pi` should have write access to them.

cedivan0718, 2020-03-09:
Hello, sorry for not replying I have been busy. But I ran the registerdevice.py script again and it had given me an error in line 24 saying that there was “no module called evdev”. I used the command “sudo pip3 install evdev” to install evdev and once I did I tried running the registerdevice.py script again and fortunately it worked. I selected my Rfid card reader then rebooted the whole system again and now it is scanning properly. Thank you for your help if any more problems occur I’ll be glad to let you guys know.

cedivan0718, 2020-03-09:
But now there seems to be a problem with iris. When i click on the spotify tab from the web interface it tells me that the site can not be reached. Any ideas to what this could be?

s-martin, 2020-03-09:
Please try
```
echo -e "Package: mopidy\nPin: version 3.*\nPin-Priority: 1001" | sudo tee /etc/apt/preferences.d/mopidy

wget -q -O - https://apt.mopidy.com/mopidy.gpg | sudo apt-key add -
sudo wget -q -O /etc/apt/sources.list.d/mopidy.list https://apt.mopidy.com/buster.list

sudo apt-get update
sudo apt-get upgrade
sudo apt-get --yes --allow-downgrades --allow-remove-essential --allow-change-held-packages install mopidy mopidy-mpd mopidy-local mopidy-spotify

sudo apt-get --yes --allow-downgrades --allow-remove-essential --allow-change-held-packages install libspotify12 python3-cffi python3-ply python3-pycparser python3-spotify

sudo python3 -m pip install Mopidy-Iris
```

cedivan0718, 2020-03-09:
upon putting in "sudo apt-get upgrade" it tells me that "after the operation, 24.1 MB of additional disk space will be used do you want to continue? [Y/n]" I selected Y and it aborts and does not continue to do anything. Also the last command "sudo python -m pip install Mopidy-Iris"  gives me this "Mopidy-Iris requires Python '>=3.7' but the running Python is 3.5.3"

s-martin, 2020-03-09:
> upon putting in "sudo apt-get upgrade" it tells me that "after the operation, 24.1 MB of additional disk space will be used do you want to continue? [Y/n]" I selected Y and it aborts and does not continue to do anything. 

This could just take a while. Wait a couple of minutes.

> Also the last command "sudo python -m pip install Mopidy-Iris" gives me this "Mopidy-Iris requires Python '>=3.7' but the running Python is 3.5.3"

Are you using Raspian Stretch or Buster?

cedivan0718, 2020-03-09:
I am using buster

s-martin, 2020-03-09:
What does `python --version` return?

cedivan0718, 2020-03-09:
Python 2.7.13

s-martin, 2020-03-09:
Sorry `python3 --version`

cedivan0718, 2020-03-09:
Python 3.5.3

s-martin, 2020-03-09:
Please try again `sudo apt update && sudo apt upgrade`

cedivan0718, 2020-03-09:
sudo apt update
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 https://apt.mopidy.com buster InRelease                                  
Hit:3 http://mirrordirector.raspbian.org/raspbian stretch InRelease            
Reading package lists... Error!
E: Problem parsing dependency 20
E: Error occurred while processing tkcts (NewVersion2)
E: Problem with MergeList /var/lib/apt/lists/mirrordirector.raspbian.org_raspbian_dists_stretch_main_binary-armhf_Packages
E: The package lists or status file could not be parsed or opened.

sudo apt upgrade
Reading package lists... Error!
E: Problem parsing dependency 20
E: Error occurred while processing tkcts (NewVersion2)
E: Problem with MergeList /var/lib/apt/lists/mirrordirector.raspbian.org_raspbian_dists_stretch_main_binary-armhf_Packages
E: The package lists or status file could not be parsed or opened.

s-martin, 2020-03-09:
You are still on stretch.

To upgrade you need to remove the path to Mopidy packages with
`sudo rm  /etc/apt/sources.list.d/mopidy.list`

and afterwards follow the instructions here
http://baddotrobot.com/blog/2019/08/29/upgrade-raspian-stretch-to-buster/
(This could take an hour)

and then again https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/832#issuecomment-596727818

cedivan0718, 2020-03-09:
I now have raspian buster on my pi and I re-installed phoniebox. the RFID scanner is working as it should be and the scans are going through and updating to the web interface. Now the problem is that when I link a song to a card and try to scan it so it can play nothing happens. do you have an idea as to what it can be?

s-martin, 2020-03-10:
Please swipe the card and afterwards post the results of `journalctl -u mopidy` here.

cedivan0718, 2020-03-11:
Logs begin at Mon 2020-03-09 18:43:03 PDT, end at Tue 2020-03-10 17:12:25 PDT
Mar 10 17:04:15 raspberrypi systemd[1]: Starting Mopidy music server...
Mar 10 17:04:15 raspberrypi systemd[1]: Started Mopidy music server.
Mar 10 17:04:22 raspberrypi mopidy[458]: /usr/lib/python2.7/dist-packages/mopidy
Mar 10 17:04:22 raspberrypi mopidy[458]:   extension_class = entry_point.load(re
Mar 10 17:04:26 raspberrypi mopidy[458]: INFO     Starting Mopidy 2.2.2
Mar 10 17:04:26 raspberrypi mopidy[458]: INFO     Loading config from builtin de
Mar 10 17:04:26 raspberrypi mopidy[458]: INFO     Loading config from /etc/mopid
Mar 10 17:04:26 raspberrypi mopidy[458]: INFO     Loading config from command li
Mar 10 17:04:26 raspberrypi mopidy[458]: INFO     Enabled extensions: iris, mpd,
Mar 10 17:04:26 raspberrypi mopidy[458]: INFO     Disabled extensions: none
Mar 10 17:04:27 raspberrypi mopidy[458]: INFO     Starting Mopidy mixer: Softwar
Mar 10 17:04:27 raspberrypi mopidy[458]: INFO     Mixer volume set to 30
Mar 10 17:04:27 raspberrypi mopidy[458]: INFO     Starting Mopidy audio
Mar 10 17:04:27 raspberrypi mopidy[458]: INFO     Starting Mopidy backends: Spot
Mar 10 17:04:27 raspberrypi mopidy[458]: INFO     No local library metadata cach
Mar 10 17:04:27 raspberrypi mopidy[458]: INFO     Loaded 0 local tracks using js
Mar 10 17:04:27 raspberrypi mopidy[458]: INFO     Audio output set to "alsasink"
Mar 10 17:04:27 raspberrypi mopidy[458]: ERROR    Spotify login error: <ErrorTyp
Mar 10 17:04:28 raspberrypi mopidy[458]: INFO     Logged into Spotify Web API as
Mar 10 17:04:28 raspberrypi mopidy[458]: INFO     Started spotify backend
Mar 10 17:04:28 raspberrypi mopidy[458]: INFO     Starting Mopidy core
Mar 10 17:04:28 raspberrypi mopidy[458]: INFO     Starting Mopidy frontends: Iri
lines 1-23...skipping...
-- Logs begin at Mon 2020-03-09 18:43:03 PDT, end at Tue 2020-03-10 17:12:25 PDT
Mar 10 17:04:15 raspberrypi systemd[1]: Starting Mopidy music server...
Mar 10 17:04:15 raspberrypi systemd[1]: Started Mopidy music server.
Mar 10 17:04:22 raspberrypi mopidy[458]: /usr/lib/python2.7/dist-packages/mopidy
Mar 10 17:04:22 raspberrypi mopidy[458]:   extension_class = entry_point.load(re
Mar 10 17:04:26 raspberrypi mopidy[458]: INFO     Starting Mopidy 2.2.2
Mar 10 17:04:26 raspberrypi mopidy[458]: INFO     Loading config from builtin de
Mar 10 17:04:26 raspberrypi mopidy[458]: INFO     Loading config from /etc/mopid
Mar 10 17:04:26 raspberrypi mopidy[458]: INFO     Loading config from command li
Mar 10 17:04:26 raspberrypi mopidy[458]: INFO     Enabled extensions: iris, mpd,
Mar 10 17:04:26 raspberrypi mopidy[458]: INFO     Disabled extensions: none
Mar 10 17:04:27 raspberrypi mopidy[458]: INFO     Starting Mopidy mixer: Softwar
Mar 10 17:04:27 raspberrypi mopidy[458]: INFO     Mixer volume set to 30
Mar 10 17:04:27 raspberrypi mopidy[458]: INFO     Starting Mopidy audio
Mar 10 17:04:27 raspberrypi mopidy[458]: INFO     Starting Mopidy backends: Spot
Mar 10 17:04:27 raspberrypi mopidy[458]: INFO     No local library metadata cach
Mar 10 17:04:27 raspberrypi mopidy[458]: INFO     Loaded 0 local tracks using js
Mar 10 17:04:27 raspberrypi mopidy[458]: INFO     Audio output set to "alsasink"
Mar 10 17:04:27 raspberrypi mopidy[458]: ERROR    Spotify login error: <ErrorTyp
Mar 10 17:04:28 raspberrypi mopidy[458]: INFO     Logged into Spotify Web API as
Mar 10 17:04:28 raspberrypi mopidy[458]: INFO     Started spotify backend
Mar 10 17:04:28 raspberrypi mopidy[458]: INFO     Starting Mopidy core
Mar 10 17:04:28 raspberrypi mopidy[458]: INFO     Starting Mopidy frontends: Iri
Mar 10 17:04:28 raspberrypi mopidy[458]: INFO     Starting Iris 3.43.0
Mar 10 17:04:28 raspberrypi mopidy[458]: INFO     MPD server running at [::ffff:
Mar 10 17:04:29 raspberrypi mopidy[458]: INFO     Starting GLib mainloop
Mar 10 17:04:29 raspberrypi mopidy[458]: INFO     HTTP server running at [::ffff
Mar 10 17:04:31 raspberrypi mopidy[458]: INFO     Loaded 13 playlists
lines 1-28

-- Logs begin at Mon 2020-03-09 18:43:03 PDT, end at Tue 2020-03-10 17:12:25 PDT. --
Mar 10 17:04:15 raspberrypi systemd[1]: Starting Mopidy music server...
Mar 10 17:04:15 raspberrypi systemd[1]: Started Mopidy music server.
Mar 10 17:04:22 raspberrypi mopidy[458]: /usr/lib/python2.7/dist-packages/mopidy/ext.py:202
Mar 10 17:04:22 raspberrypi mopidy[458]:   extension_class = entry_point.load(require=False
Mar 10 17:04:26 raspberrypi mopidy[458]: INFO     Starting Mopidy 2.2.2
Mar 10 17:04:26 raspberrypi mopidy[458]: INFO     Loading config from builtin defaults
Mar 10 17:04:26 raspberrypi mopidy[458]: INFO     Loading config from /etc/mopidy/mopidy.co
Mar 10 17:04:26 raspberrypi mopidy[458]: INFO     Loading config from command line options
Mar 10 17:04:26 raspberrypi mopidy[458]: INFO     Enabled extensions: iris, mpd, http, stre
Mar 10 17:04:26 raspberrypi mopidy[458]: INFO     Disabled extensions: none
Mar 10 17:04:27 raspberrypi mopidy[458]: INFO     Starting Mopidy mixer: SoftwareMixer
Mar 10 17:04:27 raspberrypi mopidy[458]: INFO     Mixer volume set to 30
Mar 10 17:04:27 raspberrypi mopidy[458]: INFO     Starting Mopidy audio
Mar 10 17:04:27 raspberrypi mopidy[458]: INFO     Starting Mopidy backends: SpotifyBackend,
Mar 10 17:04:27 raspberrypi mopidy[458]: INFO     No local library metadata cache found at 
[... 48 lines cut ...]
Mar 10 17:04:22 raspberrypi mopidy[458]:   extension_class = entry_point.load(require=False)
[... cut ...]
Mar 10 17:04:28 raspberrypi mopidy[458]: INFO     Starting Mopidy frontends: IrisFrontend, MpdFrontend, HttpFrontend
Mar 10 17:04:28 raspberrypi mopidy[458]: INFO     Starting Iris 3.43.0
Mar 10 17:04:28 raspberrypi mopidy[458]: INFO     MPD server running at [::ffff:0.0.0.0]:6600
Mar 10 17:04:29 raspberrypi mopidy[458]: INFO     Starting GLib mainloop
Mar 10 17:04:29 raspberrypi mopidy[458]: INFO     HTTP server running at [::ffff:0.0.0.0]:6680
Mar 10 17:04:31 raspberrypi mopidy[458]: INFO     Loaded 13 playlists
lines 1-28

ZyanKLee, 2020-03-11:
Seems like there is an error I'm the Spotify user data:

Mar 10 17:04:27 raspberrypi mopidy[458]: ERROR Spotify login error:
<ErrorType.BAD_USERNAME_OR_PASSW

cedivan0718, 2020-03-11:
any idea as to what I can do?

s-martin, 2020-03-11:
The error message suggests you Spotify login data is wrong.

cedivan0718, 2020-03-11:
I am 100% sure that my spotify login information is correct. I checked the Phonieboxinstall.conf file and the login information is what it should be. Any other suggestions as to what I can do?

ZyanKLee, 2020-03-11:
have a look at `/etc/mopidy/mopidy.conf` if the information in there is matching what you expect.

cedivan0718, 2020-03-11:
this is the output of that command "bash: /etc/mopidy/mopidy.conf: Permission denied" should I be typing in something before that?

s-martin, 2020-03-11:
Try `cat /etc/mopidy/mopidy.conf` to have a look at that file.

cedivan0718, 2020-03-11:
still getting permission denied

s-martin, 2020-03-11:
Sorry `sudo cat /etc/mopidy/mopidy.conf`

cedivan0718, 2020-03-11:
I believe I see what the error could possibly be, how could I edit my username?

s-martin, 2020-03-12:
With `sudo nano /etc/mopidy/mopidy.conf` you open the file in an editor.

`CTRL+O` saves the file, `CTRL+X` closes the file.

You need to restart mopidy with `sudo systemctl restart mopidy`.

If you don't have nano installed, try ` sudo apt install nano`.

cedivan0718, 2020-03-12:
updated the country code, locale code, and put in the correct username and everything is back up to speed. thank you very much. I just have one other question. I currently have the raspberry pi set up for my house but I want to take the pi over to my parents house as a gift because they would probably enjoy it. Would I have to change the IP address that I'm currently using for one that fits in their networks? or how would i go about doing this

s-martin, 2020-03-13:
See this section how to add another WiFi https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#i-am-moving-how-do-i-get-the-phoniebox-into-my-new-wifi-network

s-martin, 2020-03-18:
Closing this now as the question of the issue has been resolved.

Feel free to reopen it, if this issue wasn't resolved or open another issue for other problems.

closed 2020-03-18 by s-martin