MiczFlor/RPi-Jukebox-RFID #833: 🚀 | Podcasts automatisch offline verfügbar machen.

Moppel2k, 2020-03-06:
## Feature Description

### What functionality would you like to see in your phoniebox?

i.e. `Podcasts auch automatisch offline verfügbar zu machen`

### How do you envision the feature to work from a users perspective?

i.e. `Hallo zusammen
nachdem ich nun unsere PhonieBox wieder am laufen habe und meine Tochter wieder begeistert hören kann, hat sich mir eine Frage gestellt.
Ist es möglich Podcasts auch automatisch offline verfügbar zu machen?
Der Grund ist das unsere Box mobil ist und die W-Lan Abdeckung in der Wohnung nicht so super ist (Stahlbeton) und im Urlaub wäre das auch ganz praktisch :-)

Speziell am Ohrenbär ist das, die neuen Folgen «sortiert» abgelegt werden müssten, da hier zusammenhängende Geschichten erscheinen. 
Leider ist es so das beim starten über den Link immer die erste Folge des Monats abgespielt wird, die nächste kommt danach… das ist innerhalb einer Geschichte gut, allerdings wenn eine neue Geschichte dazu kommt, müssen erst die alten durchgehört werden…
Schön wäre auch eine Option die Dateiablage zu begrenzen also maximal «n» Geschichte danach first in first out.

Wäre das auch für andere eine interessante Ergänzung? 
Oder gibt es die Möglichkeit sogar schon irgendwo?

Danke schon mal für eure Hilfe!`

### Further information that might help

i.e. `Es geht bei uns um folgende Podcasts:
Die Maus Podcasts:
Die Maus - 30 - URL: https://kinder.wdr.de/radio/diemaus/audio/diemaus-30/diemaus-30-104.podcast 
Die Maus - 60 - URL: https://kinder.wdr.de/radio/diemaus/audio/diemaus-60/diemaus-60-106.podcast 
Die Sendung mit der Maus - zum Hören (120 Minuten) - URL: https://kinder.wdr.de/radio/diemaus/audio/diemaus-120/diemaus-120-104.podcast 
Die Maus - Musik - URL: https://kinder.wdr.de/radio/diemaus/audio/diemaus-musik/diemaus-musik-106.podcast 

Ohrenbär:
https://www.ohrenbaer.de/podcast/podcast.feed.podcast.xml
`

iwishitwassummer, 2020-07-08:
fände ich ja auch spannend, aber man muss ja auch noch beachten, dass man genügend Speicherplatz frei hat oder wie die vorschlägt einen X Dateien speichern dürfen. Die Geschichten noch auseinander zu halten, wäre glaube ich ganz schön aufwendig, zumal die Folgen nicht standardisiert benannt werden, also jeder Podcast macht das anders.
Dann muss man die Ordner "stream" und "download" vielleicht auch trennen, oder wie stellst du dir das vor: wenn kein Wifi, nutze geladenen Folge, wenn Wifi streame und downloade im Hintergrund?
Skripte gibt es hier auf Git z.B.:
https://github.com/dplocki/podcast-downloader

wenn man sowas nehmen würde, macht man erstmal doppelte Buchführung, oder muss die Podcast Ordner und podcast.txt auslesen und ins JSON (für obige Beispiel) schreiben. Ginge vielleicht auch.

belinea4071, 2020-07-23:
I used Podget for this task. Podget is the repository on debian buster and an therefore an easy install.
If you follow these steps you get everything set up.
1.	_sudo apt-get install -y podget_ ‘ install Podget
2.	_/usr/bin/podget_  ‘start Podget
3.	_nano .podget/podgetrc_  ‘edit the settings, Change the Location to the audiofolder, add it if it doesn’t exist.
**DIR_LIBRARY=/home/pi/RPi-Jukebox-RFID/shared/audiofolders/Podcast**
-->	save and exit
4.	_nano .podget/serverlist_  ‘Add the Podcast URL to the File

https://kinder.wdr.de/radio/diemaus/audio/diemaus-30/diemaus-30-104.podcast DieMaus-30  OPT_FEED_PLAYLIST_OLDFIRST
https://kinder.wdr.de/radio/diemaus/audio/diemaus-60/diemaus-60-106.podcast DieMaus-60  OPT_FEED_PLAYLIST_OLDFIRST
https://kinder.wdr.de/radio/diemaus/audio/diemaus-120/diemaus-120-104.podcast DieSendungmitderMaus   OPT_FEED_PLAYLIST_OLDFIRST
https://kinder.wdr.de/radio/diemaus/audio/diemaus-musik/diemaus-musik-106.podcast DieMausMusik  OPT_FEED_PLAYLIST_OLDFIRST
https://www.ohrenbaer.de/podcast/podcast.feed.podcast.xml Ohrenbär  OPT_FEED_PLAYLIST_OLDFIRST
https://www.srf.ch/feed/podcast/sd/88072903-e34f-4f51-a9b9-052562d049c2.xml ZamboHörspiele  OPT_FEED_PLAYLIST_OLDFIRST

5.	_sudo nano crontab -e_ ‘Add the Refresh to crontab with the following line
_15 05 * * * /usr/bin/podget -s_

belinea4071, 2020-07-23:
@iwishitwassummer, in Podget is also an option for minimal space left, check out **podgetrc**.
MIN_SPACE=614400 (600MB).

belinea4071, 2020-07-26:
I did some more digging at this topic and found an even nicer solution. I wasn't happy with the file names. Therefore, I ended up using **youtube-dl** to download the Podcast.

I just wrote a small sript with the Podcast URL and Crontab is checking it every morning.

Set up a new script:
`sudo nano /home/pi/RPi-Jukebox-RFID/scripts/podcast.sh`

insert the following lines:
```
#! /bin/sh

youtube-dl -o '/home/pi/RPi-Jukebox-RFID/shared/audiofolders/Podcast/Die_Maus_60/%(upload_date)s_%(title)s.%(ext)s' https://kinder.wdr.de/radio/diemaus/audio/diemaus-60/diemaus-60-106.podcast
youtube-dl -o '/home/pi/RPi-Jukebox-RFID/shared/audiofolders/Podcast/Die_Sendung_mit_der_Maus/%(upload_date)s_%(title)s.%(ext)s' https://kinder.wdr.de/radio/diemaus/audio/diemaus-120/diemaus-120-104.podcast
youtube-dl -o '/home/pi/RPi-Jukebox-RFID/shared/audiofolders/Podcast/Die_Maus_Musik/%(upload_date)s_%(title)s.%(ext)s' https://kinder.wdr.de/radio/diemaus/audio/diemaus-musik/diemaus-musik-106.podcast
youtube-dl -o '/home/pi/RPi-Jukebox-RFID/shared/audiofolders/Podcast/Ohrenbär/%(upload_date)s_%(title)s.%(ext)s' https://www.ohrenbaer.de/podcast/podcast.feed.podcast.xml
youtube-dl -o '/home/pi/RPi-Jukebox-RFID/shared/audiofolders/Podcast/Zambo_Hörgeschichten/%(upload_date)s_%(title)s.%(ext)s' https://www.srf.ch/feed/podcast/sd/88072903-e34f-4f51-a9b9-052562d049c2.xml

sudo chown -R pi:www-data /home/pi/RPi-Jukebox-RFID/shared
sudo chmod -R 777 /home/pi/RPi-Jukebox-RFID/shared
```

run the following lines so you can execute the sript.
`sudo chmod +x /home/pi/RPi-Jukebox-RFID/scripts/podcast.sh`

Add the script to crontab.
`crontab -e`

insert the line at the end.
`15 05 * * * /home/pi/RPi-Jukebox-RFID/scripts/podcast.sh`