MiczFlor/RPi-Jukebox-RFID #834: Play any Playlist from spotify

lovethisshit, 2020-03-07:
## Feature Description

### What functionality would you like to see in your phoniebox?
- Being able to cache/load any private or public spotify playlist and connect it to a rfid tag

### How do you envision the feature to work from a users perspective?
After carefully handpicking my favorite songs into the "Master Blaster Collection" Playlist on spotify, I want the phoniebox to be able to start this beautiful collection magically when I'm casually showing the related rfid tag over and over again.

### Further information that might help
Jokes aside: I couldn't find an issue/bug describing this and was surprised by this error from iris:

> Mar  6 13:59:56 raspberrypi mopidy[491]: ERROR    SpotifyBackend backend caused an exception.
Mar  6 13:59:56 raspberrypi mopidy[491]: Traceback (most recent call last):
Mar  6 13:59:56 raspberrypi mopidy[491]:   File "/usr/lib/python2.7/dist-packages/mopidy/core/library.py", line 19, in _backend_error_handling
Mar  6 13:59:56 raspberrypi mopidy[491]:     yield
Mar  6 13:59:56 raspberrypi mopidy[491]:   File "/usr/lib/python2.7/dist-packages/mopidy/core/library.py", line 237, in lookup
Mar  6 13:59:56 raspberrypi mopidy[491]:     result = future.get()
Mar  6 13:59:56 raspberrypi mopidy[491]:   File "/usr/lib/python2.7/dist-packages/pykka/threading.py", line 52, in get
Mar  6 13:59:56 raspberrypi mopidy[491]:     compat.reraise(*self._data['exc_info'])
Mar  6 13:59:56 raspberrypi mopidy[491]:   File "/usr/lib/python2.7/dist-packages/pykka/compat.py", line 12, in reraise
Mar  6 13:59:56 raspberrypi mopidy[491]:     exec('raise tp, value, tb')
Mar  6 13:59:56 raspberrypi mopidy[491]:   File "/usr/lib/python2.7/dist-packages/pykka/actor.py", line 201, in _actor_loop
Mar  6 13:59:56 raspberrypi mopidy[491]:     response = self._handle_receive(message)
Mar  6 13:59:56 raspberrypi mopidy[491]:   File "/usr/lib/python2.7/dist-packages/pykka/actor.py", line 295, in _handle_receive
Mar  6 13:59:56 raspberrypi mopidy[491]:     return callee(*message['args'], **message['kwargs'])
Mar  6 13:59:56 raspberrypi mopidy[491]:   File "/usr/local/lib/python2.7/dist-packages/Mopidy_Spotify-3.1.0-py2.7.egg/mopidy_spotify/library.py", line 34, in lookup
Mar  6 13:59:56 raspberrypi mopidy[491]:     uri)
Mar  6 13:59:56 raspberrypi mopidy[491]:   File "/usr/local/lib/python2.7/dist-packages/Mopidy_Spotify-3.1.0-py2.7.egg/mopidy_spotify/lookup.py", line 21, in lookup
Mar  6 13:59:56 raspberrypi mopidy[491]:     return lookup_playlist(session, web_session, config, uri)
Mar  6 13:59:56 raspberrypi mopidy[491]:   File "/usr/local/lib/python2.7/dist-packages/Mopidy_Spotify-3.1.0-py2.7.egg/mopidy_spotify/lookup.py", line 97, in lookup_playlist
Mar  6 13:59:56 raspberrypi mopidy[491]:     raise spotify.Error('Playlist Web API lookup failed')
Mar  6 13:59:56 raspberrypi mopidy[491]: Error: Playlist Web API lookup failed

Apparently, this feature was or is implemented lately in mopidy-spotify: e.g. https://github.com/mopidy/mopidy-spotify/pull/235

I really much like this project, if anybody could give some insight on the dependencies & missing parts I would add more details to this request.
Thanks in advance!

s-martin, 2020-03-07:
This should work with mopidy 3 and python3 (you are still using the python 2.7 version).

Try
```
# keep major verson 3 of mopidy
echo -e "Package: mopidy\nPin: version 3.*\nPin-Priority: 1001" | sudo tee /etc/apt/preferences.d/mopidy

wget -q -O - https://apt.mopidy.com/mopidy.gpg |sudo apt-key add -
sudo wget -q -O /etc/apt/sources.list.d/mopidy.list https://apt.mopidy.com/buster.list
sudo apt-get update
sudo apt-get upgrade
sudo apt-get --yes --allow-downgrades --allow-remove-essential --allow-change-held-packages install mopidy mopidy-mpd mopidy-local mopidy-spotify
sudo apt-get --yes --allow-downgrades --allow-remove-essential --allow-change-held-packages install libspotify12 python3-cffi python3-ply python3-pycparser python3-spotify

# Install necessary Python packages
sudo python3 -m pip install mopidy-iris
```
**Attention:* `sudo apt upgrade` upgrades your whole system.

See also #804

lovethisshit, 2020-03-11:
Thanks for the quick response. This somehow destroys my installation in the way that ultimately, the lighttpd server can't start anymore. From the logs it seems like there happened a bunch of 'Permission denied' issues when creating/updating config files. I completely reinstalled the project and will try again in a few weeks since I'm happy for now that spotify albums actually play.

Thanks!

s-martin, 2020-05-22:
Should be fixed with upcoming 2.0 release

closed 2020-05-22 by s-martin