MiczFlor/RPi-Jukebox-RFID #837: 🚀 | allow selection of hardware during install

ZyanKLee, 2020-03-07:
## Feature Description

### What functionality would you like to see in your phoniebox?

As a parent I would like to select during installation which hardware (RFID reader, usb soundcard, buttons, rotary encoders, etc) I got and the installer should take care of only installing the necessary dependencies.

### How do you envision the feature to work from a users perspective?

I would like to select from some lists, which hardware I have. Dependencies should be separated into a `requirements.txt` file for each hardware. After the selection has been done and confirmed the installer would run the needed pip-install commands.

s-martin, 2020-03-07:
I have a local branch where I started some installer using npyscreen.

I try to push something tonight.

ZyanKLee, 2020-03-07:
no need to rush this
I would envision it at v3 earliest.

s-martin, 2020-10-17:
This has been implemented at least for readers in 2.1, so I close this now.

closed 2020-10-17 by s-martin