MiczFlor/RPi-Jukebox-RFID #842: 🐛 Installing RFID fails in Python script

tommy-rk, 2020-03-08:
## Bug

### What I did

Re-Installed RPi. After the installer finishes it asks for an RFID device. I have answered with Y. 
The following error appears:

#
# INSTALLATION FINISHED
#
#####################################################

If you are using an USB RFID reader, connect it to your RPi.
(In case your RFID reader required soldering, consult the manual.)
Have you connected your USB Reader? [Y/n] 
Traceback (most recent call last):
  File "RegisterDevice.py", line 4, in <module>
    from Reader import get_devices
  File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 23, in <module>
    from evdev import InputDevice, categorize, ecodes, list_devices
ModuleNotFoundError: No module named 'evdev'
chown: cannot access '/home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt': No such file or directory
chmod: cannot access '/home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt': No such file or directory

DONE. Let the sounds begin.
Find more information and documentation on the github account:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/
Reboot is needed to activate all settings
Would you like to reboot now? [Y/n] 

### Seen already here:
I have seen this error documented already here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/231
 
## Software

### Base image and version

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian

### Branch / Release

Refresh index: 100% (309/309), done.
On branch master
Your branch is up to date with 'origin/master'.

### Installscript
cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh

Refresh index: 100% (309/309), done.
On branch master
Your branch is up to date with 'origin/master'.

## Hardware
### RaspberryPi version
Raspberry Pi 3 Model B Plus Rev 1.3

s-martin, 2020-03-08:
There’s an issue in the installer script.

Please try 

```
cd /home/pi/RPi-Jukebox-RFID/
sudo python3 -m pip install -r requirements.txt
```

and afterwards

```
cd /home/pi/RPi-Jukebox-RFID/scripts/
python3 RegisterDevice.py
```

See also https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/CONFIGURE-stretch#identify-and-configure-your-rfid-reader and https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/CONFIGURE-stretch#register-your-usb-device-for-the-phoniebox

tommy-rk, 2020-03-08:
This did the trick! Thanks a lot!
This should make it into the Wiki or get corrected. Right now it is buried in the issues section.

s-martin, 2020-03-08:
> This did the trick! Thanks a lot!
> This should make it into the Wiki or get corrected. Right now it is buried in the issues section.

Good to hear. It’s a bug introduced in the installer script. Should be fixed with PR #843

PR #843 (2020-03-08): Hotfix: installation of py packages now for python 3
Should fix #842 on **master** 

Untested yet.
closed 2020-03-08 by s-martin