MiczFlor/RPi-Jukebox-RFID #844: 🚀 | Create log file during installation

s-martin, 2020-03-08:
## Feature Description

### What functionality would you like to see in your phoniebox?

As a maintainer I need a log file of the installation, so issues of users can be resolved more efficient, because I can see what didn’t work during installation.

### How do you envision the feature to work from a users perspective?

If a user experience an issue during installation or afterwards, the log file of the installation can be added to a bug report, so maintainers could check for package installation errors etc. and resolve issues faster.

s-martin, 2020-03-08:
Bookmark for first idea: https://stackoverflow.com/questions/18812098/piping-an-interactive-session-to-a-file

Groovylein, 2020-03-10:
I totally agree that we need a log file. But does the user really need all information on the screen?

I am thinking about using this kind of snippet:
https://serverfault.com/questions/103501/how-can-i-fully-log-all-bash-scripts-actions

Or using a logger would be even nicer:
https://goodmami.org/2011/07/04/Simple-logging-in-BASH-scripts.html

What do you think?

ZyanKLee, 2020-03-10:
I Like hat second snippet. Writing the whole output to a log file should still be done.

Groovylein, 2020-03-10:
> I Like that second snippet. Writing the whole output to a log file should still be done. Perhaps that functionality can be added to the logging implementation?

Oh yeah, you are right. Therefore I found the following functions that I really like!
http://www.ludovicocaldara.net/dba/bash-tips-5-output-logfile/

Combining both solutions would be best.

s-martin, 2020-03-10:
> Oh yeah, you are right. Therefore I found the following functions that I really like!
> http://www.ludovicocaldara.net/dba/bash-tips-5-output-logfile/

I like that, too. Especially that you can't forget logging.

Groovylein, 2020-03-25:
So the log file is now implemented. Should we enhance the installer script to only user-relevant output?
I'm thinking about an internal logger here.

s-martin, 2020-04-10:
> Should we enhance the installer script to only user-relevant output?
I'm thinking about an internal logger here.

I agree this could be a good improvement.

@fredg02 added some more improvements to the installer in PR #888. I think we should discuss further improvements there or better in a new issue.

I would close this issue for now, as the installer log has been implemented by your PR #856.

closed 2020-04-10 by s-martin