MiczFlor/RPi-Jukebox-RFID #845: 🐛 | Web Interface: Playlist Content incomplete / does not show

Kullid, 2020-03-08:
I've got 6 folders filled with mp3 files on the phoniebox, but one of them does not display correct in the web interface. It contains 95 files. When trying to view the content of the folder by clicking the link below in the "Available Media" overview then nothing happens.
[image]
Firefox reports this error:
[image]

When the folder plays and I click "Show Playlist" only 29 files are shown and the last one is displayed without a duration like below.
[image]

When playing file 29 or files with higher number, the current total duration exceeds the calculated total duration of the folder: [image]

When deleting file 29 from the folder, file 30 becomes number 29, duration is not displayed, etc. So it does not seem to be an issue with the file.

Despite of the incomplete display in the web interface the entire folder and each file plays correctly.

How can this be fixed?

Here my configuration:

pi@Phoniebox:~/RPi-Jukebox-RFID/scripts/helperscripts $ sh Analytics_AfterInstallScript.sh
************************************
*** PHONIEBOX INFO
*** version: 1.1.9
*** edition: classic
*** Audio_iFace_Name: Speaker
*** Audio_Folders_Path: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** Audio_Volume_Change_Step: 10
*** Max_Volume_Limit: 100
*** Idle_Time_Before_Shutdown: 10
*** Second_Swipe: NOAUDIOPLAY
*** Playlists_Folders_Path: /home/pi/RPi-Jukebox-RFID/playlists
*** ShowCover: ON
************************************
*** CONF FILES DEFAULT

*** /etc/samba/smb.conf
-rw-r--r-- 1 root root 9,5K Jul  4  2019 /etc/samba/smb.conf
   path=/home/pi/RPi-Jukebox-RFID/shared
   path=/home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** /etc/lighttpd/lighttpd.conf
-rw-r--r-- 1 root root 1,1K Jul  4  2019 /etc/lighttpd/lighttpd.conf
*** /etc/lighttpd/conf-available/15-fastcgi-php.conf
-rw-r--r-- 1 root root 398 Jul  4  2019 /etc/lighttpd/conf-available/15-fastcgi-php.conf
*** /etc/php/7.0/fpm/php.ini
-rw-r--r-- 1 root root 70K Jul  4  2019 /etc/php/7.0/fpm/php.ini
*** /etc/sudoers
-r--r----- 1 root root 784 Jul  4  2019 /etc/sudoers
*** /etc/systemd/system/phoniebox*
-rw-r--r-- 1 root root 296 Jul  4  2019 /etc/systemd/system/phoniebox-rfid-reader.service
-rw-r--r-- 1 root root 340 Jul  4  2019 /etc/systemd/system/phoniebox-startup-sound.service
-rw-r--r-- 1 root root 289 Jul  4  2019 /etc/systemd/system/phoniebox-gpio-buttons.service
-rw-r--r-- 1 root root 291 Jul  4  2019 /etc/systemd/system/phoniebox-idle-watchdog.service
*** /etc/mpd.conf
-rw-r----- 1 mpd audio 14K Jul  4  2019 /etc/mpd.conf
music_directory         "/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
# music_directory are changed.
# symbolic links outside of the configured music_directory.
# symbolic links inside of the configured music_directory.
        mixer_control   "Speaker"               # optional
#       mixer_control   "Speaker"               # optional
*** /etc/dhcpcd.conf
-rw-rw-r-- 1 root netdev 178 Nov  7 20:06 /etc/dhcpcd.conf
static ip_address=192.168.205.69/24
static routers=192.168.205.1
static domain_name_servers=8.8.8.8 192.168.205.1
*** /etc/wpa_supplicant/wpa_supplicant.conf
-rw-r--r-- 1 root root 264 Nov  7 20:41 /etc/wpa_supplicant/wpa_supplicant.conf
country=DE
************************************
*** +Spotify Edition

*** /etc/locale.gen
-rw-r--r-- 1 root root 8,9K Mär 13  2018 /etc/locale.gen
*** /etc/locale.gen
-rw-r--r-- 1 root root 8,9K Mär 13  2018 /etc/locale.gen
*** /etc/mopidy/mopidy.conf
ls: Zugriff auf '/etc/mopidy/mopidy.conf' nicht möglich: Datei oder Verzeichnis nicht gefunden
cat: /etc/mopidy/mopidy.conf: Datei oder Verzeichnis nicht gefunden
*** ~/.config/mopidy/mopidy.conf
ls: Zugriff auf '/home/pi/.config/mopidy/mopidy.conf' nicht möglich: Datei oder Verzeichnis nicht gefunden
cat: /home/pi/.config/mopidy/mopidy.conf: Datei oder Verzeichnis nicht gefunden

s-martin, 2020-04-10:
Hi, sorry for the late response.

Can you reproduce the bug with a folder/files which has a much shorter path and/or contains no special characters? Maybe you could start by renaming the folder to `test1` or similar.

I can see Firefox reports expression errors, which are maybe related to the scores, underscores or whitespaces...

Kullid, 2020-04-10:
Hi s-martin,

it seems to be two problems in one. One of them is gone now after I read your answer.
It was enough to remove "(Buch)" from the folders name. The Brackets seem to have caused the javascript error in firefox. Now I can view all files corretly in the "Available Media" Overview.

However "Show Playlist" still does not work, even if I rename the folder "Test1" and all files "01.mp3" to "95.mp3".
I did a little more trial and error, deleted first 20 files, files listed at the bottom of the playlist, etc. The behaviour is strange. After deleting files 1 to 20 and restarting the phoniebox, 40 files are shown. Moving the first 20 files back to the folder does not make them reappear after a restart. The more I played around, the more files where missing in between. Some reappear some do not. I do not see a rule here. What remains is an approximate total playtime of around one hour +/- 3 minutes approximately and the last file which is shown is always displayed without duration.

Thanks for investigating!

Kullid, 2020-04-11:
I tried something else...
I split the 95 files into 5 folders with 19 files and between 30 and 41 minutes total playtime each.
All of them are displayed correctly.
Then moved 10 files from folder 5 to folder 1 --> 1:00:11 total playtime, each file displayed correctly.
Moved one more file from folder 5 to folder 1 --> 1:04:19 total playtime, each file displayed correctly.
Moved 3 more file from folder 5 to folder 1 --> 1:08:10 total playtime, last file in folder displayed as "95.mp3" instead of Title, Artist and Duration.
[image]
Deleted second last file (the one above 95.mp3) file from folder 1 --> 01:09:29 total playtime, each file displayed correctly.
[image]
Moved another file from folder 5 to folder 1 (96.mp3) --> 01:09:29 total playtime, last file in folder displayed as "96.mp3" instead of Title, Artist and Duration.
[image]

So, again, the files seem to be without errors. But as soon as there are "too many" files in a folder, problems occur. "too many" seem to vary. In my case it's between 29 and 34 files, depending on the set of files which I choose to be contained in one folder. Sometimes the last file which is shown in the playlist is displayed with title and artist and sometimes it's not. The duration is always missing.

s-martin, 2020-04-11:
@Kullid, thanks for investigating this.

We should check first, if this issue is caused by the web UI or something deeper.

s-martin, 2020-05-03:
@MiczFlor, can you have a look at this? Could be UI related.

Kullid, 2021-07-14:
I can confirm that this issue persists in version 2.2:
[image]

```************************************
*** PHONIEBOX INFO
*** version: 2.2 - 305325d - master
*** edition: classic
*** Audio_iFace_Name: Headphone
*** Audio_Folders_Path: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** Audio_Volume_Change_Step: 10
*** Max_Volume_Limit: 65
*** Idle_Time_Before_Shutdown: 10
*** Second_Swipe: NOAUDIOPLAY
*** Playlists_Folders_Path: /home/pi/RPi-Jukebox-RFID/playlists
*** ShowCover: ON
************************************
*** CONF FILES DEFAULT
 
*** /etc/samba/smb.conf
-rw-r--r-- 1 root root 8,2K Jul  9 21:41 /etc/samba/smb.conf
   path=/home/pi/RPi-Jukebox-RFID/shared
   path=/home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** /etc/lighttpd/lighttpd.conf
-rw-r--r-- 1 root root 2,1K Jul  9 21:41 /etc/lighttpd/lighttpd.conf
*** /etc/lighttpd/conf-available/15-fastcgi-php.conf
-rw-r--r-- 1 root root 575 Jul  9 21:41 /etc/lighttpd/conf-available/15-fastcgi-php.conf
*** /etc/php/7.0/fpm/php.ini
ls: Zugriff auf '/etc/php/7.0/fpm/php.ini' nicht möglich: Datei oder Verzeichnis nicht gefunden
*** /etc/sudoers
-r--r----- 1 root root 784 Jul  9 21:41 /etc/sudoers
*** /etc/systemd/system/phoniebox*
-rw-r--r-- 1 root root 295 Jul  9 21:41 /etc/systemd/system/phoniebox-rfid-reader.service
ls: Zugriff auf '/etc/systemd/system/phoniebox-startup-sound.service' nicht möglich: Datei oder Verzeichnis nicht gefunden
ls: Zugriff auf '/etc/systemd/system/phoniebox-gpio-buttons.service' nicht möglich: Datei oder Verzeichnis nicht gefunden
-rw-r--r-- 1 root root 291 Jul  9 21:41 /etc/systemd/system/phoniebox-idle-watchdog.service
*** /etc/mpd.conf
-rw-r----- 1 mpd audio 14K Jul  9 21:42 /etc/mpd.conf
music_directory		"/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
# music_directory are changed.
# symbolic links outside of the configured music_directory.
# symbolic links inside of the configured music_directory.
	mixer_control	"Headphone"		# optional
#	mixer_control	"PCM"		# optional
*** /etc/dhcpcd.conf
-rw-rw-r-- 1 root netdev 130 Jul  9 21:42 /etc/dhcpcd.conf
static ip_address=192.168.205.85/24
static routers=192.168.205.1
static domain_name_servers=8.8.8.8 192.168.205.1
*** /etc/wpa_supplicant/wpa_supplicant.conf
-rw-rw-r-- 1 root netdev 163 Jul  9 21:42 /etc/wpa_supplicant/wpa_supplicant.conf
country=DE
************************************
*** +Spotify Edition
 
*** /etc/locale.gen
-rw-r--r-- 1 root root 9,2K Mai  7 17:13 /etc/locale.gen
*** /etc/locale.gen
-rw-r--r-- 1 root root 9,2K Mai  7 17:13 /etc/locale.gen
*** /etc/mopidy/mopidy.conf
ls: Zugriff auf '/etc/mopidy/mopidy.conf' nicht möglich: Datei oder Verzeichnis nicht gefunden
cat: /etc/mopidy/mopidy.conf: Datei oder Verzeichnis nicht gefunden
*** ~/.config/mopidy/mopidy.conf
ls: Zugriff auf '/home/pi/.config/mopidy/mopidy.conf' nicht möglich: Datei oder Verzeichnis nicht gefunden
cat: /home/pi/.config/mopidy/mopidy.conf: Datei oder Verzeichnis nicht gefunden```

philister, 2021-09-07:
The bracket-thing is probably solved with PR #1231 

The other thing can be traced back to php's exec at https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/f71d1c7da1e8acf1f9ca304445692d37060cfdb2/htdocs/api/common.php#L33 called by https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/70a4a8e216f7f55f317af67bd85ab1b41439657c/htdocs/api/playlist.php#L30
It cuts off the result of `echo 'playlistinfo\nclose' | nc -w 1 localhost 6600";`
The length of the remaining content probably depends on the metadata of the files and therefore appears randomly.
I have no idea how to solve :(