MiczFlor/RPi-Jukebox-RFID #851: 🐛 When upgrading the system mopidy is been kept back - why?

tommy-rk, 2020-03-09:
This is more of a question, than a bug:
When I upgrade the system with:
`sudo apt-get update && sudo apt-get upgrade`
The mopidy package is been kept back. What is the reason for this?

Output of the update:
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:3 https://apt.mopidy.com buster InRelease [11.9 kB]                 
Fetched 26.8 kB in 1s (22.2 kB/s)                 
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
**The following packages have been kept back:
  mopidy
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.**

ZyanKLee, 2020-03-09:
perhaps there is a problem within the version pinning.
can you please run an `apt-cache show mopidy` ?

tommy-rk, 2020-03-10:
pi@raspberrypi:~ $ apt-cache show mopidy
Package: mopidy
Version: 3.0.1-2
Architecture: all
Maintainer: Stein Magnus Jodal <jodal@debian.org>
Installed-Size: 486
Depends: adduser, debconf, gir1.2-gst-plugins-base-1.0, gir1.2-gstreamer-1.0, gstreamer1.0-plugins-good, gstreamer1.0-plugins-ugly, lsb-base, python3-gst-1.0, python3-tornado (>= 4.4), debconf (>= 0.5) | debconf-2.0, python3-pkg-resources, python3-pykka, python3-requests, python3:any
Recommends: gstreamer1.0-alsa, gstreamer1.0-pulseaudio, gstreamer1.0-tools
Suggests: mopidy-doc
Homepage: https://mopidy.com/
Priority: optional
Section: sound
Filename: pool/main/m/mopidy/mopidy_3.0.1-2_all.deb
Size: 102736
SHA256: 41b194cd791dc08e1b7c5ccb307472c3ab102667d57eedd543118192e3c9253f
SHA1: d24b7672fa4a507ad2f51f3e8b31ed204142f1e1
MD5sum: 474e622d813106f52dec81f93f8fca87
Description: extensible music server
 Mopidy plays music from local disk, Spotify, SoundCloud, Google Play Music,
 and more. You can edit the playlist from any phone, tablet, or computer using
 a variety of MPD and web clients.
 .
 Vanilla Mopidy only plays music from files and radio streams. Through
 extensions, Mopidy can play music from cloud services like Spotify,
 SoundCloud, and Google Play Music.
Description-md5: 72560d4cd04b5a15f8be0cfe54877dc3

Package: mopidy
Version: 2.2.2-1
Installed-Size: 858
Maintainer: Stein Magnus Jodal <jodal@debian.org>
Architecture: all
Depends: adduser, debconf, gir1.2-gst-plugins-base-1.0, gir1.2-gstreamer-1.0, gstreamer1.0-plugins-good, gstreamer1.0-plugins-ugly, lsb-base, python-gst-1.0, python-tornado (>= 2.3), debconf (>= 0.5) | debconf-2.0, python-pkg-resources, python-pykka, python-requests, python:any (<< 2.8), python:any (>= 2.7~)
Recommends: gstreamer1.0-alsa, gstreamer1.0-pulseaudio, gstreamer1.0-tools
Suggests: mopidy-doc
Size: 207300
SHA256: 2e0291404278080e0f9370b38230e3537d2b5515a9852d591a5c577b62dd5c13
SHA1: 35d5dea3a83460a532103fcc36a9511da04eec67
MD5sum: b3e9b0f539ffa5079dbb74a711e460fb
Description: music server with support for MPD/HTTP clients
 Mopidy is an extensible music server which can play music from local
 files, radio streams, and cloud services such as Spotify and SoundCloud.
 .
 Searches combine results from all music sources, and tracks can be mixed from
 all sources in the play queue. Spotify or SoundCloud playlists are
 also available for use.
Description-md5: c3a57c8f4f235b22a38d4f2cc85eda44
Homepage: https://www.mopidy.com/
Tag: implemented-in::python, interface::commandline, interface::daemon,
 interface::web, network::server, protocol::TODO, protocol::http,
 protocol::zeroconf, role::program, sound::player, use::entertaining,
 use::playing, works-with-format::mp3, works-with-format::oggvorbis,
 works-with-format::wav, works-with::audio
Section: sound
Priority: optional
Filename: pool/main/m/mopidy/mopidy_2.2.2-1_all.deb

tommy-rk, 2020-03-10:
One solution might me to run a apt-get dist-upgrate like mentioned here:
https://askubuntu.com/questions/601/the-following-packages-have-been-kept-back-why-and-how-do-i-solve-it

Should I do it? Is anyone seeing issues in doing so?

```
pi@raspberrypi:~ $ sudo apt-get update && sudo apt-get dist-upgrade
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Get:2 https://apt.mopidy.com buster InRelease [11.9 kB]                                   
Hit:3 http://archive.raspberrypi.org/debian buster InRelease                              
Fetched 11.9 kB in 1s (8,455 B/s)         
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  python-backports-abc python-concurrent.futures python-gst-1.0 python-pykka python-singledispatch python-tornado
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  fonts-font-awesome fonts-lato fonts-roboto-slab libjs-modernizr python3-gst-1.0 python3-pykka python3-tornado sphinx-rtd-theme-common
The following packages will be upgraded:
  mopidy
1 upgraded, 8 newly installed, 0 to remove and 0 not upgraded.
Need to get 5,102 kB of archives.
After this operation, 18.1 MB of additional disk space will be used.
Do you want to continue? [Y/n] 

```

tommy-rk, 2020-03-10:
I have just tried it. Don't do it. Spotify is not working anymore after doing this.
Trying to fix it by following https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/852 also doesn't help.
Let's see, if re-running the installer helps?

s-martin, 2020-03-10:
Please post some logging data, so we can understand better what doesn't work.

tommy-rk, 2020-03-10:
Re-running the installer does help :)
All packages are up2date.

```
pi@raspberrypi:~ $ sudo apt-get update && sudo apt-get upgrade
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease            
Get:3 https://apt.mopidy.com buster InRelease [11.9 kB]                 
Fetched 11.9 kB in 2s (7,127 B/s)                 
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

```

ZyanKLee, 2020-03-11:
@tommy-rk can this issue be closed?

tommy-rk, 2020-03-11:
Yes

closed 2020-03-11 by tommy-rk