MiczFlor/RPi-Jukebox-RFID #852: Sticky: Spotify (Iris) does not work, e.g. error „404“

s-martin, 2020-03-09:
If you have installed Phoniebox using the one-line installer, you may experience problems when trying to use Spotify (Iris).

This is because currently the used third-party library Mopidy needs to be upgraded to version 3. This has been fixed for Phoniebox [2.0](https://github.com/MiczFlor/RPi-Jukebox-RFID/milestone/1), but this version has not been released yet.

To fix this in the meantime, please execute the following:

```
echo -e "Package: mopidy\nPin: version 3.*\nPin-Priority: 1001" | sudo tee /etc/apt/preferences.d/mopidy

wget -q -O - https://apt.mopidy.com/mopidy.gpg | sudo apt-key add -
sudo wget -q -O /etc/apt/sources.list.d/mopidy.list https://apt.mopidy.com/buster.list

sudo apt-get update
sudo apt-get upgrade
sudo apt-get --yes --allow-downgrades --allow-remove-essential --allow-change-held-packages install mopidy mopidy-mpd mopidy-local mopidy-spotify

sudo apt-get --yes --allow-downgrades --allow-remove-essential --allow-change-held-packages install libspotify12 python3-cffi python3-ply python3-pycparser python3-spotify

sudo python3 -m pip install Mopidy-Iris
```

For details, see #803 and #804.

MiczFlor, 2020-04-07:
Hi @s-martin 
does it make sense to add this to the install script? It might be a necessary workaround for some time. At least it's outside our control. 
If so, I will add it, but then: should I add it to master AND develop?

s-martin, 2020-04-07:
This is already integrated in the develop buster script.

When I added this issue I anticipated we would release 2.0 (i.e. develop) soon.

I think we could check the open PRs for 2.0 and then release it. From my point of view develop should be rather stable.

s-martin, 2020-05-24:
2.0 release should fix this on master.

closed 2020-05-24 by s-martin