MiczFlor/RPi-Jukebox-RFID #854: 🐛 | When creating Spotify Card IDs the title.txt and cover,jpg are not created correctly

Hamsert, 2020-03-10:
## Bug

### What I did

I created a new card ID with a Spotify Album linked to it.

### What happened

The card ID was created successfully but cover.jpg and title.txt are no longer created correctly. They are both empty. Also on the "home" page of Phoniebox the card ID just shows the Spotify icon and no name next to it.

### I expected this to happen

cover.jpg and title.txt should be set correctly and on home menu there should be the title next to the Spotify icon

### Further information that might help

cover.jpg and title.txt are also not set correctly after playing the card.

## Software

### Base image and version

newest buster release with develop phoniebox installed today.

### Branch / Release

develop

### Installscript

one line installer
cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/develop/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; GIT_BRANCH=develop bash ./buster-install-default.s

## Hardware

### RaspberryPi version

3B+

### RFID Reader

16c0:27db HXGCoLtd Keyboard

### Soundcard

no external

### Other notable hardware

bastiitsab, 2020-03-10:
I just can confirm that I have the same problem on master too

s-martin, 2020-03-10:
Possible duplicate of #799

Hamsert, 2020-03-10:
Yes, it's a duplicate.

closed 2020-03-10 by Hamsert