MiczFlor/RPi-Jukebox-RFID #855: 🚀 | Disable GPIO Buttons by default to prevent collision with GPIO-Soundcards

bastiitsab, 2020-03-11:
## Feature Description

### What functionality would you like to see in your phoniebox?

change the status of the GPIO-Buttons by default to disabled

### How do you envision the feature to work from a users perspective?
.
### Further information that might help

If you use any SoundCards which are using GPIO Pins like hifiberrys they are in conflict with the default GPIO Buttons. You have to change the PINs to use both. 
But in my opinion it should be ensured that the soundcards are working. The GPIO-Buttons can be activated or changed afterwards. 

I faced this problem now for the second time. Had all config files correct, but no sound was played. Then I remembered the gpio-buttons script. After changing the PINs the soundcard is working directly.

s-martin, 2020-03-11:
In #790 there are ideas and discussions how this issue can be addressed. Currently one idea is to have different configuration presets for hardware combinations.

veloxidSchweiz, 2020-03-15:
Hi @bastiitsab,
Thank you for the idea. I do understand your idea. It has taken quite some time to understand why the hifiberry dac/miniamp was not working when I installed the phoniebox the first time. 
I think the solution with using configuration files would be  a better way to address this issue. I am testing this feature at the moment with my box and it is working pretty good (at least for me), using buttons, rotary-encoders, rfid-reader and amplifier via GPIO. Next step are status LEDs which i want to include.

Maybe @MiczFlor can comment on the standard gpio configuration, It is not really the best when you want to be flexibel.

s-martin, 2020-10-14:
Closing this, since 2.1 gpio_control is used.

closed 2020-10-14 by s-martin