MiczFlor/RPi-Jukebox-RFID #860: 🚀 | Do not shutdown while apt updates are running

andreasbrett, 2020-03-14:
Unfortunately I had to find this the hard way... Phoniebox shut itself down during an apt update that also included a kernel update. Result: bricked kernel image that took some time fixing :-/

The shutdown timer really needs to check if apt is currently running. I think there's a lock file during updates for apt, dpkg or both. If they exist, a timer-based shutdown must be prevented.

andreasbrett, 2020-03-14:
I think we should prevent this right in `playout_controls.sh` so even button presses are not causing a shutdown without checking.

https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/5b1cac791bb7eb9d75213a4e38e24fe6b8d93992/scripts/playout_controls.sh#L125

https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/5b1cac791bb7eb9d75213a4e38e24fe6b8d93992/scripts/playout_controls.sh#L133

s-martin, 2020-03-21:
Could you make a PR?

andreasbrett, 2020-03-31:
These might be the lock files that are present during update installation. I can't tell for sure as I don't know how to simulate an update. Will have to wait for an update to come. 

/var/lib/apt/lists/lock
/var/lib/dpkg/lock
/var/cache/apt/archives/lock

andreasbrett, 2020-04-05:
> Could you make a PR?

PR is online now. https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/885

s-martin, 2020-04-10:
Closing. PR #885 is merged.

closed 2020-04-10 by s-martin