MiczFlor/RPi-Jukebox-RFID #861: 🐛 | RC522 wired RFID reader is not working

domu83, 2020-03-17:
## Bug

### What I did
After several spotify issues I have decided to run a new installation using 2020-02-13-raspbian-buster-lite. I run the Default Installation script.
reader is wired as shown in the pic in the wiki (not using the table below with the alternative wiring)

### What happened
After the installation I run through the config part of the wiki and have also using the special setup for wired RFID readers. I have run through the autstart advices for phoniebox.
However the RFID reader service is not starting up at all and throws an error.
When I try to activate the RFID reader through the webpage it gets green, but after a site refresh, turns off/red again.
BTW: also the startup and shutdown sound do not appear.

### I expected this to happen
after the service activation and reboot, startup sound should appear and RFID reader should work.

### Further information that might help

● phoniebox-rfid-reader.service - Phoniebox RFID-Reader Service
   Loaded: loaded (/etc/systemd/system/phoniebox-rfid-reader.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2020-03-17 22:03:14 GMT; 2min 2s ago
  Process: 5839 ExecStart=/home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py (code=exited, status=1/FAILURE)
 Main PID: 5839 (code=exited, status=1/FAILURE)

Mar 17 22:03:14 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Service RestartSec=100ms expired, scheduling restart.
Mar 17 22:03:14 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Scheduled restart job, restart counter is at 5.
Mar 17 22:03:14 raspberrypi systemd[1]: Stopped Phoniebox RFID-Reader Service.
Mar 17 22:03:14 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Start request repeated too quickly.
Mar 17 22:03:14 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Failed with result 'exit-code'.
Mar 17 22:03:14 raspberrypi systemd[1]: Failed to start Phoniebox RFID-Reader Service.
~

syslog liegt hier: https://1drv.ms/t/s!Ate9mXuqjEhDjgsp1fXsjKFxVvc3?e=3oQd5I

## Software

### Base image and version

2020-02-13-raspbian-buster-lite

### Branch / Release

master branch

### Installscript

installscripts/buster-install-default.sh

## Hardware

### RaspberryPi version
3B+

### RFID Reader
RC522
### Soundcard
onboard

### Other notable hardware
GPIO buttons, but not configured yet. gpio-buttons.py is untouched.
Also onoff shim and script installed

domu83, 2020-03-17:
what I see in the log is at least this topic:

> Mar 17 22:03:14 raspberrypi daemon_rfid_reader.py[5839]:   File "/home/pi/RPi-Jukebox-RFID/scripts/pirc522/rfid.py", line 68, in __init__
> Mar 17 22:03:14 raspberrypi daemon_rfid_reader.py[5839]:     self.spi.open(bus, device)
> Mar 17 22:03:14 raspberrypi daemon_rfid_reader.py[5839]: FileNotFoundError: [Errno 2] No such file or directory

so the rfid.py file is present in the "/home/pi/RPi-Jukebox-RFID/scripts/pirc522/" directory.

lovethisshit, 2020-03-18:
I think that the startup sound is not actually part of the install script. It is simply a crontab entry. Unfortunately, the wiki is a bit outdated: https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#daemon_rfid_readerpy-only-works-via-ssh-not-by-rfid-cards

The startup sound file is normally located at /home/pi/RPi-Jukebox-RFID/shared/startupsound.mp3

As for the rfid service: did you try to start the script manually?
python2 /home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py

Running scripts by command line usually was quite useful for me when trying to debug the whole app

domu83, 2020-03-19:
thanks @lovethisshit for looking into this. Yeah, I thought so as well... maybe good to open a enhancement request to get the doc updated.
For the startup sound, interesting enough, after the first reboot installing develop branch, startup sound appears. So for develop branch it's included in the install script. 
I've run through the documentation of configure the box incl the setup and autostart of the 4 scripts (RFID, GPIO, Startup & Watchdog). At least in the strech installation I did, this was the way to get everything done. 
The MP3 files themselves are available in the shared directory, so not sure whats the issue on this one.

back to the main and most critical topic for me, the RFID reader... by running the script manually I get the following output:

> pi@raspberrypi:~ $ python2 /home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.                                        py
> Traceback (most recent call last):
>   File "/home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py", line 5, in <mo                                        dule>
>     from Reader import Reader
>   File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 10, in <module>
>     import serial
> ImportError: No module named serial

If I would interprete this one, I understand that my device is not registered. However I run through the commands in the RC522 section and choose my device there. Anything I'm missing out?!

Just as of curiousity... why have you suggested python2? 
I've posted the python3 output of the manual run above and from what I've see is that the daemon is ported to python 3 already.

thanks a lot for looking into this!

s-martin, 2020-03-19:
Try the command with python3. Python 2 is not supported anymore (except you have an existing installation on stretch).

domu83, 2020-03-19:
yep, I have run the python3 and added the output to the initial case, but here again:
```
pi@raspberrypi:~ $ python3 /home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py
Traceback (most recent call last):
  File "/home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py", line 7, in <module>
    reader = Reader()
  File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 133, in __init__
    self.reader = Mfrc522Reader()
  File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 53, in __init__
    self.device = pirc522.RFID()
  File "/home/pi/RPi-Jukebox-RFID/scripts/pirc522/rfid.py", line 68, in __init__
    self.spi.open(bus, device)
FileNotFoundError: [Errno 2] No such file or directory
pi@raspberrypi:~ $
```

lovethisshit, 2020-03-19:
I'm very new to this project, so I can only guess:
* Did you check the presents of all files mentioned in the error? There is also a Pull Request which will remove those files locally soon: https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/840
* Did you actually enable SPI? Execute "sudo raspi-config" and select "Interface Options" > "SPI" > "Enable"

Side Note: I still need to use python2 because master branch wouldn't work for me otherwise and I hesitate to apt upgrade , but that's a different topic.

domu83, 2020-03-19:
@lovethisshit bummer... you are my hero... after now 1.5Y working with this project, I've missed this step. enableing the SPI did it and brought up my RFID reader! What a pitty. Thanks a lot for this. Now I have to figure out the startup sound issue

closed 2020-03-19 by domu83