MiczFlor/RPi-Jukebox-RFID #862: 🚀 | Equalizer for Phoniebox

thedipling, 2020-03-18:
## A Systemwide Equalizer based on plugequal for mpd

### It would be a good thing to be able to do some equalizer Settings for the diversity of Loudspeakers to improve Soundquality

Based on
[https://wiki.debianforum.de/Erweiterte_ALSA-Funktionen](https://wiki.debianforum.de/Erweiterte_ALSA-Funktionen) 

I've changed the /etc/mpd.conf 

``# An example of an ALSA output:
``#
``audio_output {
 ``       type            "alsa"
 ``       name            "My ALSA Device"
``#       device          "hw:1,0"        # optional
 ``       device          "plug:plugequal"
``#       mixer_type      "hardware"      # optional
``#       mixer_device    "default"       # optional
``        mixer_device    "hw:1,0"
``        mixer_control   "PCM"           # optional
``#       mixer_index     "0"             # optional
``}

and added the /etc/asound.conf

``
``ctl.equal {
``  type equal;
``  controls "/home/pi/.alsaequal.bin"
``}
``pcm.plugequal {
``  type equal;
``  slave.pcm "plughw:1,0";
``  controls "/home/pi/.alsaequal.bin"
``}
``  pcm.equal {
``  type plug;
``  slave.pcm plugequal;
``}

### Further information that might help

Audio with my attempt is acutally working on Terminal, however I havent figured out to get the general loudness working.
However with my basic attempt the loudness is greyed out in the Website...

ZyanKLee, 2020-03-20:
Is there a way to have this feature for mopidy as well?

thedipling, 2020-03-21:
Hello everyone, I've got it working! See the following Forum for information.
[https://forum-raspberrypi.de/forum/thread/46718-equalizer-fuer-die-phoniebox/?postID=421069#post421305](url)

basically the mixertype needs to be set to software in the mpd.conf.

thedipling, 2020-03-27:
> 
> 
> Is there a way to have this feature for mopidy as well?

Hi there - sorry for the late reply...

I would say that this works system-wide

MiczFlor, 2020-04-07:
Hi @thedipling 
great contribution. For the time being, I added it to the wiki:

https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Equalizer_systemwide_based_on_plugequal_for_mpd

Not sure if this should go into core. What do you think?

closed 2020-04-07 by MiczFlor

ralle211, 2020-11-18:
Hi there 

is there a working equalizer config for Phoniebox? 
I am using Phoniebox+spotify with a hifiberry miniamp. So i guess i also have to adapt **mopidy.conf** in addition to **mpd.conf** and **asound.conf**. I tried such a lot but i didnt got it working...

does somebody has a hint or got it running?

THorst92, 2020-11-26:
Hi all,
I tried to activate the EQ but was not successful. This is what I've done:
1. Install the EQ-package
2. check which audio interface my headphone-jack is with `aplay -l`. It gave me device 0:0
3. create and edit the `/etc/asound.conf ` (also tried with ~/.asoundrc):
```
ctl.equal {
  type equal;
  controls "/home/pi/.alsaequal.bin"
}
pcm.plugequal {
  type equal;
  slave.pcm "plughw:0,0";
  controls "/home/pi/.alsaequal.bin"
}
  pcm.equal {
  type plug;
  slave.pcm plugequal;
}
```
4. edit `/etc/mpd.conf`:

```
audio_output {
        type            "alsa"
        name            "My ALSA Device"
#       device          "hw:0,0"        # optional
        device          "plug:plugequal"       
#       mixer_type      "hardware"      # optional
        mixer_type      "software"
#       mixer_device    "default"       # optional
        mixer_device    "hw:0,0" 
#       mixer_control   "headphone"     # optional
        mixer_control   "PCM" 
#       mixer_index     "0"             # optional
}
```
5. restart pi
6. tried to change EQ while listening to music via `alsamixer -D equal`

Unfortunately I hear no change while changing the EQ. Do anybody have an idea or hint why?

thedipling, 2020-11-30:
Hi there, sorry for the late reply...
have you tried the decription here?
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/862#issuecomment-602037593

My initial try was not successful. It was a long time ago... I actually dont know what to do here...
sorry..

ChisSoc, 2020-12-11:
I have added a few important tips and testing methods to the Wiki page. And my setup is working without any issues. But it has taken me a long, long time to figure it out.

However, my mpd.conf differs in one aspect, which is mixer_device. I leave that at "default" (by omitting it). This means ALSA will pick an appropriate mixer device. (See here https://www.musicpd.org/doc/html/plugins.html?highlight=mixer_device). I think it should be omitted, and the asound.conf file should contain a default entry. This would make it more generally usable. Can anybody confirm / disagree?

Except from Wiki
```audio_output {
        type            "alsa"
        name            "My ALSA Device"
        device          "plug:plugequal"
        mixer_type      "software"      # optional
        mixer_device    "hw:1,0"
        mixer_control   "PCM"           # optional
}```

My configuration
```audio_output {
        type            "alsa"
        name            "HifiBerry Speakers"
        device          "hifiberry"     # optional                                                                                                                                                                                                                                                                                                                               
        auto_resample   "no"
        mixer_control   "Master"
}

THorst92, 2020-12-11:
Thank you for the good description. I have "just" one problem. As soon as I add this line to the /etc/asound.conf I get no sound any more from the phoniebox:
```
# Make the equalizer the default playback device (optional)
pcm.!default {
  type plug
  slave.pcm plugequal;
}
```
The the EQ works with the soundtest `speaker-test -c 2 -d plugequal`.
When I remove the part and restart the phoniebox everything works, but not the EQ. Even not with the soundtest.
Is this maybe a problem with the Spotify-Edition, or did I miss something?
The rest of my configuration is like in this [comment](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/862#issuecomment-734433363).

s-martin, 2020-12-11:
Please continue the discussion in a new issue or the new discussions feature.

Continuing discussions in closed issues are easily overlooked.

ChisSoc, 2020-12-13:
@THorst92 : Please test my configuration described at https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1198 and give feedback. Let's continue discussion there or in the discussion feature. For the Spotify Edition, you must make settings accordingly in mopidy configuration file.

ralle211, 2020-12-14:
I checked the #1198 but I didn't found anything about mopidy.conf. 
What exactly do I have to change there?