MiczFlor/RPi-Jukebox-RFID #864: 🚀 | FEATURE SUMMARY

Piehti, 2020-03-19:
## A touch-interface to get rid of physical GPIO Buttons AND / OR to show more interactive content, while playing a sound / track.

### What functionality would you like to see in your phoniebox?

Implement support for X"-Displays and an option to get the hardware GPIO Buttons shown as softwarebuttons on the Touchscreen-Interface. 

### How do you envision the feature to work from a users perspective?

Spares a lot of work (propably) with GPIO-Buttons, easier to use for Kids and easier to customize for the Boxbuilder (html interface propably). Could make the rfid box more buttonless and more interactive.

### Further information that might help

In a very basic version, there could be a STATIC 3x2 or 3 x 3 tiles, with the basic features to enable. Later on, the user should be enabled, to put custom commands on the tiles.

Piehti, 2020-03-19:
THIS would be what I thought off on the long run. Customizable and highly extendable for many uses. Could also work nicly with that MQTT implementation for further development in homeautom. with jukebox.

[image]

If someone gets freaky, he could even let the titels be %-to the screen for a more fance look. Or add a wallpaper to the LCD or or or. ;)

Piehti, 2020-03-19:
Thought about it and will start to get this done by myself (propably).
Will try use something as the kweb3 or something to display a selfmade webinterface on the touch.
Will be back, when it's up to connect it to the action-handler. Can take some (a lot of) time since I'm kinda rusty in webapps and even shorter on sparetime. : /

s-martin, 2020-03-19:
Pull requests are always welcome!

Piehti, 2020-04-20:
Still working on it... But had to finnish two more boxes first... ;)
Right now digging in some code to get my Video-support working first.
The webinterface shouldbt be a huge deal with kweb or something.

What's a pull request? :) xD

Piehti, 2020-05-07:
Howdy,

finally got my video-stuff done. Took a while - since I modded my Box twice in the meantime and had to rethink about stuff and implementation. Will go for the onscreen-control now. Can anyone kick me in the direction, where the actual WEB-UI sits? I'd just gonne recycle it in a flatdesign UI.

MiczFlor, 2020-05-08:
The Web UI is inside the htdocs folder.

MiczFlor, 2020-05-08:
Do you have any code to share? I would be curious to see what you mean by "video stuff"

Piehti, 2020-05-09:
Hi!

Thanks.
Its more a shame as code. ;))
Also not done yet. Maybe tomorrow

I just started to reduce the ui and blowing up icons. The Magic is kweb
with autostart the new index.php optimized for LCD.

Im in a hurry right now, will feedback as soon as something is showable.

Best regars

Piehti, 2020-05-09:
[image]

[image]

These are my Favs so. Thinking about taking the %-Volume within, but I actually like the clean look so far.

Piehti, 2020-05-10:
I tasked a friend of mine and Pi-lover to pick up the topic. And he was fast. 
This is what it is so far: https://github.com/KartoffelToby/RPi-Jukebox-RFID

Not sure if it is of use for someone yet. One might just want to pick the index.php and use it as alternative next to the origin index.php. If working adequat, you might want to consider to serve it as index-lcd-alpha.php with the next release - if someone is eager to have / use it already (as I do). I have to admit, that I haven't come to test ist, but I guess it's working as good as the original index. Can tell after some testing later / tomorrow.

s-martin, 2020-05-22:
@Piehti, as this is now added to the repo could this issue be closed?

iwishitwassummer, 2020-05-24:
@Piehti Nice work!  which touch screen have you bought for this?

Piehti, 2020-05-24:
Howdy,

> @Piehti, as this is now added to the repo could this issue be closed?

Yes thanks a lot. I might forgot to mention, that one might consider to <ALT>+Mousehweel to scroll in to the desired size or one has to use the zoom-param on chromium, to gain a better view - regarding the size of ones very own display. 

That said, I planned to change the ui (re-add songtitle, custom icons and some switches - but am still totally out of any spare time due some certifications I have to take first.

> @Piehti Nice work! which touch screen have you bought for this?

You are welcome. Yet most of the thanking goes to "KartoffelToby", who cleaned my messy braindump.

Regarding the Display: Im not 100% sure. I had it for free (bulk) with one of my Pi's I bought in used condition. But I think it's a WaveShare 3.5 Inch HDMI Touch. I did at least used theire Wiki to install and configure it: https://www.waveshare.com/wiki/3.5inch_HDMI_LCD . I did actually use it, becasue it was free. If I'd be up to buy one, I'd consider getting somewhat sligthly bigger, higher resoluted and for bigger fingers. ;) 

As said: The small one was planned as mini console, and then considered as UI for Phonie / Kodi / Steamlink and other APPs with WEB Ui, so I can seperatly work on the main display. The main Display I actually plan to implement for now is one of the following: 
a) shorturl.at/dmOR2 / b) shorturl.at/cfEHI / or c) shorturl.at/dpqr1 .

Best regards
Pete

s-martin, 2020-05-24:
Integrated in 2.0 release, therefore we close this now.
If any problems arise, please feel free to open a new issues.

closed 2020-05-24 by s-martin