MiczFlor/RPi-Jukebox-RFID #865: 🚀 | FEATURE SUMMARY

Piehti, 2020-03-19:
## SEND COMMAND TO GOOGLE HOME AND THE WAY ROUND

Can anyone point me to a tutorial on how to get the Phoniebox to send commands to my google At least I think so. I even stuck in the possibility to test it. I'm totally new to open Homeautomation. Used broadlink so far.

### What functionality would you like to see in your phoniebox?
I'd like to let my daughter put a card on the pad to turn lights on and off.

### How do you envision the feature to work from a users perspective?

I don't know. I actually think this is already possible somehow with openHAB or HASS - but I didnt get to the point. What I found was not very easy to get or lacking on any kind of basics.

s-martin, 2020-03-19:
I don't know, if this works with Google home, but you can use MQTT.

https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Smart-Home-remote-control-with-MQTT

https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/components/smart-home-automation/MQTT-protocol

Piehti, 2020-03-19:
That's what I tryed ... that's what brought me here. Since the ones you stated lost me at: 

> All relevant files can be found in the folder:
> 
> components/smart-home-automation/MQTT-protocol/

... which I don't have. So wether I or the creator is missing a very essential thing...

*feels stupid*

I found out to finde the one script in the scripts folder already. Yet missing the sample-thingy...

Where can I get that "components" folder? It's not it the Oneclickscript, neighter mentioned to be pulled as git. Not sure...

s-martin, 2020-03-19:
These files are in the develop Branch, but for regular installation the master branch is used, which doesn't have these updates yet.

Just use the second link I gave and click on each file an "Raw" and then copy the content of the file to a file on your local machine.

Piehti, 2020-03-19:
Oright. Thanks.
Got the Files there (components) now.
Got (hopefully) paho-mqtt installed. (pip3 install paho-mqtt).

Now.... How to add that autoboot sample to where-ever to make what-ever autostart? ;)
And in advance, when I run the daemon by hand:

```
 File "daemon-mqtt-client.py", line 320, in <module>
  client.tls_set(ca_certs=mqttCA, certfile=mqttCert, keyfile=mqttKey)
  File "/home/pi/.local/lib/python3.7/site-packages/paho/mqtt/client.py", line 819, in tls_set
  context.load_cert_chain(certfile, keyfile)
  FileNotFoundError: [Errno 2] No such file or directory

```
Stuck again.
Sorry for bothering.

s-martin, 2020-03-19:
Please check the Readme in the MQTT folder, I guess it's a configuration issue.

Check also #779 and maybe @nukleuz80 can help.

Piehti, 2020-03-19:
Got it running and implemented. There where some issues, like the paths for the (not existing CAs) or the not-existing "Latest_RFID" file, over chmod-issues to... well. It works.

Lets see how to get this together with my google home.

erhapp, 2020-03-22:
I can confirm the problem with MQTT.
I installed a fresh system over the weekend and MQTT does not start on boot.
If I start the script manually everything goes fine.
I did however get an error message the first time I did a manual start. Apparently "paho-mqtt" was missing on my system. This was resolved by 'pip install paho-mqtt'. Is this something that can be included in the install script?

Is there any information I can provide regarding the problem with the script not starting on startup?

closed 2020-12-15 by s-martin