MiczFlor/RPi-Jukebox-RFID #866: 🚀 | Remove hardcoded folders and stuff for "pi"

Piehti, 2020-03-19:
### What functionality would you like to see in your phoniebox?

I'd like to get rid of the user PI. unfortunatly, your oneclick installer has the user PI hardcoded at several parts of the script. If you changed the user before installing, the installation will propably fail.

ZyanKLee, 2020-03-20:
this is on the back of my mind for some weeks now, too.
I don't see this as a task for v2, though.
@MiczFlor @s-martin if you think otherwise just put it into the appropriate milestone and project

s-martin, 2020-03-21:
I agree that’s something for 3.0

jo-me, 2020-06-16:
Will this allow the script to work on other OS than raspbian?

I'm thinking about using DietPi as base (smaller and less actiity on SD card) but it has a different user and probably other packages installed.

s-martin, 2020-06-16:
Phoniebox depends on packages like RPi.GPIO and others, so if these packages are available, it **could** work.

However, I'm not aware that anybody has run Phoniebox on another platform, so it definitely would be pioneer work.

PR #1543 (2022-02-27): Remove hardcoded homedir in install scritp
### Will resolve #866 
Replacing /home/pi with the home folder of the current user (`echo $HOME`)

### What I've tested and looked good so far:

- Created docker test with alternative user `hans`
- Installed with user `hugo` on my RPi3 (classic installation)
  - uploaded and listened to one mp3 file
- Installed with user `pi` on my RPi3 (classic installation)
  - uploaded and listened to one mp3 file

### Not Tested since I don't have any components:

- GPIO
- RFID
- Buttons
closed 2022-02-27 by MiczFlor