MiczFlor/RPi-Jukebox-RFID #867: 🚀 | (*Hint*) Small Android App

Piehti, 2020-03-19:
## Feature Description

### What functionality would you like to see in your phoniebox?

I'd like to have a small Android app to controll the Box - if this doesn't go to far.

***HINT***

_For now I use an old smartphone with "Progressive Web Apps"-Apps and wrap the IP of the Phoniebox into a fullscreen app. Maybe theres a android dev around with some time?_ 

### How do you envision the feature to work from a users perspective?
Would just be nais to have, to advance on to the "T***box" and make the remote controleven more convinient for wifey and older people. For a better out of the box experience.

### Further information that might help

Got none. This is what it looks with the wrapper:

[image]

ZyanKLee, 2020-03-20:
I like the idea. It should be a separate project, though.

Groovylein, 2020-05-15:
I thought about this too and like it! +1

iroQuai, 2020-10-27:
I use Hermit for stuff like this! Creates stand-alone browser-based "apps" for android. Works like a charm, and perfectly customisable (like fullscreen or not, etc)

Philippdosch, 2020-12-06:
Hi. I just found the following app, available in Google play store:
https://github.com/tkem/mopidy-mobile
It even has and android notification bar music controller which was the reason why I would wanted an App.
So far for me it works well.

Luegengladiator, 2021-02-10:
Since the system is based on a music player daemon, all music player clients (overview here: https://www.musicpd.org/clients/) can be used to control the typical music functions of the player. Change playlist, adjust volume, search for pieces of music ...
There are various clients in Android that support this function. 

I myself use MPDroid in Android and gmpc in the PC environment or rompr as a web service.

Once again clear: any function of the RFIDs cannot be controlled via these clients! No new cards can be registered or other functions can be controlled that were developed for Phoniebox itself.

s-martin, 2024-04-19:
From my point of view if we want to implement this feature we should make sure that the Web App works seamlessly on mobile.
Advantages:
- implementing only one code base 
- works on Android and iPhone (and other mobile devices, too).

We should implement this for V3.