MiczFlor/RPi-Jukebox-RFID #868: 🐛 | Spotifyversion and no Local MP3 play

hyperbit00, 2020-03-20:
## Bug
@ 19.03.2020 i installd a fresh raspbian lite buster os and make a
fresh on line developer installation of phoniebox

the Phoniebox works like a charm, excepts the local mp3 play...
### What I did

I added a Local mp3 file to the Phoniebox and conect this to a RFID Chip
When i try to load this file (with the chip or webinterface) there is no output

so I run a local libary scan (and it never stops?)

in console i typed  `tail -n 500 /var/log/syslog`

##Output

`Mar 20 11:58:14 phoniebox daemon_rfid_reader.py[753]: 2020-03-20 11:58:14,900 - root - INFO - Trigger Play Cardid=0002940831
Mar 20 11:58:15 phoniebox mopidy[773]: INFO     [MpdSession-910] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:54790
Mar 20 11:58:15 phoniebox mopidy[773]: INFO     [MpdSession-911] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:54794
Mar 20 11:58:15 phoniebox mopidy[773]: INFO     [MpdSession-912] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:54798
Mar 20 11:58:15 phoniebox mopidy[773]: INFO     [MpdSession-913] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:54802
Mar 20 11:58:15 phoniebox daemon_rfid_reader.py[753]: volume: 30%   repeat: off   random: off   single: off   consume: off
Mar 20 11:58:15 phoniebox mopidy[773]: INFO     [MpdSession-914] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:54806
Mar 20 11:58:15 phoniebox daemon_rfid_reader.py[753]: volume: 30%   repeat: off   random: off   single: off   consume: off
Mar 20 11:58:15 phoniebox mopidy[773]: INFO     [MpdSession-915] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:54810
Mar 20 11:58:15 phoniebox mopidy[773]: ERROR    [LocalBackend-7] mopidy_local.library Lookup error for local:track:Testordner/DE47B2~1.MP3: no such table: tracks
Mar 20 11:58:15 phoniebox mopidy[773]: ERROR    [LocalBackend-7] mopidy_local.library Lookup error for local:track:Testordner/DERMAI~1.MP3: no such table: tracks
Mar 20 11:58:15 phoniebox daemon_rfid_reader.py[753]: loading: Testordner
Mar 20 11:58:15 phoniebox mopidy[773]: INFO     [MpdSession-916] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:54814
Mar 20 11:58:15 phoniebox daemon_rfid_reader.py[753]: volume: 30%   repeat: off   random: off   single: off   consume: off
Mar 20 11:58:15 phoniebox mopidy[773]: INFO     [MpdSession-917] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:54818
Mar 20 11:58:15 phoniebox daemon_rfid_reader.py[753]: volume: 30%   repeat: off   random: off   single: off   consume: off
Mar 20 11:58:15 phoniebox mopidy[773]: INFO     [MpdSession-918] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:54822
Mar 20 11:58:15 phoniebox daemon_rfid_reader.py[753]: OK MPD 0.19.0
Mar 20 11:58:15 phoniebox daemon_rfid_reader.py[753]: OK
`

Internetradio, Spotify playlists works like a charm.. but local playback not :(

Any Advice?

s-martin, 2020-03-21:
Please try `mopidy local scan` and then post the output of `tail -n 500 /var/log/syslog`.

hyperbit00, 2020-03-21:
Did it via local Scan (webinterface) with the Output postet above
Is there a difference when i do it via console?

s-martin, 2020-03-21:
> Did it via local Scan (webinterface) with the Output postet above
> Is there a difference when i do it via console?

Not sure, but worth a try.

hyperbit00, 2020-03-21:
oky i did it via console
no change :(

[(https://pastebin.com/HUjadbVv)]

s-martin, 2020-03-22:
Are these the real file names?

`Mar 21 21:44:30 phoniebox mopidy[682]: ERROR    [LocalBackend-7] mopidy_local.library Lookup error for local:track:Testordner/DE47B2~1.MP3: no such table: tracks`

I found a hint here as well https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/640#issuecomment-568188673

hyperbit00, 2020-03-22:
was correct but i reuploaded the test files. No Change :(

https://pastebin.com/kQtHF6tB

hyperbit00, 2020-03-22:
i tryd 

> sudo apt-get install gstreamer1.0-plugins-bad -y

no Change :(

Maybee there is a error with add_callback?

> Mar 22 09:45:15 phoniebox mopidy[686]: INFO     [MpdSession-79] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56044
> Mar 22 09:45:15 phoniebox mopidy[686]: INFO     [MpdSession-80] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56048
> Mar 22 09:45:15 phoniebox mopidy[686]: ERROR    [IrisFrontend-13] mopidy.listener Triggering event failed: tracklist_changed()
> Mar 22 09:45:15 phoniebox mopidy[686]: Traceback (most recent call last):
> Mar 22 09:45:15 phoniebox mopidy[686]:   File "/usr/local/lib/python3.7/dist-packages/mopidy/listener.py", line 40, in on_event
> Mar 22 09:45:15 phoniebox mopidy[686]:     getattr(self, event)(**kwargs)
> Mar 22 09:45:15 phoniebox mopidy[686]:   File "/usr/local/lib/python3.7/dist-packages/mopidy_iris/frontend.py", line 29, in tracklist_changed
> Mar 22 09:45:15 phoniebox mopidy[686]:     iris.ioloop.add_callback(functools.partial(iris.clean_queue_metadata))
> Mar 22 09:45:15 phoniebox mopidy[686]: AttributeError: 'NoneType' object has no attribute 'add_callback'
> Mar 22 09:45:15 phoniebox mopidy[686]: INFO     [MpdSession-81] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56052
> Mar 22 09:45:15 phoniebox mopidy[686]: ERROR    [LocalBackend-7] mopidy_local.library Lookup error for local:track:Testordner/1.mp3: no such table: tracks
> Mar 22 09:45:15 phoniebox mopidy[686]: ERROR    [LocalBackend-7] mopidy_local.library Lookup error for local:track:Testordner/2.mp3: no such table: tracks
> Mar 22 09:45:15 phoniebox mopidy[686]: INFO     [MpdSession-82] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56056

s-martin, 2020-03-22:
According to this ticket https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/700#issuecomment-596083433 scanning and then restarting mopidy helped.

hyperbit00, 2020-03-22:
> According to this ticket [#700 (comment)](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/700#issuecomment-596083433) scanning and then restarting mopidy helped.

s-Martin i have installd the whole machine again.
This Time with One Line Raspbian Buster (recommended version) Installer.

But i get a 404 error @ iris --> so i follow your suggestion (Sticky: Spotify (Iris) does not work, e.g. error „404“ - Howto #852 

but now i have to start mopidy via console after each reboot of the machine :(

and i found out, that only the "mipody local scan" cmd make the local files work ... (the same via webinterface doesnt work although the search process was successful)

is that a bug? and how can i fix the automatic startup of mopidy

s-martin, 2020-03-22:
Try `sudo systemctl enable mopidy` and `sudo systemctl start mopidy`.

If that doesn't work, please show the logs again

hyperbit00, 2020-03-22:
Now everythin works (have to refresh db in phoniebox after each fileadd to get it work)

closed 2020-03-22 by hyperbit00