MiczFlor/RPi-Jukebox-RFID #870: 🐛 | BUG SUMMARY

erhapp, 2020-03-22:
## Bug

### What I did

2.0 rc7 install with script on Rasbian running on Raspberry Pi Zero W

### What happened

MQTT isn't sending "general server info" in on_connect event.

### I expected this to happen

Receiving state, version, edition, disk_total and disk_avail in MQTT server.

### Further information that might help

I can get the message to be send by adding "time.sleep(1)" right before the client.publish commands

## Software

### Base image and version

Raspbian GNU/Linux 10 (buster)

### Branch / Release

master

### Installscript

buster-install-default.sh

## Hardware

### RaspberryPi version

Raspberry Pi Zero W

s-martin, 2020-03-23:
If you have a possible fix, could you submit a pull request?

s-martin, 2022-12-20:
Closing as there was no activity for a very long time.

Please check the current release 2.4 or our new implementation 3.2, if this issue still persists.

Feel free to open a new issue, if necessary.

closed 2022-12-20 by s-martin