MiczFlor/RPi-Jukebox-RFID #873: 🐛 | BUG SUMMARY Rotary Encoder uses GPIOs outside of the gpio_buttons.py

hyperbit00, 2020-03-23:
Recently there was a change to the Jukebox Release Candidates.
The Rotary Encoder was introduced.

Usualy the GPIOs are configured in the gpio_buttons.py but for the rotary its done in the rotary-encoder.py.

It takes a long time for me to find out that the rotary-encoder causes the problems in my project (cause i use the GPIOs 5 and 6)

If the PINS were configured in the GPIOs-buttons.py, not the whole Code must be searchd for the unwanted GPIO use.

For Quick solution it would be nice, if the rotarty Encoder is disabled after new Setup

s-martin, 2020-03-23:
I see your issue, but I'm not sure, if there's a quick fix. We will look into that, thanks for reporting.

I wonder since when this issue exists, because the last change on `master` was in September last year.

hyperbit00, 2020-03-23:
maybee nobody uses the GPIOs 5 and 6 or stuck in an old software version :)

But some people use the step by step instructions from http://splittscheid.de/selfmade-phoniebox/ so they will run into this problem :)

Thanks for this nice Project!

splitti, 2020-03-29:
Truly, i get several questions while boxes are updatet and some things doesn't work. In my opinion, there should be a few points in the installer, asking if buttons-script should be activeted and with which GPIOs. The same on for the Rotary-Script. The predefied GPIOs in the Button-Script make hifiberry-Users also unhappy. Maybe the GPIO-Button- and Rotary-Script should be disabled and the Setup could be taken in the Web Interface, also the activation.

s-martin, 2020-03-30:
> Truly, i get several questions while boxes are updatet and some things doesn't work. In my opinion, there should be a few points in the installer, asking if buttons-script should be activeted and [...]

This problem is already noted in #790.

s-martin, 2020-10-14:
Closing this, since 2.1 gpio_control is used.

closed 2020-10-14 by s-martin