MiczFlor/RPi-Jukebox-RFID #874: 🐛 | Local File Play not working on Buster install

theogillie, 2020-03-26:
I have had a very difficult time trying to get local file play to work and I am at a loss at this point. I installed the raspberry pi with the buster image ran through the installer script without issue.  All other features work, Spotify works, webapp works, RFID reader works, Adding files via webapp to shared folder works, folder appears in webapp, etc. But when I try to play local mp3's, no luck. I have done probably half a dozen installs at this point and nothing seems to fix it.

I believe it is something happening between the scan of the files, because when I use `mopidy local scan` this is what gets returned again and again no matter the file:

pi@raspberrypi:~ $ mopidy local scan
/usr/lib/python2.7/dist-packages/mopidy/ext.py:202: PkgResourcesDeprecationWarning: Parameters to load are deprecated.  Call .resolve and .require separately.
  extension_class = entry_point.load(require=False)
INFO     Starting Mopidy 2.2.2
INFO     Loading config from builtin defaults
INFO     Loading config from /home/pi/.config/mopidy/mopidy.conf
INFO     Loading config from command line options
INFO     Enabled extensions: iris, mpd, http, stream, spotify, m3u, softwaremixer, file, local-images, local
INFO     Disabled extensions: none
INFO     Found 18 files in media_dir.
INFO     Checking 0 tracks from library.
INFO     Removing 0 missing tracks.
INFO     Found 15 tracks which need to be updated.
INFO     Scanning...
WARNING  Failed local:track:Your%20Old%20Droog%20-%20Transportation/01.%20Stillwell%20Baby.mp3: Timeout after 1000ms
WARNING  Failed local:track:Your%20Old%20Droog%20-%20Transportation/02.%20Monthly.mp3: Timeout after 1000ms
WARNING  Failed local:track:Your%20Old%20Droog%20-%20Transportation/03.%20Loosey%20Spot%20on%20Wheels.mp3: Timeout after 1000ms
WARNING  Failed local:track:Your%20Old%20Droog%20-%20Transportation/04.%20Under%20the%20Train%20%28Transporting%29.mp3: Timeout after 1000ms
WARNING  Failed local:track:Your%20Old%20Droog%20-%20Transportation/05.%20Head%20over%20Wheels%20II%20%28An%20Ode%20to%20G%20Dep%29.mp3: Timeout after 1000ms
WARNING  Failed local:track:Your%20Old%20Droog%20-%20Transportation/06.%20The%20Cheese.mp3: Timeout after 1000ms
WARNING  Failed local:track:Your%20Old%20Droog%20-%20Transportation/07.%20Rocket%20Launch%C2%A0Interlude.mp3: Timeout after 1000ms
WARNING  Failed local:track:Your%20Old%20Droog%20-%20Transportation/08.%20My%20Plane.mp3: Timeout after 1000ms
WARNING  Failed local:track:Your%20Old%20Droog%20-%20Transportation/09.%20Getting%20Off%20the%20Plane%C2%A0Interlude.mp3: Timeout after 1000ms
WARNING  Failed local:track:Your%20Old%20Droog%20-%20Transportation/10.%20SS%20YOD.mp3: Timeout after 1000ms
WARNING  Failed local:track:Your%20Old%20Droog%20-%20Transportation/11.%20Train%20Love.mp3: Timeout after 1000ms
WARNING  Failed local:track:Your%20Old%20Droog%20-%20Transportation/12.%20Bikes%2C%20Bleeps%2C%20Busses.mp3: Timeout after 1000ms
WARNING  Failed local:track:Your%20Old%20Droog%20-%20Transportation/13.%20207.mp3: Timeout after 1000ms
WARNING  Failed local:track:Your%20Old%20Droog%20-%20Transportation/14.%20Taxi%20%28Bonus%29.mp3: Timeout after 1000ms
WARNING  Failed local:track:Your%20Old%20Droog%20-%20Transportation/15.%20Vigilantes%20%28Bonus%20Remix%29.mp3: Timeout after 1000ms
INFO     Scanned 15 of 15 files in 25s.
INFO     Done scanning.

I have a Pi Zero, fresh install. Any help would be greatly appreciated

s-martin, 2020-03-26:
Currently you need a workaround described in #852

axelslade, 2020-03-31:
Hi, I have the same issue. Done the Script in #852, but local files does not get played - as well as the start sound does not getting played / the shutdown gets played. 

I did a clean install with the buster script including spotify.
Any ideas how I can solve this?

s-martin, 2020-03-31:
Did you run `mopidy local scan` again?

And please post the result of `journalctl -u mopidy`.

axelslade, 2020-03-31:
`**mopidy local scan**
Traceback (most recent call last):
  File "/usr/local/bin/mopidy", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 3191, in <module>
    @_call_aside
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 3175, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 3204, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 583, in _build_master
    ws.require(__requires__)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 900, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 786, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'Mopidy==2.2.2' distribution was not found and is required by the application`

`journalctl -u mopidy
-- Logs begin at Tue 2020-03-31 21:29:25 BST, end at Tue 2020-03-31 21:32:54 BST. --
Mar 31 21:29:37 raspberrypi systemd[1]: Starting Mopidy music server...
Mar 31 21:29:37 raspberrypi systemd[1]: Started Mopidy music server.
Mar 31 21:29:41 raspberrypi mopidy[541]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.1
Mar 31 21:29:41 raspberrypi mopidy[541]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Mar 31 21:29:41 raspberrypi mopidy[541]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Mar 31 21:29:41 raspberrypi mopidy[541]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Mar 31 21:29:41 raspberrypi mopidy[541]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Mar 31 21:29:41 raspberrypi mopidy[541]: INFO     [MainThread] mopidy.config Loading config from command line options
Mar 31 21:29:41 raspberrypi mopidy[541]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
Mar 31 21:29:42 raspberrypi mopidy[541]: INFO     [MainThread] mopidy.__main__ Enabled extensions: file, stream, local, mpd, http, softwaremixer, spotify, iris, m3u
Mar 31 21:29:42 raspberrypi mopidy[541]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Mar 31 21:29:43 raspberrypi mopidy[541]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Mar 31 21:29:43 raspberrypi mopidy[541]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Mar 31 21:29:43 raspberrypi mopidy[541]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Mar 31 21:29:43 raspberrypi mopidy[541]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, SpotifyBackend, LocalBackend
Mar 31 21:29:43 raspberrypi mopidy[541]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Mar 31 21:29:44 raspberrypi mopidy[541]: INFO     [SpotifyBackend-6] mopidy_spotify.web Logged into Spotify Web API as q6i39042ajj1agx4kf8io2brl
Mar 31 21:29:44 raspberrypi mopidy[541]: INFO     [SpotifyEventLoop] mopidy_spotify.backend Logged in to Spotify in online mode
Mar 31 21:29:44 raspberrypi mopidy[541]: INFO     [SpotifyBackend-6] mopidy_spotify.playlists Refreshed 3 Spotify playlists
Mar 31 21:29:44 raspberrypi mopidy[541]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Mar 31 21:29:44 raspberrypi mopidy[541]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: IrisFrontend, HttpFrontend, MpdFrontend
Mar 31 21:29:44 raspberrypi mopidy[541]: INFO     [IrisFrontend-11] mopidy_iris.core Starting Iris 3.46.0
Mar 31 21:29:44 raspberrypi mopidy[541]: INFO     [HttpFrontend-13] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Mar 31 21:29:44 raspberrypi mopidy[541]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Mar 31 21:29:44 raspberrypi mopidy[541]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
Mar 31 21:29:50 raspberrypi mopidy[541]: INFO     [MpdSession-15] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36380
Mar 31 21:29:56 raspberrypi mopidy[541]: INFO     [MpdSession-16] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36384
Mar 31 21:30:12 raspberrypi mopidy[541]: INFO     [MpdSession-17] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36388
Mar 31 21:30:18 raspberrypi mopidy[541]: INFO     [MpdSession-18] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36392
Mar 31 21:30:24 raspberrypi mopidy[541]: INFO     [MpdSession-19] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36396
Mar 31 21:30:30 raspberrypi mopidy[541]: INFO     [MpdSession-20] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36400
Mar 31 21:30:36 raspberrypi mopidy[541]: INFO     [MpdSession-21] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36404
Mar 31 21:30:42 raspberrypi mopidy[541]: INFO     [MpdSession-22] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36408
Mar 31 21:30:48 raspberrypi mopidy[541]: INFO     [MpdSession-23] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36412
[... cut ...]
Mar 31 21:32:42 raspberrypi mopidy[541]: INFO     [MpdSession-42] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36488
Mar 31 21:32:48 raspberrypi mopidy[541]: INFO     [MpdSession-43] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36492
Mar 31 21:32:54 raspberrypi mopidy[541]: INFO     [MpdSession-44] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36496
lines 2-56/56 (END)                                                                                                                                                                      `

s-martin, 2020-03-31:
Did you reboot?

axelslade, 2020-03-31:
sry, here you go

`pi@raspberrypi:~ $ journalctl -u mopidy
-- Logs begin at Tue 2020-03-31 22:17:09 BST, end at Tue 2020-03-31 22:19:45 BST. --
Mar 31 22:17:20 raspberrypi systemd[1]: Starting Mopidy music server...
Mar 31 22:17:20 raspberrypi systemd[1]: Started Mopidy music server.
Mar 31 22:17:24 raspberrypi mopidy[543]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.1
Mar 31 22:17:24 raspberrypi mopidy[543]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Mar 31 22:17:24 raspberrypi mopidy[543]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Mar 31 22:17:24 raspberrypi mopidy[543]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Mar 31 22:17:24 raspberrypi mopidy[543]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Mar 31 22:17:24 raspberrypi mopidy[543]: INFO     [MainThread] mopidy.config Loading config from command line options
Mar 31 22:17:24 raspberrypi mopidy[543]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
Mar 31 22:17:25 raspberrypi mopidy[543]: INFO     [MainThread] mopidy.__main__ Enabled extensions: m3u, iris, stream, file, softwaremixer, mpd, local, spotify, http
Mar 31 22:17:25 raspberrypi mopidy[543]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Mar 31 22:17:26 raspberrypi mopidy[543]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Mar 31 22:17:26 raspberrypi mopidy[543]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Mar 31 22:17:26 raspberrypi mopidy[543]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Mar 31 22:17:26 raspberrypi mopidy[543]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, SpotifyBackend, LocalBackend
Mar 31 22:17:26 raspberrypi mopidy[543]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Mar 31 22:17:27 raspberrypi mopidy[543]: INFO     [SpotifyBackend-6] mopidy_spotify.web Logged into Spotify Web API as q6i39042ajj1agx4kf8io2brl
Mar 31 22:17:27 raspberrypi mopidy[543]: INFO     [SpotifyEventLoop] mopidy_spotify.backend Logged in to Spotify in online mode
Mar 31 22:17:28 raspberrypi mopidy[543]: INFO     [SpotifyBackend-6] mopidy_spotify.playlists Refreshed 3 Spotify playlists
Mar 31 22:17:28 raspberrypi mopidy[543]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Mar 31 22:17:28 raspberrypi mopidy[543]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: IrisFrontend, HttpFrontend, MpdFrontend
Mar 31 22:17:28 raspberrypi mopidy[543]: INFO     [IrisFrontend-11] mopidy_iris.core Starting Iris 3.46.0
Mar 31 22:17:28 raspberrypi mopidy[543]: INFO     [HttpFrontend-13] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Mar 31 22:17:28 raspberrypi mopidy[543]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Mar 31 22:17:28 raspberrypi mopidy[543]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
Mar 31 22:17:28 raspberrypi mopidy[543]: INFO     [MpdSession-15] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52516
Mar 31 22:17:29 raspberrypi mopidy[543]: INFO     [MpdSession-16] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52520
Mar 31 22:17:34 raspberrypi mopidy[543]: INFO     [MpdSession-17] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52528
Mar 31 22:17:36 raspberrypi mopidy[543]: INFO     [MpdSession-18] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52532
Mar 31 22:17:36 raspberrypi mopidy[543]: INFO     [MpdSession-19] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52536
Mar 31 22:17:36 raspberrypi mopidy[543]: INFO     [MpdSession-20] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52540
Mar 31 22:17:36 raspberrypi mopidy[543]: INFO     [MpdSession-21] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52544
Mar 31 22:17:36 raspberrypi mopidy[543]: INFO     [MpdSession-22] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52548
Mar 31 22:17:36 raspberrypi mopidy[543]: INFO     [MpdSession-23] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52552
Mar 31 22:17:36 raspberrypi mopidy[543]: INFO     [MpdSession-24] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52556
Mar 31 22:17:38 raspberrypi mopidy[543]: INFO     [MpdSession-27] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52560
Mar 31 22:17:38 raspberrypi mopidy[543]: INFO     [MpdSession-28] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52564
Mar 31 22:17:39 raspberrypi mopidy[543]: INFO     [MpdSession-29] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52568
Mar 31 22:17:41 raspberrypi mopidy[543]: INFO     [MpdSession-30] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52572
Mar 31 22:17:42 raspberrypi mopidy[543]: INFO     [MpdSession-31] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52576
Mar 31 22:17:42 raspberrypi mopidy[543]: INFO     [MpdSession-32] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52580
Mar 31 22:17:42 raspberrypi mopidy[543]: INFO     [MpdSession-33] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52584
[... cut ...]
Mar 31 22:19:40 raspberrypi mopidy[543]: INFO     [MpdSession-130] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52972
Mar 31 22:19:42 raspberrypi mopidy[543]: INFO     [MpdSession-131] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52976
Mar 31 22:19:42 raspberrypi mopidy[543]: INFO     [MpdSession-132] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52980
Mar 31 22:19:45 raspberrypi mopidy[543]: INFO     [MpdSession-133] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52984`

axelslade, 2020-03-31:
Update: it works... i pressed renew database in the webfrontent and now it works

Viatorus, 2020-04-04:
I had the same issue. After renewing the database, local files work but streams (like the sample  LiveStream-Bayern2) don't. I get this error inside Mopidy:

```
Apr 04 22:21:02 phoniebox mopidy[1967]: Traceback (most recent call last):
Apr 04 22:21:02 phoniebox mopidy[1967]:   File "/usr/lib/python3/dist-packages/mopidy/listener.py", line 40, in on_event
Apr 04 22:21:02 phoniebox mopidy[1967]:     getattr(self, event)(**kwargs)
Apr 04 22:21:02 phoniebox mopidy[1967]:   File "/usr/local/lib/python3.7/dist-packages/mopidy_iris/frontend.py", line 29, in tracklis
Apr 04 22:21:02 phoniebox mopidy[1967]:     iris.ioloop.add_callback(functools.partial(iris.clean_queue_metadata))
Apr 04 22:21:02 phoniebox mopidy[1967]: AttributeError: 'NoneType' object has no attribute 'add_callback'
```

Viatorus, 2020-04-05:
Here is a detailed log:

```
 INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
 INFO     [SpotifyEventLoop] mopidy_spotify.backend Logged in to Spotify in online mode
 INFO     [SpotifyBackend-6] mopidy_spotify.web Logged into Spotify Web API as 1131874961
 INFO     [SpotifyBackend-6] mopidy_spotify.playlists Refreshed 0 Spotify playlists
 INFO     [MainThread] mopidy.commands Starting Mopidy core
 INFO     [MainThread] mopidy.commands Starting Mopidy frontends: IrisFrontend, HttpFrontend, MpdFrontend
 INFO     [IrisFrontend-11] mopidy_iris.core Starting Iris 3.45.0
 INFO     [HttpFrontend-13] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
 INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
 INFO     [MainThread] mopidy.commands Starting GLib mainloop
 INFO     [MpdSession-15] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39072
 INFO     [MpdSession-16] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39076
 INFO     [MpdSession-17] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39080
 INFO     [MpdSession-18] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39086
 INFO     [MpdSession-19] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39090
 INFO     [MpdSession-20] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39094
 INFO     [MpdSession-21] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39106
 INFO     [MpdSession-22] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39114
 INFO     [MpdSession-23] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39118
 gst_element_make_from_uri: assertion 'gst_uri_is_valid (uri)' failed
 WARNING  [StreamBackend-5] mopidy.internal.http Download of 's:\x1a' failed due to unsupported schema
 [206B blob data]
 WARNING  [StreamBackend-5] mopidy.stream.actor Problem looking up https://tracking.neuland.br.de/file/1792078/c/feed/waermster-januar-in-europa-seit-39-jahren.mp3
 INFO     [MpdSession-24] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39130
 INFO     [MpdSession-25] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39134
 ERROR    [Core-10] mopidy.core.library StreamBackend backend caused an exception.
 Traceback (most recent call last):
   File "/usr/lib/python3/dist-packages/mopidy/core/library.py", line 17, in _backend_error_handling
     yield
   File "/usr/lib/python3/dist-packages/mopidy/core/library.py", line 217, in lookup
     result = future.get()
   File "/usr/lib/python3/dist-packages/pykka/_threading.py", line 45, in get
     _compat.reraise(*self._data['exc_info'])
   File "/usr/lib/python3/dist-packages/pykka/_compat/__init__.py", line 29, in reraise
     raise value
   File "/usr/lib/python3/dist-packages/pykka/_actor.py", line 193, in _actor_loop
     response = self._handle_receive(envelope.message)
   File "/usr/lib/python3/dist-packages/pykka/_actor.py", line 299, in _handle_receive
     return callee(*message.args, **message.kwargs)
   File "/usr/lib/python3/dist-packages/mopidy/stream/actor.py", line 70, in lookup
     requests_session=self.backend._session,
   File "/usr/lib/python3/dist-packages/mopidy/stream/actor.py", line 138, in _unwrap_stream
     scan_result = scanner.scan(uri, timeout=scan_timeout)
   File "/usr/lib/python3/dist-packages/mopidy/audio/scan.py", line 66, in scan
     pipeline, signals = _setup_pipeline(uri, self._proxy_config)
   File "/usr/lib/python3/dist-packages/mopidy/audio/scan.py", line 83, in _setup_pipeline
     src = Gst.Element.make_from_uri(Gst.URIType.SRC, uri)
 gi.repository.GLib.GError: gst-uri-error-quark: No URI handler for the c. protocol found (0)
 INFO     [MpdSession-26] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39142
 INFO     [MpdSession-27] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39146
 gst_element_make_from_uri: assertion 'gst_uri_is_valid (uri)' failed
 WARNING  [StreamBackend-5] mopidy.internal.http Download of 'K:\x00\x00\x00' failed due to unsupported schema
 INFO     [StreamBackend-5] mopidy.stream.actor Unwrapping stream from URI (https://tracking.neuland.br.de/file/1790365/c/feed/zaehlung-der-wintervoegel-2020.mp3) failed: error downloading URI K:
 WARNING  [StreamBackend-5] mopidy.stream.actor Problem looking up https://tracking.neuland.br.de/file/1790365/c/feed/zaehlung-der-wintervoegel-2020.mp3
 INFO     [MpdSession-28] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39154
 INFO     [MpdSession-29] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39158
 ERROR    [Core-10] mopidy.core.library StreamBackend backend caused an exception.
 Traceback (most recent call last):
   File "/usr/lib/python3/dist-packages/mopidy/core/library.py", line 17, in _backend_error_handling
     yield
   File "/usr/lib/python3/dist-packages/mopidy/core/library.py", line 217, in lookup
     result = future.get()
   File "/usr/lib/python3/dist-packages/pykka/_threading.py", line 45, in get
     _compat.reraise(*self._data['exc_info'])
   File "/usr/lib/python3/dist-packages/pykka/_compat/__init__.py", line 29, in reraise
     raise value
   File "/usr/lib/python3/dist-packages/pykka/_actor.py", line 193, in _actor_loop
     response = self._handle_receive(envelope.message)
   File "/usr/lib/python3/dist-packages/pykka/_actor.py", line 299, in _handle_receive
     return callee(*message.args, **message.kwargs)
[... cut ...]
 WARNING  [StreamBackend-5] mopidy.internal.http Download of 'x:' failed due to unsupported schema
 INFO     [StreamBackend-5] mopidy.stream.actor Unwrapping stream from URI (https://tracking.neuland.br.de/file/1787223/c/feed/bauernproteste-in-berlin.mp3) failed: error downloading URI x:
 WARNING  [StreamBackend-5] mopidy.stream.actor Problem looking up https://tracking.neuland.br.de/file/1787223/c/feed/bauernproteste-in-berlin.mp3
 INFO     [MpdSession-32] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39192
```

s-martin, 2020-04-05:
Looks like either the stream is broken or incompatible.

Did you try other stream sources?

Viatorus, 2020-04-07:
Ah, all podcasts don't work with mopidy some how.. Like the ones bundled with CreatePodcastsKidsDeutsch.sh.

s-martin, 2020-04-10:
did you by any change edit the text files with a Windows editor?

s-martin, 2020-12-15:
@Viatorus  could you check, if the following fixes your issue:

1. Replace content of requirements-spotify.txt with `Mopidy-Iris==3.54.2`
2. `sudo pip install --upgrade --force-reinstall -r requirements-spotify.txt`
3. Reboot

closed 2020-12-18 by s-martin