MiczFlor/RPi-Jukebox-RFID #875: 🐛 | Phoniebox starts randomly playing music

georgbachmann, 2020-03-26:
Lately my phoniebox randomly starts playing the last audiobook that was played on it. I noticed it a couple of times the last few days that it was playing and wasn't sure it it were the kids who turned it on, but right now I was in there with them to put them to sleep when the box would just start playing. There was no card on it... so I have no idea why it's doing that?
Are there some logs that I can have a look at to see the reason it started playing?
Thanks a lot for your help
Cheers,
Georg

ozone89, 2020-03-27:
Going on a limb here, but... maybe the cards are near the reader enough to trigger it? have you tried moving them further and/or shielding the reader with some foil?

georgbachmann, 2020-03-27:
Thank you for your help @ozone89 . The cards were actually in an other room (my kids like to spread them a lot ;) ). So that could not have been the case... :(

mduesi, 2020-03-27:
I have observed the same behavior. The box is in my son's room and in the middle of the night a radio play suddenly went on...

MiczFlor, 2020-03-31:
@georgbachmann @mduesi 
Are you using the *Classic* edition of Phoniebox without Spotify or the *+Spotify* version?

georgbachmann, 2020-03-31:
@MiczFlor I am using the classic one. I did setup everything new like 2-3 weeks ago from git.
Is there any information I can give you to help find the problem? Logs or so?

s-martin, 2020-03-31:
Please post the output of `tail -n 500 /var/log/syslog`

georgbachmann, 2020-03-31:
It's all from today though... I guess there was no error today...
But I'll update the logs if the problem occurs again...
Here are the logs from today:
https://pastebin.com/DcQ5itjX

MiczFlor, 2020-04-08:
This was 8 days ago. @georgbachmann did it happen again or was it a glitch in the early hours after the Phoniebox first launch (not that that should explain anything ;)

georgbachmann, 2020-04-09:
Hm... at least I didn't notice it... But it's in the kids room and my office has 2 doors in-between...
I post logs if it happens again, ok? Thank you very much for your help! (and this awesome project in general :) )

georgbachmann, 2020-04-14:
Ok... It just happed again right now. Here are the logs

https://pastebin.com/dpf0T9Lq

in the logs I saw:

```Apr 14 09:08:16 raspberrypi gpio-buttons.py[697]: Peppa Wutz/Peppa Wutz - E026 - Schnee.mp3
Apr 14 09:08:16 raspberrypi gpio-buttons.py[697]: [playing] #2/6   0:14/4:45 (4%)
```

but nobody was even in the room to touch the GPIO Buttons?!? Hm... Any idea?

ozone89, 2020-04-14:
Buttons got dirty somehow and are now shorted?

georgbachmann, 2020-04-14:
Hm... I guess not?!? They are as they always have been...
I'll open my box tonight and have a look if I can see something. Do you have any tips on how to see such a problem?
Thanks for helping!!!

ozone89, 2020-04-14:
Post a picture of the buttons and gpio whenever possible

s-martin, 2020-12-18:
Closing this issue now, feel free to reopen, if the problem persists

closed 2020-12-18 by s-martin