MiczFlor/RPi-Jukebox-RFID #876: No spotify after restart

Marleme, 2020-03-28:
## Hi I'm new, couldn't find a solution in the documentation. Thanks for the great project!!

Every time I restart my Phoniebox card linked to a spotify stream won't work. All local folders or other internet streams do work.

### When I scan the folders and go in the Spotify menu in the webinterface it seems to restart the service and it always works flawlessly til the next restart. It would be great if you could help me. Thanks a lot

s-martin, 2020-03-28:
Please try 
`sudo systemctl enable mopidy`
`sudo systemctl start mopidy`

Marleme, 2020-03-28:
Thank you for your reply, but unfortunately it is still the same.

s-martin, 2020-03-28:
Please post results of `tail -n 500 /var/log/syslog` and `journalctl -u mopidy`>

Marleme, 2020-03-31:
Sorry my account was flagged (don't kknow why) and so my posts were hidden. There are login errors in the file, but I don't know why actually.