MiczFlor/RPi-Jukebox-RFID #878: 🐛 | Resume Probleme

splitti, 2020-03-29:
I make a fresh installation of my two boxes three weeks ago by the OneLineInstaller. Everything works fine, but Resume does not work. The Playlist always starts from the beginning.  We use the box with large playlists, with about 1.000 titles (Spotify Playlist).

The setup wasn't changed. Resume is activated.

s-martin, 2020-03-30:
Do you have anything in the log files?

splitti, 2020-03-30:
First of all a few more information.... I work with the shutdown via OnOffShim. In the daemon, i added the following lines for the shutdown:

> /home/pi/RPi-Jukebox-RFID/scripts/./resume_play.sh -c=savepos && mpc clear

So, there is another step i really recognized firstly today! After starting the box or make 'mpc clear' the first RFId Command does not work. You need to swipe two times!

The user-persmissons in the audiofolder directory are pi:pi. 

folder.conf will be written, while the command "resume_play.sh -c savepos && mpc clear" will be executed. Looks like this:

> CURRENTFILENAME="spotify:track:6VJuRRLuMUmMrrU78KD7pL"
> ELAPSED="9.794"
> PLAYSTATUS="Stopped"
> RESUME="ON"
> SHUFFLE="OFF"
> LOOP="OFF"
> SINGLE="OFF"

Next Swipe seems not to be recongnized.

latestID.txt  looks like this:
> Card ID '0008830168' was used at '2020-03-30.18:34:02'.
> This ID has been used before.
> The shortcut points to audiofolder 'Ninjago_All'.

The Logs shows this:

> ########### SCRIPT playout_controls.sh (2020-03-30.20:15:35) ##
>   ######### SCRIPT inc.readArgsFromCommandLine.sh (2020-03-30.20:15:35) ##
>   # VAR CARDID: 
>   # VAR COMMAND: playlistaddplay
>   # VAR FOLDER: Ninjago_All
>   # VAR VALUE: Ninjago_All
> VAR COMMAND: playlistaddplay
> VAR VALUE: Ninjago_All
>    playlistaddplay playlist name VALUE: Ninjago_All
>    playlistaddplay FOLDER: Ninjago_All
> mpc load Ninjago_All
>   ######### SCRIPT inc.readArgsFromCommandLine.sh (2020-03-30.20:15:37) ##
>   # VAR CARDID: 
>   # VAR COMMAND: single_check
>   # VAR FOLDER: Ninjago_All
>   # VAR VALUE: 
> ## SCRIPT single_play.sh (2020-03-30.20:15:37) ##
> VAR AUDIOFOLDERSPATH: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
> VAR COMMAND: single_check
> VAR VALUE: 
> VAR FOLDER: Ninjago_All
>   content of /home/pi/RPi-Jukebox-RFID/shared/audiofolders/Ninjago_All/folder.conf
> CURRENTFILENAME="spotify:track:6oYvJ4wq2HPBkJySKDqw8T"
> ELAPSED="10.476"
> PLAYSTATUS="Stopped"
> RESUME="ON"
> SHUFFLE="OFF"
> LOOP="OFF"
> SINGLE="OFF"
>   Now doing what COMMAND wants: single_check
>   entering: single_check with value OFF
>   ######### SCRIPT inc.readArgsFromCommandLine.sh (2020-03-30.20:15:37) ##
>   # VAR CARDID: 
>   # VAR COMMAND: shuffle_check
>   # VAR FOLDER: Ninjago_All
>   # VAR VALUE: 
> ## SCRIPT shuffle_play.sh (2020-03-30.20:15:37) ##
> VAR AUDIOFOLDERSPATH: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
> VAR COMMAND: shuffle_check
> VAR VALUE: 
> VAR FOLDER: Ninjago_All
> # found content in folder.conf:
> CURRENTFILENAME="spotify:track:6oYvJ4wq2HPBkJySKDqw8T"
> ELAPSED="10.476"
> PLAYSTATUS="Stopped"
> RESUME="ON"
> SHUFFLE="OFF"
> LOOP="OFF"
> SINGLE="OFF"
>   Now doing what COMMAND wants: shuffle_check
>   entering: shuffle_check with value OFF
> # new content in folder.conf:
> CURRENTFILENAME="spotify:track:6oYvJ4wq2HPBkJySKDqw8T"
> ELAPSED="10.476"
> PLAYSTATUS="Stopped"
> RESUME="ON"
> SHUFFLE="OFF"
> LOOP="OFF"
> SINGLE="OFF"
>   ######### SCRIPT inc.readArgsFromCommandLine.sh (2020-03-30.20:15:37) ##
>   # VAR CARDID: 
>   # VAR COMMAND: resume
>   # VAR FOLDER: Ninjago_All
>   # VAR VALUE: 
> #START##### SCRIPT resume_play.sh (2020-03-30.20:15:37) ##
> VAR AUDIOFOLDERSPATH: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
> VAR COMMAND: resume
> VAR VALUE: 
> VAR FOLDER: Ninjago_All
>   content of /home/pi/RPi-Jukebox-RFID/shared/audiofolders/Ninjago_All/folder.conf
> CURRENTFILENAME="spotify:track:6oYvJ4wq2HPBkJySKDqw8T"
> ELAPSED="10.476"
> PLAYSTATUS="Stopped"
> RESUME="ON"
> SHUFFLE="OFF"
> LOOP="OFF"
> SINGLE="OFF"
>   Now doing what COMMAND wants: resume
>   savepos FOLDER: Ninjago_All
>   entering: resume with value ON
>   entering: single with value OFF
>   ######### SCRIPT inc.readArgsFromCommandLine.sh (2020-03-30.20:15:37) ##
>   # VAR CARDID: 
>   # VAR COMMAND: resume
>   # VAR FOLDER: Ninjago_All
>   # VAR VALUE: 
> ########### SCRIPT inc.writeFolderConfig.sh (2020-03-30.20:15:37) ##
> VAR COMMAND: resume
> VAR FOLDER: Ninjago_All
> VAR AUDIOFOLDERSPATH: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
> CHECK FOLDER EXISTS: /home/pi/RPi-Jukebox-RFID/shared/audiofolders/Ninjago_All
>   KEEP NEW VARS IN MIND
> VAR NEWCURRENTFILENAME: spotify:track:6oYvJ4wq2HPBkJySKDqw8T
>   KEEP SINGLE OFF IN MIND
>   content of /home/pi/RPi-Jukebox-RFID/shared/audiofolders/Ninjago_All/folder.conf
> CURRENTFILENAME="spotify:track:6oYvJ4wq2HPBkJySKDqw8T"
> ELAPSED="10.476"
> PLAYSTATUS="Stopped"
> RESUME="ON"
> SHUFFLE="OFF"
> LOOP="OFF"
> SINGLE="OFF"
>   REPLACE VALUES FROM THE CONFIG FILE WITH NEW ONES
> VAR CURRENTFILENAME: spotify:track:6oYvJ4wq2HPBkJySKDqw8T
>   WRITE THE VALUES INTO THE NEWLY CREATED RAW CONFIG
> #END####### SCRIPT resume_play.sh (2020-03-30.20:15:37) ##
> mpc load Ninjago_All && /home/pi/RPi-Jukebox-RFID/scripts/resume_play.sh -c=resume -d=Ninjago_All
>   echo Ninjago_All > /home/pi/RPi-Jukebox-RFID/scripts/../settings/Latest_Folder_Played
>   VAR Latest_Folder_Played: Ninjago_All
>   # end playout_controls.sh playlistaddplay
>   Command: /home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=playlistaddplay -v="Ninjago_All" -d="Ninjago_All"

But the box plays nothing. Next card swipe make a fresh start of the playlist.

quottax, 2020-04-01:
Which MPD-Version do you have?
There is a problem with the 2.x version. Maybe try to update to 3.x.

s-martin, 2020-04-01:
What does `tail -n 500 /var/log/syslog` show?

s-martin, 2020-05-17:
@splitti, some fixes regarding resume were merged to `develop`, could you check, if your issue still persists.

s-martin, 2020-05-26:
We've released 2.0, i.e. merged develop to master with changes on the resume implementation. 
Could you check, if this helps.

StefanMinke, 2020-08-11:
Hello,

I experienced the same behaviour, even with version 2.0. (master)
With my few programmer skills I found out that obviously the seek command of mpd does not work correctly in this case.

In /scripts/resume_play.sh I replaced:

```
echo -e seek $PLAYLISTPOS $ELAPSED \\nclose | nc -w 1 localhost 6600
```
with:

```
echo -e "play $PLAYLISTPOS" | nc -w 1 localhost 6600
echo -e seekcur $ELAPSED \\nclose | nc -w 1 localhost 6600
```

What this does is to play the needed title from the beginning an after that seek to the specified position within this current track. You can hear this behaviour, but with these little changes the resume funcion works good for me. And I guess this workaround may solve other resume correlated problems / bugs (playlist empty, stop after 1 song), too.  

Splitti, regarding the OnOffShim issue I don't have any problems with this. I also save the position and clear mpc within the /etc/cleanshutd.conf and this works good for me with the next startup and resume. Maybe you placed it at a wrong position?

s-martin, 2020-08-12:
Cool, that you found this workaround.

Do you want to open a PR?

StefanMinke, 2020-11-19:
Hey.

I updated my Phoniebox to the current version, hopefully the resume problem would disappear. And obviously there is a different behaviour. 
When I would like to resume a playlist the seek (resume_play.sh echo -e seek $PLAYLISTPOS $ELAPSED \\nclose | nc -w 1 localhost 6600) command seems to be broken. I don't know why, but the current song will be played till the end and thereafter the playlist wouldn't be continued. The result is that the software thinks the playlist was finished with the last song and starts from the beginning.

Can anyone reproduce this?

There is no problem with the savepos command, the config file is filled correctly. I checked it by the debug mode.
There is a problem with the resume command.

With my solution mentioned above, this issue is no problem anymore.

Maybe my solution could solve this problem als well (https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1041)

PR #1149 (2020-11-22): Fix resume
Fixes #878.
commit c40ab6c2 (2020-11-22): Fix resume

Fixes #878
closed 2020-11-22 by s-martin