MiczFlor/RPi-Jukebox-RFID #880: Local works, spotify play🐛 | BUG SUMMARY

johannescramer, 2020-03-30:
## Bug
When linking a spotify playlist to a card it links, but when pressing play on phoniebox home screen or swiping the card, nothing happens. When I go to mopidy (i.e. http://192.168.1.27:6680/iris/library/tracks) and start playing the tracks from there, it works. So mopidy works, spotify does, too, but for some reason communication with phoniebox doesn't.
Local files work flawlessly.

### Base image and version
latest buster
### Branch / Release
master
### Installscript
scripts/installscripts/buster-install-default.sh`
## Hardware
-
### RaspberryPi version
3 B+
### RFID Reader
RFID card reader via GPIOs (works)
### Soundcard
none (integrated)

s-martin, 2020-03-30:
Please post the output of `tail -n 500 /var/log/syslog` and `journalctl -u mopidy`

johannescramer, 2020-03-30:
thanks for the quick reply 

tail -n 500 /var/log/syslog
`Mar 30 20:16:58 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:17:02 raspberrypi CRON[7492]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Mar 30 20:17:04 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:17:10 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:17:14 raspberrypi dhcpcd[412]: wlan0: Router Advertisement from fe80::925c:44ff:fe79:2f16
Mar 30 20:17:16 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:17:22 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:17:28 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:17:34 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:17:40 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:17:46 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:17:52 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:17:58 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:18:04 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:18:10 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:18:16 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:18:18 raspberrypi dhcpcd[412]: wlan0: Router Advertisement from fe80::925c:44ff:fe79:2f16
Mar 30 20:18:22 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:18:28 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:18:34 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:18:36 raspberrypi dhcpcd[412]: wlan0: Router Advertisement from fe80::925c:44ff:fe79:2f16
Mar 30 20:18:40 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:18:46 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:18:52 raspberrypi dhcpcd[412]: wlan0: Router Advertisement from fe80::925c:44ff:fe79:2f16
Mar 30 20:18:52 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:18:58 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:19:04 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:19:08 raspberrypi dhcpcd[412]: wlan0: Router Advertisement from fe80::925c:44ff:fe79:2f16
Mar 30 20:19:10 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:19:16 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:19:22 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:19:28 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:19:34 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:19:40 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:19:46 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:19:52 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:19:58 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:20:04 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:20:10 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:20:16 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:20:22 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:20:28 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:20:34 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:20:40 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:20:46 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:20:52 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:20:58 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:21:04 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:21:10 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:21:16 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:21:22 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:21:28 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:21:34 raspberrypi mopidy[1343]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:21:35 raspberrypi dhcpcd[412]: wlan0: Router Advertisement from fe80::925c:44ff:fe79:2f16
[... cut ...]
Mar 30 20:40:04 raspberrypi mopidy[9101]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:40:10 raspberrypi mopidy[9101]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:40:16 raspberrypi mopidy[9101]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:40:22 raspberrypi mopidy[9101]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:40:28 raspberrypi mopidy[9101]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 20:40:34 raspberrypi mopidy[9101]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
`

johannescramer, 2020-03-30:
journalctl -u mopidy

`-- Logs begin at Sat 2020-03-21 21:22:27 GMT, end at Mon 2020-03-30 20:49:22 BST. --

Mar 21 21:22:32 raspberrypi systemd[1]: Starting Mopidy music server...
Mar 21 21:22:32 raspberrypi systemd[1]: Started Mopidy music server.
Mar 21 21:22:39 raspberrypi mopidy[461]: /usr/lib/python2.7/dist-packages/mopidy/ext.py:202: PkgResourcesDeprecationWarning: Pa
Mar 21 21:22:39 raspberrypi mopidy[461]:   extension_class = entry_point.load(require=False)
Mar 21 21:22:44 raspberrypi mopidy[461]: INFO     Starting Mopidy 2.2.2
Mar 21 21:22:44 raspberrypi mopidy[461]: INFO     Loading config from builtin defaults
Mar 21 21:22:44 raspberrypi mopidy[461]: INFO     Loading config from /etc/mopidy/mopidy.conf
Mar 21 21:22:44 raspberrypi mopidy[461]: INFO     Loading config from command line options
Mar 21 21:22:45 raspberrypi mopidy[461]: INFO     Enabled extensions: iris, mpd, http, stream, spotify, m3u, softwaremixer, fil
Mar 21 21:22:45 raspberrypi mopidy[461]: INFO     Disabled extensions: none
Mar 21 21:22:45 raspberrypi mopidy[461]: INFO     Starting Mopidy mixer: SoftwareMixer
Mar 21 21:22:45 raspberrypi mopidy[461]: INFO     Mixer volume set to 30
Mar 21 21:22:45 raspberrypi mopidy[461]: INFO     Starting Mopidy audio
Mar 21 21:22:45 raspberrypi mopidy[461]: INFO     Starting Mopidy backends: SpotifyBackend, StreamBackend, M3UBackend, FileBack
Mar 21 21:22:46 raspberrypi mopidy[461]: INFO     Loaded 1 local tracks using json
Mar 21 21:22:46 raspberrypi mopidy[461]: INFO     Audio output set to "alsasink"
Mar 21 21:22:47 raspberrypi mopidy[461]: INFO     Logged in to Spotify in online mode
Mar 21 21:22:47 raspberrypi mopidy[461]: ERROR    Authorization failed, not attempting Spotify API request. Please get new cred
Mar 21 21:22:47 raspberrypi mopidy[461]: ERROR    OAuth token refresh failed: invalid_client Client not known.
Mar 21 21:22:47 raspberrypi mopidy[461]: ERROR    Failed to load Spotify user profile
Mar 21 21:22:47 raspberrypi mopidy[461]: INFO     Started spotify backend
Mar 21 21:22:47 raspberrypi mopidy[461]: INFO     Loaded 0 playlists
Mar 21 21:22:47 raspberrypi mopidy[461]: INFO     Refresh Playlists took 1ms
Mar 21 21:22:47 raspberrypi mopidy[461]: INFO     Starting Mopidy core
Mar 21 21:22:47 raspberrypi mopidy[461]: INFO     Starting Mopidy frontends: IrisFrontend, MpdFrontend, HttpFrontend
Mar 21 21:22:47 raspberrypi mopidy[461]: INFO     Starting Iris 3.43.0
Mar 21 21:22:47 raspberrypi mopidy[461]: INFO     MPD server running at [::ffff:0.0.0.0]:6600
Mar 21 21:22:47 raspberrypi mopidy[461]: INFO     Starting GLib mainloop
Mar 21 21:22:47 raspberrypi mopidy[461]: INFO     HTTP server running at [::ffff:0.0.0.0]:6680
Mar 30 14:26:19 raspberrypi mopidy[461]: INFO     Logged in to Spotify in online mode
Mar 30 15:39:17 raspberrypi mopidy[461]: INFO     Logged in to Spotify in online mode
Mar 30 16:01:41 raspberrypi mopidy[461]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 16:01:42 raspberrypi mopidy[461]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 16:01:46 raspberrypi mopidy[461]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 16:01:46 raspberrypi mopidy[461]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 16:01:46 raspberrypi mopidy[461]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 16:01:46 raspberrypi mopidy[461]: ERROR    SpotifyBackend backend caused an exception.
Mar 30 16:01:46 raspberrypi mopidy[461]: Traceback (most recent call last):
Mar 30 16:01:46 raspberrypi mopidy[461]:   File "/usr/lib/python2.7/dist-packages/mopidy/core/library.py", line 19, in _backend
Mar 30 16:01:46 raspberrypi mopidy[461]:     yield
Mar 30 16:01:46 raspberrypi mopidy[461]:   File "/usr/lib/python2.7/dist-packages/mopidy/core/library.py", line 237, in lookup
Mar 30 16:01:46 raspberrypi mopidy[461]:     result = future.get()
Mar 30 16:01:46 raspberrypi mopidy[461]:   File "/usr/lib/python2.7/dist-packages/pykka/threading.py", line 52, in get
Mar 30 16:01:46 raspberrypi mopidy[461]:     compat.reraise(*self._data['exc_info'])
Mar 30 16:01:46 raspberrypi mopidy[461]:   File "/usr/lib/python2.7/dist-packages/pykka/compat.py", line 12, in reraise
Mar 30 16:01:46 raspberrypi mopidy[461]:     exec('raise tp, value, tb')
Mar 30 16:01:46 raspberrypi mopidy[461]:   File "/usr/lib/python2.7/dist-packages/pykka/actor.py", line 201, in _actor_loop
Mar 30 16:01:46 raspberrypi mopidy[461]:     response = self._handle_receive(message)
Mar 30 16:01:46 raspberrypi mopidy[461]:   File "/usr/lib/python2.7/dist-packages/pykka/actor.py", line 295, in _handle_receive
Mar 30 16:01:46 raspberrypi mopidy[461]:     return callee(*message['args'], **message['kwargs'])
Mar 30 16:01:46 raspberrypi mopidy[461]:   File "/usr/local/lib/python2.7/dist-packages/Mopidy_Spotify-3.1.0-py2.7.egg/mopidy_s
Mar 30 16:01:46 raspberrypi mopidy[461]:     uri)
Mar 30 16:01:46 raspberrypi mopidy[461]:   File "/usr/local/lib/python2.7/dist-packages/Mopidy_Spotify-3.1.0-py2.7.egg/mopidy_s
Mar 30 16:01:46 raspberrypi mopidy[461]:     return lookup_playlist(session, web_session, config, uri)
Mar 30 16:01:46 raspberrypi mopidy[461]:   File "/usr/local/lib/python2.7/dist-packages/Mopidy_Spotify-3.1.0-py2.7.egg/mopidy_s
Mar 30 16:01:46 raspberrypi mopidy[461]:     raise spotify.Error('Playlist Web API lookup failed')
Mar 30 16:01:46 raspberrypi mopidy[461]: Error: Playlist Web API lookup failed
lines 1-58
[... cut ...]
Mar 30 16:13:30 raspberrypi mopidy[461]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 16:13:36 raspberrypi mopidy[461]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 16:13:42 raspberrypi mopidy[461]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 16:13:48 raspberrypi mopidy[461]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Mar 30 16:13:54 raspberrypi mopidy[461]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
`
and a lot more of the same

johannescramer, 2020-03-30:
looks like an authentication error...
what files need my clientid/secret?

s-martin, 2020-03-30:
#852 describes a workaround you need right now to install mopidy 3.0

johannescramer, 2020-04-06:
@s-martin thanks a lot. This worked. Sorry, for the late reply. I destroyed my config file and had to rebuild it, but now it works!
Cheers!

s-martin, 2020-04-06:
> @s-martin thanks a lot. This worked. Sorry, for the late reply. I destroyed my config file and had to rebuild it, but now it works!
> Cheers!

Great! If it works for you, you could close the issue.

closed 2020-04-07 by johannescramer