MiczFlor/RPi-Jukebox-RFID #881: 🐛 | getVolume returns 0 at random

bionicTOADY, 2020-03-31:
When i increase or decrease the volume via webinterface OR gpio-wired button the getVolume method returns 0 instead of the actual volume level.

debugging modded:
[image]

log:
[image]

Since I am not very skilled at python, shell-scripts etc. I hope I'll get some help here.

Software and Hardware:
Phoniebox-Version: 2.0 rc7
Raspi 4 2gb
Soundcard: Behringer U-Control UCA202 USB Audio Interface 2-in/2-out

Thanks for suggestions and help.

bionicTOADY, 2020-04-04:
Hi there,
i reinastalled my box since my child's birthday is some days down the road...
I evaluated, that the error occures after I got the startupsound to work with the instructiond found here:
https://forum-raspberrypi.de/forum/thread/45333-phoniebox-kein-startup-sound-nach-update/

Now I can have a startup sound OR a working volume control =(. I have not found a solution which got both to work. I hope someone may help me.

MiczFlor, 2020-04-08:
If you change the sound in the web app, does the volume change at all?
It might be the case that you have not added the soundcard to the conf file?
Please have a look here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#audio-is-not-working
under the second subheader: **b) Audio worked, then you changed the sound card and it stopped working**
This is for the bare bones Phoniebox classic without spotify. 
If you use the *+Spotify* edition, you might need to change the mopidy conf

bionicTOADY, 2020-04-09:
The volume changes just as I want. The steps are the preset steps from the settings and everything works just finde apart from setting the volumelevel to 0. This happens randomly by changing the volumelevel. The source of the change is indifferent (web app or physical button).

bionicTOADY, 2020-04-09:
Since the debugmessage:

`    getvolume)
        # read volume in percent
 if [ "${DEBUG_playout_controls_sh}" == "TRUE" ]; then echo "getVolumeMyDebug" >> $PATHDATA/../logs/debug.log; fi
`

does not return a printout when getVolume fails I think the method is not called correctly when this problem occures.
Quite difficult for me to see why this happens. Any sugestion?

bionicTOADY, 2020-04-10:
```
            # $VOLFILE does NOT exist == audio on
            # read volume in percent
if [ "${DEBUG_playout_controls_sh}" == "TRUE" ]; then echo "volfile does not exist" >> $PATHDATA/../logs/debug.log; fi
            VOLPERCENT=$(echo -e status\\nclose | nc -w 1  localhost 6600 | grep -o -P '(?<=volume: ).*')
```

this line causes the trouble. But i have no clue how to fix it....

s-martin, 2020-04-10:
@fredg02, people with bash skills needed here :)

s-martin, 2021-01-05:
@bionicTOADY do you still have this issue?

s-martin, 2022-12-27:
Can you check, if the issue still persists with current 2.4 version?

s-martin, 2023-05-05:
Closing for now. Feel free to reopen this issue, if the problem persists

closed 2023-05-05 by s-martin