MiczFlor/RPi-Jukebox-RFID #883: 🚀 | Raspotify (spotify-connect integration)

quottax, 2020-03-31:
## Feature Description
I would like to connect Spotify by using spotify-connect (Raspotify) to my Phoniebox

### What functionality would you like to see in your phoniebox?

I would like to have the feature to run "[Raspotify](https://github.com/dtcooper/raspotify)" on the Phoniebox.

### How do you envision the feature to work from a users perspective?

Would like to connect from my mobile phone with spotify app and with spotify-connect with the phoniebox.

### Further information that might help

I tried to install raspotify and it works, but then the mopidy/Phoniebox doesnt work anymore. So it just works one or the other, not in harmony.
Once one of both is running and playing music I’m not able to start playing the other one. In order to play music using Spotify Connect I need to manually stop the mopidy server and vice versa.

ddbkva, 2020-04-03:
i`m running raspotify on my phoniebox with a raspberry zero w. Its working well. I`m running too this shairport, so the box is an wlan speaker too. 

https://github.com/dtcooper/raspotify
https://github.com/mikebrady/shairport-sync

quottax, 2020-04-04:
But how you do that? If i install raspotify it blocks the RPi-Jukebox-Software or it doesnt work when playing music over the MPD.

veloxidSchweiz, 2020-04-04:
Hi @quottax ,
I am playing around with raspotify as well, using mopidy. But what i realized is that i need to stop the music. it is not enough to pause the song.... Otherwise it gets the error that the device is still busy...
Maybe one could add a command ```mpc stop``` somehow to raspotify.

quottax, 2020-04-05:
Now i have a solution for the blocked audio output!

In the file /etc/default/raspotify i added this line:

`OPTIONS="-v --device hw:sndrpihifiberry hw:0 --onevent /home/pi/setaudiodevice.sh"`

File setaudiodevice.sh:
`#!/bin/bash
if [ "$PLAYER_EVENT" = "start" ]; then
  mpc stop
elif [ "$PLAYER_EVENT" = "stop" ]; then
  mpc stop
fi`

Now i can play spotify over raspotify even its running. I just need the other way around (Raspotify playing, Jukebox starting).

And the Volume-GPIO-Buttons dont work with Raspotify. I will look for a solution for this too.

Piehti, 2020-05-09:
Worked out of the box on my one. Got one Box at the Bed to play sleepmusic only via spotify connect. :) Also RaspiCast and stuff wasnt making too much trouble. Youll propably have to restart MPD "now and then" in ya customs. 

Cant help with GPIO though. Ain't using it.

veloxidSchweiz, 2020-05-10:
I tried to get a gpio control posibility via gpio buttons. There is an option using spotipy, creating a control app. (It is also not very easy to setup the system the first time if you are not used to it).
I will look a little bit more into it.  But need to think how it is made that it does not interfear with the main  funtionality...

s-martin, 2020-05-17:
Could you explain a little more what you expect?

Do you want that functionality in addition to the current Spotify integration?

What’s the difference between raspotify and the current integration (I don’t use Spotify, so I need some more info :) )?

Groovylein, 2021-04-02:
I would like to have it in the current Spotify integration.
This would enable to see the Phoniebox as a device in the Spotify app, like a Google Home. 

I will try to install it on my test setup and see if I can somehow integrate it into the current installation process.

Groovylein, 2021-08-05:
It does not look like there is a possibility with raspotify :(
[https://github.com/librespot-org/librespot/discussions/830#discussioncomment-1125550](discussion)

So I will try
[https://github.com/Spotifyd/spotifyd](spotifyd)
let's see how I can stop and restart the daemon properly so that the playback is not doubled.

pabera, 2021-08-05:
`Spotifyd` will work on your Pi but independently from the Phoniebox software / Mopidy I believe. Not sure if they will sync as their are different approaches on how to play Spotify music.