MiczFlor/RPi-Jukebox-RFID #887: 🐛 | Some Audiofiles don't work, and make that the hole folder/phoniebox don't play anymore

Neo778, 2020-04-07:
## Bug

Hello alltogether,
at first, thank you for this great project. And sorry for my english. :-) This is my first Phoniebox, and my first Raspberry/Linux project. So a big thank you for all the good work with the manual, wiki and the youtube videos! 

Everything works great. But I have one problem. If I add this youtube links, the files don't start to play and the hole folder with a file in it which worked before don't start to play anymore, too. In the actual installation all other file work anymore until I delete the "problematic" files again.
This are the links/files which causes the problems:
https://www.youtube.com/watch?v=4m1EFMoRFvY
https://www.youtube.com/watch?v=gwA9GfYIYus
https://www.youtube.com/watch?v=0OWj0CiM8WU

It happens too, if I download the audio via the Android App NewPipe and put it via the Web App in the folders. Although when i rename them to files without special symbols in it.

Can someone give me a hint, what to do? 

I tried some issues from this board, which I though have similar problems. But no one worked.

Of course I can post some log files, but I don't know which may help you and where to find them. Please tell me the command or location, then I give my best. :-)

### What I did

I installed buster lite via headless ssh, because I have no Screen and Keyboard. I tried the one-line-skript with and without spotify. And the developer-script with spotify. Same problem.

### What happened

As written before.

### I expected this to happen

A wonderful working phoniebox. :-)

### Further information that might help

Radiostreams are working fine. I haven't tried out spotify, because I have no premium account. And this two youtube-links working too:
https://www.youtube.com/watch?v=DBOxXxdg5yQ
https://www.youtube.com/watch?v=iqND8VTQK14

## Software

### Base image and version

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Branch / Release

At the moment I have installed the normal version without spotify

### Installscript

At the moment i use the recommend one-line-skript without spotify.

## Hardware

### RaspberryPi version

Raspberry Pi 3 version B

### RFID Reader

I have the one from the shopping list.

### Soundcard

No extra soundcard. I use the sound from the Raspberry Pi.

### Other notable hardware

No other notable hardware. I use the USB-Boxes from the shopping list. I don't use Extra Buttons, only RFID-Cards for the command: Volume up/down, Play/Pause, Next/Privious Song, Shutdown.

Thank you for reading till here. I hope someone can help me.

MiczFlor, 2020-04-07:
Hi @Neo778 
not quite sure what you did with the YouTube files. Did you download them and then they won't play? Or did you add the YouTube links in a file like a playlist?
YouTube files can only be played when you downloaded the files from YouTube. You can do that in the web interface of Phoniebox.

Neo778, 2020-04-07:
Hi @MiczFlor 
yes, i downloaded them with the web app under "register new card ID"/point C. Youtube. I added the youtube link and created a new folder. As mentioned this works with some youtube links, and with some not...

MiczFlor, 2020-04-08:
And the files are in the folder you created?
It's strange that the whole phoniebox should stall because of some corrupt mp3 files. I can't see a link there. 
Did you reboot and try again? (I am sure you did :)

Neo778, 2020-04-08:
Yes, the files are in the folder. And they are showen in the web app on the main site, when i click the small arrow. Yes, i reboot. Only when I delete the files again, the phoniebox is working again.

MiczFlor, 2020-04-08:
Bizarre. 
* Did you delete and try to download them again?
* Can you play the files with another audio player?
* How big are the files?

Neo778, 2020-04-08:
1.: Yes several times.
2.: Yes, on my Android phone I can play them.
3.: They are depending on the length of the song about 1,5 MB till 4 MB big.

s-martin, 2020-04-10:
After you tried to play files, please post the output of `tail -n 500 /var/log/syslog`.

Neo778, 2020-04-10:
Hello this is the output, after adding https://www.youtube.com/watch?v=4m1EFMoRFvY in the folder Beats.

Apr 10 10:05:20 raspberrypi systemd[1]: Created slice system-bthelper.slice.
Apr 10 10:05:20 raspberrypi systemd[1]: Starting Bluetooth service...
Apr 10 10:05:20 raspberrypi bluetoothd[476]: Bluetooth daemon 5.50
Apr 10 10:05:20 raspberrypi systemd[1]: Started Bluetooth service.
Apr 10 10:05:20 raspberrypi bluetoothd[476]: Starting SDP server
Apr 10 10:05:20 raspberrypi systemd[1]: Started Raspberry Pi bluetooth helper.
Apr 10 10:05:20 raspberrypi systemd[1]: Reached target Bluetooth.
Apr 10 10:05:20 raspberrypi kernel: [   12.939301] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
Apr 10 10:05:20 raspberrypi kernel: [   12.939314] Bluetooth: BNEP filters: protocol multicast
Apr 10 10:05:20 raspberrypi kernel: [   12.939338] Bluetooth: BNEP socket layer initialized
Apr 10 10:05:20 raspberrypi bluetoothd[476]: Bluetooth management interface 1.14 initialized
Apr 10 10:05:20 raspberrypi dbus-daemon[325]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.5' (uid=0 pid=476 comm="/usr/lib/bluetooth/bluetoothd ")
Apr 10 10:05:20 raspberrypi bluetoothd[476]: Sap driver initialization failed.
Apr 10 10:05:20 raspberrypi bluetoothd[476]: sap-server: Operation not permitted (1)
Apr 10 10:05:20 raspberrypi systemd[1]: bthelper@hci0.service: Succeeded.
Apr 10 10:05:20 raspberrypi systemd[1]: Starting Hostname Service...
Apr 10 10:05:20 raspberrypi bluetoothd[476]: Failed to set privacy: Rejected (0x0b)
Apr 10 10:05:21 raspberrypi dbus-daemon[325]: [system] Successfully activated service 'org.freedesktop.hostname1'
Apr 10 10:05:21 raspberrypi systemd[1]: Started Hostname Service.
Apr 10 10:05:22 raspberrypi dhcpcd[330]: wlan0: using static address 192.168.178.37/24
Apr 10 10:05:22 raspberrypi avahi-daemon[354]: Joining mDNS multicast group on interface wlan0.IPv4 with address 192.168.178.37.
Apr 10 10:05:22 raspberrypi avahi-daemon[354]: New relevant interface wlan0.IPv4 for mDNS.
Apr 10 10:05:22 raspberrypi avahi-daemon[354]: Registering new address record for 192.168.178.37 on wlan0.IPv4.
Apr 10 10:05:22 raspberrypi dhcpcd[330]: wlan0: adding route to 192.168.178.0/24
Apr 10 10:05:22 raspberrypi dhcpcd[330]: wlan0: adding default route via 192.168.178.1
Apr 10 10:05:22 raspberrypi dhcpcd[330]: Database /etc/iproute2/rt_scopes is corrupted at iB'`P#001#0340H2g]#030s4*7ը#031͔9T$C#013#020#017(h2-#022]EMH=Ј#,˰/
Apr 10 10:05:23 raspberrypi dhcpcd[330]: forked to background, child pid 526
Apr 10 10:05:23 raspberrypi systemd[1]: Started dhcpcd on all interfaces.
Apr 10 10:05:23 raspberrypi systemd[1]: Reached target Network.
Apr 10 10:05:23 raspberrypi systemd[1]: Starting Permit User Sessions...
Apr 10 10:05:23 raspberrypi systemd[1]: Started Phoniebox Idle Watchdog Service.
Apr 10 10:05:23 raspberrypi systemd[1]: Reached target Network is Online.
Apr 10 10:05:23 raspberrypi systemd[1]: Starting Lighttpd Daemon...
Apr 10 10:05:23 raspberrypi systemd[1]: Starting LSB: exim Mail Transport Agent...
Apr 10 10:05:23 raspberrypi systemd[1]: Starting /etc/rc.local Compatibility...
Apr 10 10:05:23 raspberrypi systemd[1]: Started Phoniebox Rotary Encoder Service.
Apr 10 10:05:23 raspberrypi systemd[1]: Started Phoniebox RFID-Reader Service.
Apr 10 10:05:23 raspberrypi systemd[1]: Starting The PHP 7.3 FastCGI Process Manager...
Apr 10 10:05:23 raspberrypi systemd[1]: Started Phoniebox GPIO Buttons Service.
Apr 10 10:05:23 raspberrypi systemd[1]: Condition check resulted in fast remote file copy program daemon being skipped.
Apr 10 10:05:23 raspberrypi systemd[1]: Starting OpenBSD Secure Shell server...
Apr 10 10:05:23 raspberrypi systemd[1]: Starting Mopidy music server...
Apr 10 10:05:23 raspberrypi systemd[1]: Starting Samba NMB Daemon...
Apr 10 10:05:23 raspberrypi systemd[1]: Started Permit User Sessions.
Apr 10 10:05:23 raspberrypi systemd[1]: Started /etc/rc.local Compatibility.
Apr 10 10:05:23 raspberrypi systemd[1]: Started Mopidy music server.
Apr 10 10:05:23 raspberrypi systemd[1]: Started Getty on tty1.
Apr 10 10:05:23 raspberrypi systemd[1]: Reached target Login Prompts.
Apr 10 10:05:24 raspberrypi systemd[1]: Started OpenBSD Secure Shell server.
Apr 10 10:05:24 raspberrypi systemd[1]: Started The PHP 7.3 FastCGI Process Manager.
Apr 10 10:05:25 raspberrypi systemd[1]: Started Lighttpd Daemon.
Apr 10 10:05:25 raspberrypi systemd[1]: systemd-rfkill.service: Succeeded.
Apr 10 10:05:25 raspberrypi systemd[1]: Started Samba NMB Daemon.
Apr 10 10:05:25 raspberrypi systemd[1]: Starting Samba SMB Daemon...
Apr 10 10:05:25 raspberrypi exim4[530]: Starting MTA: exim4.
Apr 10 10:05:25 raspberrypi systemd[1]: Started LSB: exim Mail Transport Agent.
Apr 10 10:05:26 raspberrypi systemd[1]: Started Samba SMB Daemon.
Apr 10 10:05:26 raspberrypi systemd[1]: Reached target Multi-User System.
Apr 10 10:05:26 raspberrypi systemd[1]: Reached target Graphical Interface.
Apr 10 10:05:26 raspberrypi systemd[1]: Starting Update UTMP about System Runlevel Changes...
Apr 10 10:05:26 raspberrypi systemd[1]: systemd-update-utmp-runlevel.service: Succeeded.
Apr 10 10:05:26 raspberrypi systemd[1]: Started Update UTMP about System Runlevel Changes.
Apr 10 10:05:27 raspberrypi systemd[1]: Startup finished in 1.785s (kernel) + 17.534s (userspace) = 19.320s.
Apr 10 10:05:29 raspberrypi mopidy[548]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
[... cut ...]
Apr 10 10:11:42 raspberrypi mopidy[548]: INFO     [MpdSession-163] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56218
Apr 10 10:11:42 raspberrypi mopidy[548]: INFO     [MpdSession-164] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56222
Apr 10 10:11:44 raspberrypi kernel: [  385.602217] Under-voltage detected! (0x00050005)
Apr 10 10:11:44 raspberrypi mopidy[548]: INFO     [MpdSession-165] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56226
Apr 10 10:11:44 raspberrypi mopidy[548]: INFO     [MpdSession-166] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56230

s-martin, 2020-04-10:
Some findings in the log:

```
Apr 10 10:10:51 raspberrypi mopidy[548]: WARNING  [MainThread] mopidy.audio.actor Could not find a MPEG-4 AAC decoder to handle media.
Apr 10 10:10:51 raspberrypi mopidy[548]: WARNING  [MainThread] mopidy.audio.gst GStreamer warning: No decoder available for type 'audio/mpeg, mpegversion=(int)4, framed=(boolean)true, stream-format=(string)raw, level=(string)2, base-profile=(string)lc, profile=(string)lc, codec_data=(buffer)12100000000000000000000000000000, rate=(int)44100, channels=(int)2'.
Apr 10 10:10:51 raspberrypi mopidy[548]: ERROR    [MainThread] mopidy.audio.gst GStreamer error: Your GStreamer installation is missing a plug-in.
```
There seems to be an issue with GStreamer.
You could try `apt-get install gstreamer1.0-plugins-bad` according to https://discourse.mopidy.com/t/missing-gstreamer-plugin-for-m4a/1486

In addition you could check: https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#fix-broken-youtube-dl-while-staying-on-same-rpi-jukebox-rfid-releaseversion

```
Apr 10 10:08:41 raspberrypi lighttpd[836]: WARNING: Writing cache to '/var/www/.cache/youtube-dl/youtube-sigfuncs/js_vflset_112.json' failed: Traceback (most recent call last):
Apr 10 10:08:41 raspberrypi lighttpd[836]:   File "/usr/local/lib/python3.7/dist-packages/youtube_dl/cache.py", line 49, in store
Apr 10 10:08:41 raspberrypi lighttpd[836]:     os.makedirs(os.path.dirname(fn))
Apr 10 10:08:41 raspberrypi lighttpd[836]:   File "/usr/lib/python3.7/os.py", line 211, in makedirs
Apr 10 10:08:41 raspberrypi lighttpd[836]:     makedirs(head, exist_ok=exist_ok)
Apr 10 10:08:41 raspberrypi lighttpd[836]:   File "/usr/lib/python3.7/os.py", line 211, in makedirs
Apr 10 10:08:41 raspberrypi lighttpd[836]:     makedirs(head, exist_ok=exist_ok)
Apr 10 10:08:41 raspberrypi lighttpd[836]:   File "/usr/lib/python3.7/os.py", line 221, in makedirs
Apr 10 10:08:41 raspberrypi lighttpd[836]:     mkdir(name, mode)
Apr 10 10:08:41 raspberrypi lighttpd[836]: PermissionError: [Errno 13] Permission denied: '/var/www/.cache'
```

The web server (user `www-data`) cannot write to this directory. You should fix the permissions of `/var/www`

The following should work:
```
sudo chown -R pi:www-data /var/www
sudo chmod -R 775 /var/www
```

Neo778, 2020-04-10:
Hello @s-martin 
thank you but it don't solve the problem. Everything like before.

The first command gives this output:
pi@raspberrypi:~ $ apt-get install gstreamer1.0-plugins-bad
E: Could not open lock file /var/lib/dpkg/lock-frontend - open (13: Permission denied)
E: Unable to acquire the dpkg frontend lock (/var/lib/dpkg/lock-frontend), are you root?

s-martin, 2020-04-10:
Sorry you need the appropriate rights, please try:
`sudo apt-get install gstreamer1.0-plugins-bad`

Neo778, 2020-04-10:
Thank you, I am a newbie to this whole raspberry world.

There is one Error shown in the process. And the problem is still there.

sudo apt-get install gstreamer1.0-plugins-bad
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  python-backports-abc python-certifi python-chardet python-concurrent.futures python-gst-1.0 python-idna
  python-openssl python-pykka python-requests python-singledispatch python-tornado python-urllib3
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  libdca0 libde265-0 libgssdp-1.0-3 libgstreamer-plugins-bad1.0-0 libgupnp-1.0-4 libgupnp-igd-1.0-4 libilmbase23
  libkate1 liblcms2-2 libmjpegutils-2.1-0 libmpeg2encpp-2.1-0 libmplex2-2.1-0 libnice10 libofa0 libopenexr23 libsbc1
  libsoundtouch1 libspandsp2 libsrtp2-1 libvo-aacenc0 libvo-amrwbenc0 libvulkan1 libwebrtc-audio-processing1 libzbar0
Suggested packages:
  frei0r-plugins liblcms2-utils
The following NEW packages will be installed:
  gstreamer1.0-plugins-bad libdca0 libde265-0 libgssdp-1.0-3 libgstreamer-plugins-bad1.0-0 libgupnp-1.0-4
  libgupnp-igd-1.0-4 libilmbase23 libkate1 liblcms2-2 libmjpegutils-2.1-0 libmpeg2encpp-2.1-0 libmplex2-2.1-0
  libnice10 libofa0 libopenexr23 libsbc1 libsoundtouch1 libspandsp2 libsrtp2-1 libvo-aacenc0 libvo-amrwbenc0
  libvulkan1 libwebrtc-audio-processing1 libzbar0
0 upgraded, 25 newly installed, 0 to remove and 0 not upgraded.
Need to get 7,332 kB of archives.
After this operation, 18.1 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf libvo-aacenc0 armhf 0.1.3-1 [71.1 kB]
Get:2 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf libvo-amrwbenc0 armhf 0.1.3-1 [67.3 kB]
Get:3 http://mirror.de.leaseweb.net/raspbian/raspbian buster/main armhf libdca0 armhf 0.0.6-1 [85.1 kB]
Get:4 http://mirror1.hs-esslingen.de/pub/Mirrors/archive.raspbian.org/raspbian buster/main armhf libde265-0 armhf 1.0.3-1+rpi1+b9 [180 kB]
Get:5 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf libgssdp-1.0-3 armhf 1.0.2-4 [22.3 kB]
Get:7 http://mirror.de.leaseweb.net/raspbian/raspbian buster/main armhf libgupnp-igd-1.0-4 armhf 0.2.5-3 [24.0 kB]
Err:6 http://ftp.agdsn.de/pub/mirrors/raspbian/raspbian buster/main armhf libgupnp-1.0-4 armhf 1.0.3-3
  Could not connect to ftp.agdsn.de:80 (141.76.119.131). - connect (113: No route to host)
Get:9 http://mirror1.hs-esslingen.de/pub/Mirrors/archive.raspbian.org/raspbian buster/main armhf libkate1 armhf 0.4.1-9 [37.0 kB]
Get:8 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf libilmbase23 armhf 2.2.1-2 [65.7 kB]
Get:10 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf liblcms2-2 armhf 2.9-3 [116 kB]
Get:11 http://mirror1.hs-esslingen.de/pub/Mirrors/archive.raspbian.org/raspbian buster/main armhf libmjpegutils-2.1-0 armhf 1:2.1.0+debian-5 [24.9 kB]
Get:12 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf libmpeg2encpp-2.1-0 armhf 1:2.1.0+debian-5 [66.0 kB]
Get:15 http://mirror.de.leaseweb.net/raspbian/raspbian buster/main armhf libofa0 armhf 0.9.3-19 [48.0 kB]
Get:17 http://mirror.de.leaseweb.net/raspbian/raspbian buster/main armhf libsbc1 armhf 1.4-1 [30.5 kB]
Get:13 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf libmplex2-2.1-0 armhf 1:2.1.0+debian-5 [45.1 kB]
Get:14 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf libnice10 armhf 0.1.14-1 [112 kB]
Get:16 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf libopenexr23 armhf 2.2.1-4.1 [519 kB]
Get:18 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf libsoundtouch1 armhf 2.1.2+ds1-1 [28.7 kB]
Get:19 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf libspandsp2 armhf 0.0.6+dfsg-2 [237 kB]
Get:20 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf libsrtp2-1 armhf 2.2.0-1 [65.0 kB]
Get:21 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf libvulkan1 armhf 1.1.97-2 [77.3 kB]
Get:22 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf libwebrtc-audio-processing1 armhf 0.3-1 [230 kB]
Get:23 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf libzbar0 armhf 0.22-1 [118 kB]
Get:24 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf libgstreamer-plugins-bad1.0-0 armhf 1.14.4-1+b1 [1,914 kB]
Get:25 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf gstreamer1.0-plugins-bad armhf 1.14.4-1+b1 [3,083 kB]
Fetched 7,267 kB in 3min 2s (39.9 kB/s)
E: Failed to fetch http://ftp.agdsn.de/pub/mirrors/raspbian/raspbian/pool/main/g/gupnp/libgupnp-1.0-4_1.0.3-3_armhf.deb  Could not connect to ftp.agdsn.de:80 (141.76.119.131). - connect (113: No route to host)
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

Neo778, 2020-04-10:
Two more things i have noticed:
1. In the Web App it says under "Info" MPD Server Status: inactive (dead)
2. And the phoniebox plays a sound when i shutdown with Card or with Web app, but it don't plays the welcome sound when I start the raspberry, like I saw in youtubevideos.
I don't know if this have something to do with this problem.

s-martin, 2020-04-10:
Please try
```
sudo apt update
sudo apt-get install gstreamer1.0-plugins-bad
```

and afterwards `sudo systemctl restart mpd` to fix the dead mpd

Neo778, 2020-04-10:
The problem is solved!!! Thank you very much!!! 

Only in the Web App it says still MPD Server Status: inactive (dead)

This is the output of the mpd command you posted.

pi@raspberrypi:~ $ sudo systemctl restart mpd
Job for mpd.service failed because the control process exited with error code.
See "systemctl status mpd.service" and "journalctl -xe" for details.

s-martin, 2020-04-10:
Oh, I remember you are using the Spotify edition. If I remember correctly then the MPD status in the webapp is always dead, because you don't use mpd, but mopidy

s-martin, 2020-04-10:
If it's now working please close the issue.

Neo778, 2020-04-10:
No, at the moment I use classic version without spotify.

closed 2020-04-10 by Neo778