MiczFlor/RPi-Jukebox-RFID #889: WebUI Player - Repeat-Button malfunction in 2.0RC7 (List / Off)

c4t0r, 2020-04-10:
## Bug
WebUI Player - Repeat-Button malfunction in 2.0RC7 (List/Off seem to be mixed up)
### What I did
Pushed the repeat button on the WebUI until "Off" is showing

### What happened
After a reload of the page (no matter if using F5 or waiting for refresh) repeat is set to "List". Repeat mode is acutally "List".

### I expected this to happen
Repeat keeps the state "Off" and behave like "Repeat=Off"

### Further information that might help
"List" and "Off" seem to be mixed somewhere in the frontend. Since I see a
`function execSuccessfully: /home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=playerprev -c=playerrepeat -v=off`
in the debug.log when setting to "List" in WebUI and 
`function execSuccessfully: /home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=playerprev -c=playerrepeat -v=playlist`
when setting to "Off"

## Software

### Base image and version
`2020-02-13-raspbian-buster-lite.img`

### Branch / Release
`master`

### Installscript
`scripts/installscripts/buster-install-default.sh`

## Hardware

### RaspberryPi version
`Raspberry Pi 3 Model B Rev 1.2`

### RFID Reader
`16c0:27db Van Ooijen Technische Informatica Keyboard`

### Soundcard
OnBoard

### Other notable hardware

nothing yet

s-martin, 2020-04-10:
I can confirm, issue still exists on `develop` branch.

c4t0r, 2020-04-13:
Did a small change on my machine in `htdocs/inc.loadControls.php` and "fixed" lines 40-42 (repeat cycle) assuming a desired order of "Off->Playlist->Single->Off" by changing the "id" and the "onclick" action.

`print '<a id="repeatPlaylist" onclick="repeatSingle();" class="btn  btn-lg" title="' . $lang['playerLoop'] . ': ' . $lang['globalList'] . '" style="display: none"><i class="mdi mdi-repeat"></i> ' . $lang['globalList'] . '</a>';`
`print '<a id="repeatSingle" onclick="repeatOff();" class="btn  btn-lg" title="' . $lang['playerLoop'] . ": " . $lang['globalTrack'] . '" style="display: none"><i class="mdi mdi-repeat-once"></i> ' . $lang['globalTrack'] . '</a>';`
`print '<a id="repeatOff" onclick="repeatPlaylist();" class="btn btn-lg" title="' . $lang['playerLoop'] . ': ' . $lang['globalOff'] . '" style="display: none"><i class="mdi mdi-repeat-off"></i> ' . $lang['globalOff'] . '</a>';`

Now a least the immediate behaviour is as expected but after a reload the shown icon is wrong (Off<->Playlist).
Unfortunately my web coding got a bit rusty but I will stick on this issue - maybe I can find a solution.

s-martin, 2020-04-16:
Thanks for fixing this.

Could you open a pull request? 

If you have problems, please ping me and I could open one for you.

s-martin, 2020-04-18:
@c4t0r, I have opened a PR #909  to incorporate your proposed changes.

Could you review the PR, if the changes are correct?

c4t0r, 2020-04-19:
@s-martin, thanks for creating a pull request for this. I currently have no IDE running for this. It was just a live fix on my "production" system.
I reviewed the PR and it matches the changes I have done. Thank you for your work on this great project!

closed 2020-05-02 by MiczFlor