MiczFlor/RPi-Jukebox-RFID #890: Release 2.0

s-martin, 2020-04-10:
This is an issue to coordinate all necessary steps to release [version 2.0](https://github.com/MiczFlor/RPi-Jukebox-RFID/milestone/1).

Some PRs and issues for 2.0 are still open and should be fixed, before 2.0 can be released.

- [ ] Fix all [issues for 2.0](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues?q=is%3Aissue+is%3Aopen+milestone%3A2.0)
- [ ] Merge all [PRs for 2.0](https://github.com/MiczFlor/RPi-Jukebox-RFID/pulls?q=is%3Apr+is%3Aopen+milestone%3A2.0)
- [ ] ....

topas-rec, 2020-04-16:
If you plan a release candidate (which I want to suggest here) I would like to test the led functionality from #387 which seems to have issues with stable branch.

s-martin, 2020-05-19:
From my point of view we could think about releasing 2.0 and merge develop to master.
Some minor documentation issues are open, but that should be easy.

I also suggest to make master the default branch again, so Readme from master is visible when visiting GitHub

closed 2020-05-24 by s-martin