MiczFlor/RPi-Jukebox-RFID #892: Unnecessary wget in install script

Viatorus, 2020-04-11:
I don't get it why we need this inside the install script:

https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/0027f433a35599f76080a099b50fc7494e3e47f8/scripts/installscripts/buster-install-default.sh#L581-L586

Is this redundant?

fredg02, 2020-04-11:
I have the same question. If there is a good reason for it, a short comment should be added that explains it.

s-martin, 2020-04-11:
I don't know, but maybe this was necessary when the repo wasn't checked out at that point.
For me it looks redundant now, too.

@MiczFlor, do you know why this is still there?

MiczFlor, 2020-05-02:
I commented them out. Let's see if somebody complains :)

The good reason was, I believe, that when mopidy created trouble, @splitti made a hotfix for master, calling some files from the `develop` branch. I am 99% sure that these lines made it into the `develop` install via copy and paste and there they are not needed. For the reasons stated here.
If my theory is correct, we can delete the lines alltogether and close this issue. Any thoughts?

closed 2020-05-12 by MiczFlor