MiczFlor/RPi-Jukebox-RFID #894: 🐛 | Add Iris to sudoers during install

JuliH29, 2020-04-12:
Hi!

Coding is really not part of my expertise, however I might have found something that's missing in the install scripts.
After having installed phoniebox via the developer script, Iris is installed in version 3.45.
However under settings it advertises the update to 3.46. However the update never seemed to be completed successfully.
Via `tail -n 500 /var/log/syslog` I found this:
`ERROR mopidy_iris.system Password-less access to file:///usr/local/lib/python3.7/dist-packages/mopidy_iris/system.sh was refused. Check your /etc/sudoers file.`

which lead me to [this discussion](https://github.com/jaedb/Iris/issues/509) and [this wiki entry](https://github.com/jaedb/Iris/wiki/Getting-started#installing).

May you want to add
`sudo sh -c 'echo "mopidy ALL=NOPASSWD: /usr/local/lib/python3.7/dist-packages/mopidy_iris/system.sh" >> /etc/sudoers'`
to the install script to allow Iris to be easily updated?

s-martin, 2020-04-12:
We currently fix the version to 3.45 (see https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/requirements-spotify.txt), because in the past we had issue with different versions behaving differently and 3.45 seemed to work for all users.

So this version constraint could be relaxed in that file, if a new version would be necessary.

JuliH29, 2020-04-12:
OK, thanks for the clarification.

closed 2020-04-12 by JuliH29

s-martin, 2020-04-12:
If you want to update Iris manually (I guess you try to fix #893), you could use
`sudo python3 -m pip install Mopidy-Iris`