MiczFlor/RPi-Jukebox-RFID #896: 🐛 | BUG SUMMARY

Hamsert, 2020-04-13:
## Bug

### What I did

When I want to skip to track 11 in a folder and I press the button 10 times I end up at track 5 (on the old Phoniebox version every button press was recognized no matter how fast and often a button was pressed). Same thing happens when pressing the volume button multiple times.

### What happened

Button pressed 10 times, 4 presses recognized

### I expected this to happen

10/10 recognition

## Software

### Base image and version

Buster

### Branch / Release

Development

## Hardware

### RaspberryPi version

3 B

### Other notable hardware

RPi-Jukebox-RFID/scripts/gpio-buttons.py (nothing changed by me):

#!/usr/bin/python3
import RPi.GPIO as GPIO
from signal import pause
from subprocess import check_call
import time
from time import sleep

# This script will block any I2S DAC e.g. from Hifiberry, Justboom, ES9023, PCM5102A
# due to the assignment of GPIO 19 and 21 to a buttons
#
# 2019-12-17
# Added support to detect RFID tag removal. Requires an appropriate switch
# makes no difference which hardware implementation you choose, could be a simple switch,
# or hidden magnetic switch or hacking the rfid reader as suggested here:
# https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/62
#
# 2019-05-14
# * Rewrite using RPi.GPIO for Raspberry pi zero compatibility.
# * Added power LED functionality
#
# 2018-10-31
# Added the function on holding volume + - buttons to change the volume in 0.3s interval
#
# 2018-10-15
# this script has the `pull_up=True` for all pins. See the following link for additional info:
# https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/259#issuecomment-430007446
#
# 2017-12-12
# This script was copied from the following RPi forum post:
# https://forum-raspberrypi.de/forum/thread/13144-projekt-jukebox4kids-jukebox-fuer-kinder/?postID=312257#post312257
# I have not yet had the time to test is, so I placed it in the misc folder.
# If anybody has ideas or tests or experience regarding this solution, please create pull requests or contact me.

# This function takes a holding time (fractional seconds), a channel, a GPIO state and an action reference (function).
# It checks if the GPIO is in the state since the function was called. If the state
# changes it return False. If the time is over the function returns True.
def checkGpioStaysInState(holdingTime, gpioChannel, gpioHoldingState):
    # Get a reference start time (https://docs.python.org/3/library/time.html#time.perf_counter)
    startTime = time.perf_counter()
    # Continously check if time is not over
    while(holdingTime >= (time.perf_counter() - startTime)):
        # Return if state does not match holding state
        if(gpioHoldingState != GPIO.input(gpioChannel)):
           return False
        # Else: Wait
    return True
# Actions that call other processes of the phoniebox (Channel Parameter needed by RPi.GPIO)
def shutdown_action(channel):
    check_call("./scripts/playout_controls.sh -c=shutdown", shell=True)
def vol0_action(channel):
    check_call("./scripts/playout_controls.sh -c=mute", shell=True)
def volD_action(channel):
    check_call("./scripts/playout_controls.sh -c=volumedown", shell=True)
def volU_action(channel):
    check_call("./scripts/playout_controls.sh -c=volumeup", shell=True)
def next_action(channel):
    check_call("./scripts/playout_controls.sh -c=playernext", shell=True)
def prev_action(channel):
    check_call("./scripts/playout_controls.sh -c=playerprev", shell=True)
def halt_action(channel):
    check_call("./scripts/playout_controls.sh -c=playerpause", shell=True)
def recordstart_action(channel):
    check_call("./scripts/playout_controls.sh -c=recordstart", shell=True)
def recordstop_action(channel):
    check_call("./scripts/playout_controls.sh -c=recordstop", shell=True)
def recordplaylatest_action(channel):
    check_call("./scripts/playout_controls.sh -c=recordplaylatest", shell=True)
def rfidoff_action():
    check_call("./scripts/playout_controls.sh -c=playerpauseforce", shell=True)

# Handlers that handle special behavior of the push of a button
# When the shutdown button was held for more than 2 seconds LED flashed and afterwards
def shutdown_handler(channel):
    # Detect holding of button
    if True == checkGpioStaysInState(shutdownHoldTime, btn_shut, GPIO.LOW):
        # Blink LED
        for x in range(0, 10):
            GPIO.output(led_power, x & 1)
            sleep(PledBlinkTime)
        # Keep LED on until power off
        GPIO.output(led_power, GPIO.HIGH)
        # Shutdown afterwards
        shutdown_action(channel)
# When the Volume Down button was held for more than 0.3 seconds every 0.3 seconds he will lower t$
def volU_handler(channel):
    # Rise volume as requested
    volU_action(channel)
    # Detect holding of button
    while checkGpioStaysInState(volumeHoldTime, btn_volU, GPIO.LOW):
        volU_action(channel)

# When the Volume Up button was held for more than 0.3 seconds every 0.3 seconds he will call a ra$
def volD_handler(channel):
    # Rise volume as requested
    volD_action(channel)
    # Detect holding of button
    while checkGpioStaysInState(volumeHoldTime, btn_volD, GPIO.LOW):
        volD_action(channel)

# Define the used pins of the raspberry board
btn_shut = 3
btn_vol0 = 13
btn_volU = 16
btn_volD = 19
btn_next = 26
btn_prev = 20
btn_halt = 21
led_power = 12
#reco =
#play =

# Set GPIO numbering to BCM instead of board numbering
GPIO.setmode(GPIO.BCM)
# Bounce tolerance time for buttons
bouncetime = 500
volumeHoldTime = 0.3 # Seconds
shutdownHoldTime = 2 # Seconds
PledBlinkTime = 0.3  # Seconds

# Set up GPIO pins and the power led
GPIO.setup(btn_shut       , GPIO.IN, pull_up_down=GPIO.PUD_UP)
GPIO.setup(btn_vol0       , GPIO.IN, pull_up_down=GPIO.PUD_UP)
GPIO.setup(btn_volU       , GPIO.IN, pull_up_down=GPIO.PUD_UP)
GPIO.setup(btn_volD       , GPIO.IN, pull_up_down=GPIO.PUD_UP)
GPIO.setup(btn_next       , GPIO.IN, pull_up_down=GPIO.PUD_UP)
GPIO.setup(btn_prev       , GPIO.IN, pull_up_down=GPIO.PUD_UP)
GPIO.setup(btn_halt       , GPIO.IN, pull_up_down=GPIO.PUD_UP)
#GPIO.setup(btn_rfidoff, GPIO.IN, pull_up_down=GPIO.PUD_UP)
GPIO.setup(led_power      , GPIO.OUT)
# btn_rfidoff = Button(12,pull_up=True)

# Set standard events for the buttons. Callback functions define the actions of the events (THey are defined above)
GPIO.add_event_detect(btn_shut, GPIO.FALLING, callback=shutdown_handler, bouncetime=bouncetime)
GPIO.add_event_detect(btn_vol0, GPIO.FALLING, callback=vol0_action, bouncetime=bouncetime)
GPIO.add_event_detect(btn_volU, GPIO.FALLING, callback=volU_handler, bouncetime=bouncetime)
GPIO.add_event_detect(btn_volD, GPIO.FALLING, callback=volD_handler, bouncetime=bouncetime)
GPIO.add_event_detect(btn_next, GPIO.FALLING, callback=next_action, bouncetime=bouncetime)
GPIO.add_event_detect(btn_prev, GPIO.FALLING, callback=prev_action, bouncetime=bouncetime)
GPIO.add_event_detect(btn_halt, GPIO.FALLING, callback=halt_action, bouncetime=bouncetime)
#reco.when_pressed = recordstart_action
#reco.when_released = recordstop_action
#play.when_pressed = recordplaylatest_action
#GPIO.add_event_detect(btn_rfidoff, GPIO.FALLING, callback=rfidoff_action, bouncetime=bouncetime)

# Switch on power led after boot to indicate state "on" for phoniebox
GPIO.output(led_power, GPIO.HIGH)

pause()

s-martin, 2020-04-14:
Do you recognize any difference, if you press the buttons faster or slower? Did you try to press them super slow, i.e. waiting couple of seconds between each push?

Could be a also hardware issue.

I have acarde buttons, which don't always recognize the pressing of the button, if you don't press the button deep enough. But this can also happen, if you press the button just once.

Hamsert, 2020-04-14:
Yes there is a difference if I press the buttons slower. If I press and wait for the new Song to start and then press again the button presses are always recognized.
I doubt it's a hardware issue since as I said the behavior was different on the first version of phoniebox. I also tried again and made sure that I fully pressed the button.

"#!/usr/bin/python3"
This is quoted out in the script, is this how it's supposed to be?

s-martin, 2020-04-14:
> Yes there is a difference if I press the buttons slower. If I press and wait for the new Song to start and then press again the button presses are always recognized.

Ok, so I think that's a timing issue. We should look into this, but I don't think we will have a quick fix.

> "#!/usr/bin/python3"
> This is quoted out in the script, is this how it's supposed to be?

Yeah, that's a shebang and supposed to be like that.

Hamsert, 2020-05-30:
Hi, thanks for the release of Version 2! I checked on this issue and unfortunately it persists.

s-martin, 2020-05-31:
Maybe you could play around with the value
`bouncetime = 500`
in the gpio-buttons.py file.

I think the value is in ms, so you could try if maybe 100 works better. Or also try 700 or 800.

Hamsert, 2020-05-31:
Thanks for the recommendation, I set it to 200 and it seems a lot better now.

closed 2020-05-31 by MiczFlor