MiczFlor/RPi-Jukebox-RFID #898: Question to Mopidy / Spotify version base function

domu83, 2020-04-14:
As far as I understand and experience there is no playback possible, when using the spotify version, until Mopidy is connected to Spotify.
Is this observation correct and working as designed?

s-martin, 2020-04-15:
I would expect that local files could be played even when there's no connection to Spotify.

Issue #893 may be related to your problem.

kingosticks, 2020-04-15:
That is correct. Mopidy will not respond to commands until the backends have started. If a backend is taking a long time to start (e.g. Mopidy-Spotify is refreshing playlists) then you have to wait for that to finish before you can do anything, even play local files. Most people, who use the Mopidy service and have it start automatically on boot, will find Mopidy is ready by the time the rest of their system becomes ready for use. However, if you don't use the Mopidy service or if your system otherwise comes up very quickly, the delay is very noticeable. I do agree that isn't very good and needs improving.

In the meantime as a workaround, if you exclusively control Mopidy using Iris then you can disable playlists in Mopidy-Spotify by setting `spotify/allow_playlists = False` in your Mopidy config. You can then have Iris fetch your Spotify playlists directly from Spotify's servers so you'll still have them available. Alternatively, if you have some sort of offline-only setup then you can (and probably should) entirely disable Mopidy-Spotify in your config file.

closed 2020-12-15 by s-martin