MiczFlor/RPi-Jukebox-RFID #899: web interface loads very slow🐛 | BUG SUMMARY

TrustCompact, 2020-04-14:
## Bug
The web interface takes huge amount of time for loading. I'm talking about intervals greater than 80 seconds.
When loaded, usability is ok. Not fast but ok.
Everything else works fine.
Ping of IP works as intended and in normal time.

### What I did

### I expected this to happen

### Further information that might help

## Software

### Base image and version

### Branch / Release

### Installscript

## Hardware

### RaspberryPi version

### RFID Reader

### Soundcard

### Other notable hardware

s-martin, 2020-04-14:
Hi, could you add some of the info the bug template asked for?
This would help a lot. Thanks!

TrustCompact, 2020-04-14:
> Hi, could you add some of the info the bug template asked for?
> This would help a lot. Thanks!

Totally sorry for the missing details. I put them into the OP, but forgot to remove the commenting-tags and therefore these details were not shown :( Fixed this.

s-martin, 2020-04-14:
What’s the output of `journalctl -u lighttpd`?

TrustCompact, 2020-04-14:
> What’s the output of `journalctl -u lighttpd`?

`journalctl -u lighttpd
-- Logs begin at Tue 2020-04-14 20:17:01 CEST, end at Tue 2020-04-14 20:30:08 CEST. --
Apr 14 20:17:10 Phoniebox systemd[1]: Starting Lighttpd Daemon...
Apr 14 20:17:12 Phoniebox systemd[1]: Started Lighttpd Daemon.
Apr 14 20:17:26 Phoniebox sudo[959]: www-data : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/htdocs/api ; USER=root ; COMMAND=/bin/echo status
                                           currentsong
                                           close
Apr 14 20:17:26 Phoniebox sudo[959]: pam_unix(sudo:session): session opened for user root by (uid=0)
Apr 14 20:17:26 Phoniebox sudo[959]: pam_unix(sudo:session): session closed for user root
Apr 14 20:17:27 Phoniebox sudo[970]: www-data : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/htdocs/api ; USER=root ; COMMAND=/bin/echo playlistinfo
                                           close
Apr 14 20:17:27 Phoniebox sudo[970]: pam_unix(sudo:session): session opened for user root by (uid=0)
Apr 14 20:17:27 Phoniebox sudo[970]: pam_unix(sudo:session): session closed for user root
Apr 14 20:17:36 Phoniebox sudo[981]: www-data : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/htdocs/api ; USER=root ; COMMAND=/bin/echo status
                                           currentsong
                                           close
Apr 14 20:17:36 Phoniebox sudo[981]: pam_unix(sudo:session): session opened for user root by (uid=0)
Apr 14 20:17:36 Phoniebox lighttpd[904]: echo: write error: Broken pipe
Apr 14 20:17:36 Phoniebox sudo[981]: pam_unix(sudo:session): session closed for user root
Apr 14 20:28:37 Phoniebox sudo[992]: www-data : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/htdocs/api ; USER=root ; COMMAND=/bin/echo status
                                           currentsong
                                           close
Apr 14 20:28:37 Phoniebox sudo[992]: pam_unix(sudo:session): session opened for user root by (uid=0)
Apr 14 20:28:37 Phoniebox sudo[992]: pam_unix(sudo:session): session closed for user root
lines 1-23...skipping...
-- Logs begin at Tue 2020-04-14 20:17:01 CEST, end at Tue 2020-04-14 20:30:08 CEST. --
Apr 14 20:17:10 Phoniebox systemd[1]: Starting Lighttpd Daemon...
Apr 14 20:17:12 Phoniebox systemd[1]: Started Lighttpd Daemon.
Apr 14 20:17:26 Phoniebox sudo[959]: www-data : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/htdocs/api ; USER=root ; COMMAND=/bin/echo status
                                           currentsong
                                           close
Apr 14 20:17:26 Phoniebox sudo[959]: pam_unix(sudo:session): session opened for user root by (uid=0)
Apr 14 20:17:26 Phoniebox sudo[959]: pam_unix(sudo:session): session closed for user root
Apr 14 20:17:27 Phoniebox sudo[970]: www-data : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/htdocs/api ; USER=root ; COMMAND=/bin/echo playlistinfo
                                           close
Apr 14 20:17:27 Phoniebox sudo[970]: pam_unix(sudo:session): session opened for user root by (uid=0)
Apr 14 20:17:27 Phoniebox sudo[970]: pam_unix(sudo:session): session closed for user root
Apr 14 20:17:36 Phoniebox sudo[981]: www-data : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/htdocs/api ; USER=root ; COMMAND=/bin/echo status
                                           currentsong
                                           close
Apr 14 20:17:36 Phoniebox sudo[981]: pam_unix(sudo:session): session opened for user root by (uid=0)
Apr 14 20:17:36 Phoniebox lighttpd[904]: echo: write error: Broken pipe
Apr 14 20:17:36 Phoniebox sudo[981]: pam_unix(sudo:session): session closed for user root
Apr 14 20:28:37 Phoniebox sudo[992]: www-data : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/htdocs/api ; USER=root ; COMMAND=/bin/echo status
                                           currentsong
                                           close
Apr 14 20:28:37 Phoniebox sudo[992]: pam_unix(sudo:session): session opened for user root by (uid=0)
Apr 14 20:28:37 Phoniebox sudo[992]: pam_unix(sudo:session): session closed for user root
Apr 14 20:28:44 Phoniebox sudo[1005]: www-data : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/htdocs/api ; USER=root ; COMMAND=/bin/echo status
                                            currentsong
                                            close
Apr 14 20:28:44 Phoniebox sudo[1005]: pam_unix(sudo:session): session opened for user root by (uid=0)
Apr 14 20:28:44 Phoniebox sudo[1005]: pam_unix(sudo:session): session closed for user root
Apr 14 20:28:44 Phoniebox sudo[1016]: www-data : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/htdocs/api ; USER=root ; COMMAND=/bin/echo playlistinfo
                                            close
Apr 14 20:28:44 Phoniebox sudo[1016]: pam_unix(sudo:session): session opened for user root by (uid=0)
Apr 14 20:28:44 Phoniebox sudo[1016]: pam_unix(sudo:session): session closed for user root
Apr 14 20:28:50 Phoniebox sudo[1027]: www-data : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/htdocs/api ; USER=root ; COMMAND=/bin/echo status
                                            currentsong
                                            close
Apr 14 20:28:50 Phoniebox sudo[1027]: pam_unix(sudo:session): session opened for user root by (uid=0)
Apr 14 20:28:50 Phoniebox sudo[1027]: pam_unix(sudo:session): session closed for user root
Apr 14 20:28:56 Phoniebox sudo[1039]: www-data : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/htdocs/api ; USER=root ; COMMAND=/bin/echo status
[... cut ...]
Apr 14 20:29:03 Phoniebox sudo[1050]: pam_unix(sudo:session): session closed for user root
` 

> Did you try to reboot?

Sure - several times. No difference.
It even makes no difference from which machine I open the web interface. Mobile, Desktop, Notebook - also tried different browsers.
 
> And what’s the output of `journalctl -u mpd`?

`journalctl -u mpd
-- Logs begin at Tue 2020-04-14 20:17:01 CEST, end at Tue 2020-04-14 20:31:29 CEST. --
Apr 14 20:17:10 Phoniebox systemd[1]: Starting Music Player Daemon...
Apr 14 20:28:40 Phoniebox systemd[1]: Started Music Player Daemon.
`

TrustCompact, 2020-04-15:
Well, I found the problem.
I had mounted two network folders from OMV via etc/fstab to avoid having file duplicates.
Removing these mounted folders gets the normal speed of the web interface back.

Anyway it would be nice to be able to use mounted folders without having these preformance problems. Access to the mounted folders via console or file explorer works in normal speed.

Or are there any other ways to achieve using network files?

Thanks!