MiczFlor/RPi-Jukebox-RFID #900: 🐛 | BUG SUMMARY rotary encoder on developer

narf0815, 2020-04-14:
## Bug

### What I did

as the status LED wasn't introduced to stable release yet i tried the developer branch and i tried to install the rotary-encoder followig these steps: https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Audio-RotaryKnobVolume
### What happened
on the last step
`sudo chmod +x /home/pi/RPi-Jukebox-RFID/scripts/rotary-encoder.py`
tells me the file does not exist.

i found the file rotary_encoder.py in the folder ~/RPi-Jukebox-RFID/components/gpio_control/GPIODevices and tried to copy the file to /home/pi/RPi-Jukebox-RFID/scripts/ but the rotary encoder still didn't work

last thing i tried was to change the ExecStart in phoniebox-rotary-encoder.service to "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/rotary_encoder.py"
but this didn't work either...

same wiring and gpio definitions work fine with stable.

s-martin, 2020-04-14:
@veloxidSchweiz, can you help here?

veloxidSchweiz, 2020-04-16:
Hey,
sorry for the late answer.
Maybe #906 would be a possibile solution for you.
You would need to run the ./install.sh script in the components/gpio_control folder

narf0815, 2020-04-16:
```
pi@PhonieBerry:~/RPi-Jukebox-RFID/components/gpio_control $ ./install.sh
Installing GPIO_Control service

Please choose a default configuration
1) ./example_configs/gpio_settings_test.ini
2) ./example_configs/gpio_settings.ini
3) Stop the script
#? 2
Configuration  file ./example_configs/gpio_settings.ini selected
Copy file to /home/pi/.config/phoniebox/gpio_settings.ini
cp -v ./example_configs/gpio_settings.ini /home/pi/.config/phoniebox/gpio_settin                                                                                                                                                             gs.ini
'./example_configs/gpio_settings.ini' -> '/home/pi/.config/phoniebox/gpio_settin                                                                                                                                                             gs.ini'
cp: cannot create regular file '/home/pi/.config/phoniebox/gpio_settings.ini': N                                                                                                                                                             o such file or directory

Installing GPIO_Control service, this will require to enter your password up to                                                                                                                                                              3 times to enable the service
Press enter to continue
'./example_configs/phoniebox_gpio_control.service' -> '/etc/systemd/system/phoni                                                                                                                                                             ebox_gpio_control.service'
systemctl start phoniebox_gpio_control.service
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ===
Authentication is required to start 'phoniebox_gpio_control.service'.
Authenticating as: root
Password:
polkit-agent-helper-1: pam_authenticate failed: Authentication failure
==== AUTHENTICATION FAILED ===
Failed to start phoniebox_gpio_control.service: Access denied
See system logs and 'systemctl status phoniebox_gpio_control.service' for detail                                                                                                                                                             s.
systemctl enable phoniebox_gpio_control.service
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-unit-files ===
Authentication is required to manage system service or unit files.
Authenticating as: root
Password:
polkit-agent-helper-1: pam_authenticate failed: Authentication failure
==== AUTHENTICATION FAILED ===
Failed to enable unit: Access denied
● phoniebox_gpio_control.service - Phoniebox GPIO Control Service
   Loaded: loaded (/etc/systemd/system/phoniebox_gpio_control.service; disabled;
   Active: inactive (dead)
```
authentication was not wrong, tried a second time:
```
pi@PhonieBerry:~/RPi-Jukebox-RFID/components/gpio_control $ ./install.sh
Installing GPIO_Control service

Please choose a default configuration
1) ./example_configs/gpio_settings_test.ini
2) ./example_configs/gpio_settings.ini
3) Stop the script
#? 2
Configuration  file ./example_configs/gpio_settings.ini selected
Copy file to /home/pi/.config/phoniebox/gpio_settings.ini
cp -v ./example_configs/gpio_settings.ini /home/pi/.config/phoniebox/gpio_settin                                                                                                                                                             gs.ini
'./example_configs/gpio_settings.ini' -> '/home/pi/.config/phoniebox/gpio_settin                                                                                                                                                             gs.ini'
cp: cannot create regular file '/home/pi/.config/phoniebox/gpio_settings.ini': N                                                                                                                                                             o such file or directory

Installing GPIO_Control service, this will require to enter your password up to                                                                                                                                                              3 times to enable the service
Press enter to continue
/etc/systemd/system/phoniebox_gpio_control.service exists.
Press enter to continue
● phoniebox_gpio_control.service - Phoniebox GPIO Control Service
   Loaded: loaded (/etc/systemd/system/phoniebox_gpio_control.service; disabled;
   Active: inactive (dead)
```
didn't even ask for auth on second run.

s-martin, 2020-04-16:
You need to run the script with sudo:
`sudo install.sh`

narf0815, 2020-04-16:
i already tried to run as root:
```
root@PhonieBerry:/home/pi/RPi-Jukebox-RFID/components/gpio_control# ./install.sh
Installing GPIO_Control service

Please choose a default configuration
1) ./example_configs/gpio_settings_test.ini
2) ./example_configs/gpio_settings.ini
3) Stop the script
#? 2
Configuration  file ./example_configs/gpio_settings.ini selected
Copy file to /root/.config/phoniebox/gpio_settings.ini
cp -v ./example_configs/gpio_settings.ini /root/.config/phoniebox/gpio_settings.ini
'./example_configs/gpio_settings.ini' -> '/root/.config/phoniebox/gpio_settings.ini'
cp: cannot create regular file '/root/.config/phoniebox/gpio_settings.ini': No such file or directory

Installing GPIO_Control service, this will require to enter your password up to 3 times to enable the service
Press enter to continue
/etc/systemd/system/phoniebox_gpio_control.service exists.
Press enter to continue
● phoniebox_gpio_control.service - Phoniebox GPIO Control Service
   Loaded: loaded (/etc/systemd/system/phoniebox_gpio_control.service; disabled; vendor preset: enabled)
   Active: inactive (dead)
```
"sudo install.sh" with user pi didn't work
neither did "sudo ./install.sh":
```
Installing GPIO_Control service

./install.sh: 13: ./install.sh: Syntax error: redirection unexpected
```

s-martin, 2020-04-17:
@veloxidSchweiz provided a possible fix in #906

veloxidSchweiz, 2020-04-17:
Hi,
Hi Thank you for the infos. I tried to improve the script such  that it is more bullet proof. it is up to now only tested on my pc.
the script should not be run as root as usually the file owner is wrong afterwards. sorry for not clarifying that. I added a check in the install script. 
For debugging you can run the component locally to see the problems:
`cd /home/pi/RPi-Jukebox-RFID/components/gpio_control;
python gpio_control.py`
Please check and give feedback.

veloxidSchweiz, 2020-04-17:
At the end you will need to configure the config file correctly:
the file can be found in
`/home/pi/.config/phoniebox/gpio_control.ini`
I haven't written a manual for configuration yet. Which  gpio devices do you have connected to which pin?

veloxidSchweiz, 2020-04-17:
And for debugging it could be helpful to check the output of
`systemctl list-units phoniebox*`
and 
`systemctl list-unit-files phoniebox*`
this shows you all (running) services which are there...

narf0815, 2020-04-17:
GPIOS:
_Buttons:_
shut = Button(**3**, hold_time=2)
next = Button(**26**,pull_up=True)
prev = Button(**20**,pull_up=True)
halt = Button(**23**,pull_up=True)

_Rotary encoder:_
**5
6**

_MiniAmp:_
**18
19
21**

"Buttons" are configured in gpio-buttons.py
Rotary encoder via rotary-encoder.py (? - just followed steps from [wiki](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Audio-RotaryKnobVolume))
MiniAmp without any additional configuration (as explained [here](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/HiFiBerry-Soundcard-Details))

i'll try with a clean develop setup after work: 1) configuring hifiberry > 2) run one-line install script > then what? first install.sh or first activating rotary encoder?

veloxidSchweiz, 2020-04-17:
Hi,
The component gpio_control is a replacement of ALL gpio input/output devices, you cannot have roatary-encoder nor gpio_buttons.py running in parallel.
Please check that these services are not running when you use the gpio_control service instead.
I wrote a small script which checks the installation. Check my PR.

The gpio_settings.ini  (/home/pi/.config/phoniebox/gpio_settings.ini) should look like this:
`[DEFAULT]
enabled: True
used_pinds = [13]

[VolumeControl]
enabled: True
Type: RotaryEncoder
PinUp: 5
PinDown: 6
pull_up: True
hold_time: 0.3
hold_repeat: True
timeBase: 0.2
# only for rotary encoder
functionCallDown: functionCallVolD
functionCallUp: functionCallVolU
functionCallTwoButtons: functionCallStop
#only for TwoButtonControl
functionCallButton: functionCallPlayerPause 
# only for RotaryEncoderClickable

[PrevNextControl]
enabled: True
Type: TwoButtonControl
Pin1: 20
Pin2: 26
functionCall1: functionCallPlayerPrev
functionCall2: functionCallPlayerNext
functionCallTwoButtons: functionCallPlayerStop
pull_up: True
hold_time: 0.3
hold_repeat: False

[Shutdown]
enabled: True
Type:  Button
Pin: 3
hold_time: 2
functionCall: functionCallShutdown

[PlayPause]
enable: True
Type: Button
Pin: 23
pull_up: True
functionCall: functionCallPlayerPause

[StatusLED]
enable: True
Type: MPDStatusLED
Pin: 12`

narf0815, 2020-04-17:
hm okay.
1) configuring hifiberry > 2) run one-line **develop** install script > configuring gpio_settings.ini (nothing to configure in gpio_control.py, right?) ?

you're referring to [this script](https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/906/commits/63d39f6bba81971258d1dde6dcf50c2d76b832a2), right?

veloxidSchweiz, 2020-04-17:
I am not 100percent sure what happens  when you install the one-line develop install script.
Is there actually the develop branch checked out?
I couldn't check that yet, that's why the original PR was originally only meant to discuss details....
you need to have the  the folder `/home/pi/RPi-Jukebox-RFID/components/gpio_control`
Does that exist?
If yes you can run the install script of the gpio_control by:
```
cd /home/pi/RPi-Jukebox-RFID/components/gpio_control;
./install.sh
#HERE You will be asked to install the config and the corresponding service
./check_installation.sh #<-- THis will show you if installation of gpio_controls worked
```
if check_installationsh fails,you need to resolve the issue (stopping & disabling the corresponding services...)
Afterwards edit the gpio_settings.ini File  which is in the folder ```/home/pi/.config/phoniebox/gpio_settings.ini```
If you have changed the file you will need to restart the gpio_control service 

You don't need to configure anything in the gpio_control.py
You should not have any other gpio_button service or gpio-rotary-encoder running it is all handled by 
```
systemctl restart phoniebox_gpio_control.service

```
you can check everything again, by runnig the check_installation.py or 
`systemctl status phoniebox_gpio_control.service`

s-martin, 2020-04-17:
To install from develop branch, see here: https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/INSTALL-stretch#b-developer-install-script-for-buster

But this doesn't include the changes from the PR (yet).

veloxidSchweiz, 2020-04-17:
Thanks,
Could I create a branch on the main git repo not using a fork? Then It would be accessable using explicilty the corresponding branch ...

s-martin, 2020-04-17:
> Thanks,
> Could I create a branch on the main git repo not using a fork? Then It would be accessable using explicilty the corresponding branch ...

@MiczFlor, could you elevate @veloxidSchweiz to Collaborator?

narf0815, 2020-04-17:
@veloxidSchweiz i'll wait in the meantime.
please let me know when/how to proceed so i can test LED + rotary!

Benzinbruder, 2020-04-19:
I'm stuck with the very same problem as mentioned in the OP by narf0815.
Did a fresh install with the one-line script (developer branch, as described in [the wiki](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/INSTALL-stretch#b-developer-install-script-for-buster)).

closed 2020-04-20 by s-martin

s-martin, 2020-04-20:
Issues should be fixed with merged PR #906

narf0815, 2020-04-21:
@veloxidSchweiz i'm ready when you are!
i have prepared a fresh buster with miniamp config...

Benzinbruder, 2020-04-21:
I too would be willing an eager to test.
I have tried a new install wieder the one-line-script and the develop branch but could not get the rotary encoder to work. It seems as if, by default, the gpio_control service is not enabled.

By the way: I have tried installing it with `./install.sh` in `/home/pi/RPi-Jukebox-RFID/components/gpio_control`, and while the installation now seems to be starting as intended, authentication always fails for me (also see https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/900#issuecomment-614880750 above).

s-martin, 2020-04-21:
> By the way: I have tried installing it with `./install.sh` in `/home/pi/RPi-Jukebox-RFID/components/gpio_control`, and while the installation now seems to be starting as intended, authentication always fails for me 

Please use `sudo install.sh`

Benzinbruder, 2020-04-21:
I did initially, but that did not work either. Installation wouldn't start.
That's why I used `./install.sh` as a 'fallback' method.

narf0815, 2020-04-24:
so... still the same?

veloxidSchweiz, 2020-04-24:
Hi,
I actually do not understand why there is this issue with the authentication.
It should work as a pi user.
I hope that i find time this weekend to setup a new rpi this weekend.
COuld you maybe test it not using the graphical ui? i.e. change to on of the other virtual console?

Benzinbruder, 2020-04-24:
JFYI, I was running my Pi in headless mode - so no GUI involved.
I have run into another issue (possibly hardware related) though, so I am sorry to say that I can not promise to be of much help for further testing at the moment.

veloxidSchweiz, 2020-04-26:
Hey @narf0815, @Benzinbruder 
I did a fresh installation of the Phoniebox using the current develop branch and for me everything is more or less working. I cannot reproduce the error with the authentication and actually do not have any idea where this kind of error could come from....
I used the install.sh in components/gpio_control to install the new service and used check_installation.sh to check everything.
I actually got the information that the phoniebox-gpio-buttons.service was still running:
```checking phoniebox-gpio-buttons.service
 **phoniebox-gpio-buttons.service is still active**
checking phoniebox-rotary-encoder.service
Phoniebox GPIO Service started correctly .....

PROBLEM IN INSTALLATION
```
So this makes sense as the phoniebox-gpio-buttons.service is installed and enabled automatically during the installation process. therefore this behaviour is to be expected.
You need to stop and disable the service by:
```systemctl stop phoniebox-gpio-buttons.service
systemctl disable phoniebox-gpio-buttons.service
```
after doing this one get's the following information 
 ```checking phoniebox-gpio-buttons.service
checking phoniebox-rotary-encoder.service
Phoniebox GPIO Service started correctly .....

Installation seems to be fine
```
By running the command `systemctl status phoniebox_gpio_control.service`
you can see if the gpio_control is running and some last outputs.
if you need more information run
`journalctl -fu phoniebox_gpio_control.service`

don't hesitate to ask additional questions...

narf0815, 2020-04-26:
no idea what i'm doing wrong. nothing changes :(
just made a fresh install with one line develop install, then this (code isn't parsed correctly with ``): https://pastebin.com/sAFKCR4S

narf0815, 2020-04-27:
i tried the single steps from install.sh:

`pi@raspberrypi:~/RPi-Jukebox-RFID/components/gpio_control $ cp -v example_configs/gpio_settings_rotary_and_led.ini ~/.config/phoniebox/g                                                                                                     pio_settings.ini
'example_configs/gpio_settings_rotary_and_led.ini' -> '/home/pi/.config/phoniebox/gpio_settings.ini'
`

ok

`pi@raspberrypi:~/RPi-Jukebox-RFID/components/gpio_control $ systemctl restart phoniebox_gpio_control.service
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ===
Authentication is required to restart phoniebox_gpio_control.service.
Authenticating as: root
Password:
polkit-agent-helper-1: pam_authenticate failed: Authentication failure
==== AUTHENTICATION FAILED ===
Failed to restart phoniebox_gpio_control.service: Access denied
See system logs and systemctl status phoniebox_gpio_control.service for details.
`

not ok

`pi@raspberrypi:/var/log $ sudo systemctl restart phoniebox_gpio_control.service
`

went through without any (error) message

`pi@raspberrypi:~/RPi-Jukebox-RFID/components/gpio_control $ ./check_installation.sh
checking phoniebox-gpio-buttons.service
 phoniebox-gpio-buttons.service is still active
checking phoniebox-rotary-encoder.service

Problem during installation occured
   Service not running, please check functionallity by running gpio_control.py
   in the directory ~/RPi-Jukebox-RFID/components/gpio_control:
      $ cd ~/RPi-Jukebox-RFID/components/gpio_control
      $ python gpio_control.py
   or check output of journaclctl by:
      $ journalctl -u phoniebox_gpio_control.service -f
`
hmm, k

`pi@raspberrypi:~/RPi-Jukebox-RFID/components/gpio_control $ journalctl -u phoniebox_gpio_control.service -f
-- Logs begin at Thu 2019-02-14 10:11:59 GMT. --
Apr 27 20:28:03 raspberrypi PhonieboxGPIOControl[1359]: Traceback (most recent call last):
Apr 27 20:28:03 raspberrypi PhonieboxGPIOControl[1359]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py", line 6, in <module>
Apr 27 20:28:03 raspberrypi PhonieboxGPIOControl[1359]:     from GPIODevices import *
Apr 27 20:28:03 raspberrypi PhonieboxGPIOControl[1359]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/__init__.py", line 7, in <module>
Apr 27 20:28:03 raspberrypi PhonieboxGPIOControl[1359]:     from .led import *
Apr 27 20:28:03 raspberrypi PhonieboxGPIOControl[1359]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/led.py", line 4, in <module>
Apr 27 20:28:03 raspberrypi PhonieboxGPIOControl[1359]:     import mpd
Apr 27 20:28:03 raspberrypi PhonieboxGPIOControl[1359]: ModuleNotFoundError: No module named 'mpd'
Apr 27 20:28:03 raspberrypi systemd[1]: phoniebox_gpio_control.service: Main process exited, code=exited, status=1/FAILURE
Apr 27 20:28:03 raspberrypi systemd[1]: phoniebox_gpio_control.service: Failed with result 'exit-code'.
Apr 27 20:28:06 raspberrypi systemd[1]: phoniebox_gpio_control.service: Service RestartSec=3s expired, scheduling restart.
Apr 27 20:28:06 raspberrypi systemd[1]: phoniebox_gpio_control.service: Scheduled restart job, restart counter is at 84.
Apr 27 20:28:06 raspberrypi systemd[1]: Stopped Phoniebox GPIO Control Service.
Apr 27 20:28:06 raspberrypi systemd[1]: Started Phoniebox GPIO Control Service.
`

`pi@raspberrypi:~/RPi-Jukebox-RFID/components/gpio_control $ python gpio_control.py
Traceback (most recent call last):
  File "gpio_control.py", line 6, in <module>
    from GPIODevices import *
  File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/__init__.py", line 7, in <module>
    from .led import *
  File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/led.py", line 4, in <module>
    import mpd
ModuleNotFoundError: No module named 'mpd'
`

(no idea why the part in the middle is not displayed as code...)

maybe i should mention that i did NOT choose to install spotify in the one-line-setup...

honestly... i don't even see where checking phoniebox-rotary-encoder.service should be enabled in install.sh (but that's probably just me not understanding things...)
@veloxidSchweiz any help would be welcome :)

s-martin, 2020-04-27:
You need to install the Python mpd package, maybe @veloxidSchweiz can help which package exactly.

narf0815, 2020-04-27:
oh. that might explain why it's not working. i started with buster-LITE, and according to [this site](https://www.raspberrypi.org/documentation/linux/software/python.md), pip doesn't come with lite version...
also @Benzinbruder states that he has a headless setup so maybe he has the same problem?

s-martin, 2020-04-28:
You need to execute `systemctl` **always** with `sudo`.

After `disable` you need to restart or use `sudo systemctl stop phoniebox-gpio-buttons`.
`disable` only disables the service when booting. Please disable first, then stop.

Please make also sure you installed the mpd dependency also with `sudo`, so 
`sudo python3 -m pip install -r requirements.txt` (make sure you use the correct path for the requirements.txt file).

narf0815, 2020-04-28:
i tried with sudo, but after that, ssh console loses connection (tried more times) and on reboot the service is still active.

i don't understand why i (and Benzinbruder) have auth issues while @veloxidSchweiz has no problem with ./install.sh...

i'm trying again with a fresh install (non-lite). maybe we're still missing some packages?

s-martin, 2020-04-28:
If you loose SSH connection I think something else could be broken, too.

Raspian Lite should be fine, but a fresh install is maybe a good idea.

narf0815, 2020-04-28:
now i have massive problems with one-line-installer (several "lost connections").
(since yesterday my test-raspberry is a rasp3 - my son is still using the rasp2 on master)

narf0815, 2020-04-28:
ok. tried with pi-zero, buster (not lite!) and develop install and it worked
while with lite auth was not possible (could NOT choose between users pi and root) it worked with full buster image (and choosing pi+pw).

finally:

`checking phoniebox-gpio-buttons.service
checking phoniebox-rotary-encoder.service
Phoniebox GPIO Service started correctly

Installation seems to be fine`

(but why is code not displayed as "code"?!)

no i'm trying with my raspberry3 (zero is very slow...)

thanks for your help @s-martin & @veloxidSchweiz 
@Benzinbruder give it a try with full image of raspian-buster!

Benzinbruder, 2020-04-28:
Thanks so much for the update!
I can confirm that I was using buster lite as well for my setup. So it seems indeed likely that this was the problem.

I have two running boxes now on release, and at the moment I do not really like to work with them. My other Pi seems to be fried (my fault) but if I can get it running again, I will try to reproduce your solution.

veloxidSchweiz, 2020-04-28:
Hi,
Sorry,
I always tried the full not the lite. 
I will try lite some time in the future.
but great to hear that you got it runnning.

@narf0815 : What do you mean with 

> why is code not displayed as "code"?!)

narf0815, 2020-04-30:
installed on (another) pi3 yesterday and rotary + buttons work perfectly (LED not connected yet).

> why is code not displayed as "code"?!)

usually if i put a text between two of these guy: ` it's displayes like this:
`this`
but sometimes it doesn't work (see my previous post, lines after "finally:")

s-martin, 2020-04-30:
If you have multiple lines you need 3 ` at the start and 3 at the end (in it's own line).

veloxidSchweiz, 2020-05-02:
Hey,
Thanks all for the feedback. And good to hear that it is working. I hope you see an advantage in having just one script.... 
I will try Buster Lite at some point. I do not have worked with it and it looks somehow like a more general problem when using sudo there

thiesschneider, 2020-05-24:
I am also using Light. Any progress with this?

veloxidSchweiz, 2020-05-24:
I am sorry at the moment i do not have much time to work with the rpi and could not try to check what is the cause for this (for me really strange) behaviour.
If anyone could try to fix or find a way to reproduce it in a docker that would be great..

thiesschneider, 2020-05-24:
That would be cool. I am tryinig to get the rotary knob to work but I cannot use GPIO 5 and 6 for the rotary knob, because I use the Hiifiberry and the onoffshim.

Benzinbruder, 2020-05-27:
I too am using a HiFiBerry (MiniAmp) and a Pimoroni OnOff SHIM. Neither of these devices interferes with GPIOs 5 and 6. Do you use a different HiFiBerry amp?

thiesschneider, 2020-05-27:
Perhaps I miscounted... I will have a look later. 
Nevertheless I do not know, where to change the used GPIO for the rotary knob and it would be nice to be able to use this file which is handling all of it.
But if they are free I will start with the existing one :)

narf0815, 2020-05-29:
@thiesschneider look at this post https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/900#issuecomment-615140282
these are the settings that work for me with OnOff SHIM, rotary knob & Hifiberry. you can also see in that post where to edit GPIO.
at the moment i am NOT using raspian lite!

thiesschneider, 2020-05-31:
Nice, thank you! Hope this feature will soon also work with the lite version, too. At the moment I am using the time to prepare the box design :)

Joemoe134, 2020-08-07:
I am not sure if this fits here, but after dealing with the setup of GPIO_control I finally have a running service and a running gpio_buttons.py script. Turning the knob results in input read by the script, however, here I get the following error messages:
```
INFO:GPIODevices.rotary_encoder:RotaryVolumeControl: Calling functionIncr 1
expr: syntax error: unexpected argument '3'
../../scripts/playout_controls.sh: line 238: [: -le: unary operator expected
OK MPD 0.21.4
OK
INFO:GPIODevices.rotary_encoder:RotaryVolumeControl: Calling functionDecr 1
expr: syntax error: unexpected argument '('
../../scripts/playout_controls.sh: line 276: [: -ge: unary operator expected
OK MPD 0.21.4
OK
```

I adjusted in rotary_encoder.py on line 142
```
    pin1 = int(input('please enter first pin'))
    pin2 = int(input('please enter second pin'))
```
to my GPIO 5 and 6 between ' ' but this shouldnt be causing this? 

any pointers as to what is going on? I am a bit lost.
Thanks a lot in advance.

Benzinbruder, 2020-08-13:
What does your code look like exactly?

`pin1 = int(input(5)`

or

`pin1 = int(input('5')`

Joemoe134, 2020-08-14:
I used `pin1 = int(input('5')` 

I am at this point not sure if this has to do with that script or if it might be an issue with the other scripts involved.
As the https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Audio-RotaryKnobVolume refers to scripts that are not available on github I used wget to download ky040.py from here: https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/267
 
ky040.py.txt
rotary-encoder.py.txt
rotary-encoder.service.txt

Not sure where else I would find this script in the most up to date version?
The rotary encoder worked for me to control alsamixer volume on a different installation using mopidy and pummeluff, but as there were to many things that needed setting up i wanted to go for a preset install like RPI_musicbox.

Benzinbruder, 2020-08-14:
I am afraid I can't help you out here.
However, it seems as if you were using scripts from previous versions that should not / can no longer be used with 2.0.

You might want to switch to using gpio_control instead.
Have a look [here](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/components/gpio_control/README.md).

I have found a few similar threads and obviously the integration of a rotary switch is still tricky in 2.0.
If you find a solution, I am sure it'd be worth posting it.

veloxidSchweiz, 2020-08-15:
Hi,
Sorry for the late answer. 
The link of @Benzinbruder  is right. It should be done only with the component gpio_control. The separate files mentioned in the wiki are not right anymore. I will update this probably some time next week. Hope that i ll find tie again after summer holiday time... 
@Joemoe134 : Could you pleas pinpoint me to the exact version you used? I ll try to replicate the error....
Could you also tell me what is the output  when you check the installation:
seeL https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/900#issuecomment-615238433