MiczFlor/RPi-Jukebox-RFID #901: Spotify doesn't wortk (any more)

Chapuisat, 2020-04-14:
Hi all, I had a good working Phoniebox for 1,5 years. In the last weeks all spotify playlists stopped working but streaming radio channels worked.

Today I reset the entire box to RPi buster and did a fresh installation from scratch using the one-line-install script. Everything works fine, soundcard, buttons, RFI card reader. But I still cannot play any song or playlist from spotify. What happened? Can anybody support?

I'm running buster on RPi Jukebox 2.0 64a42f1 develop. Connected my account several times already with mopidy and spotify. I'm pretty much sure that all: user, pass, client id and secret are correct. I have a premium family account. 

Any ideas?

Chapuisat, 2020-04-14:
That's really strange, now I was able to start one playlist by coincidence. But after rebooting the box it doesn't work any more. Mopidy seems to be quite buggy. It tells me it's connected but I cannot start any playlist any more neither from Phoniebox nor directly from Mopidy (Mopidy: Failed to add some tracks).

s-martin, 2020-04-15:
What's the output of `journalctl -u mopidy`?

Did you try `mopidy local scan`?

Chapuisat, 2020-04-15:
Here is the output of `journalctl -u mopidy` directly after reboot: 
```
pi@raspberrypi:~ $ journalctl -u mopidy
-- Logs begin at Thu 2019-02-14 11:11:59 CET, end at Wed 2020-04-15 10:20:57 CEST. --
Apr 15 09:54:22 raspberrypi systemd[1]: Starting Mopidy music server...
Apr 15 09:54:23 raspberrypi systemd[1]: Started Mopidy music server.
Apr 15 09:54:29 raspberrypi mopidy[450]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
Apr 15 09:54:29 raspberrypi mopidy[450]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Apr 15 09:54:29 raspberrypi mopidy[450]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Apr 15 09:54:29 raspberrypi mopidy[450]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Apr 15 09:54:29 raspberrypi mopidy[450]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Apr 15 09:54:29 raspberrypi mopidy[450]: INFO     [MainThread] mopidy.config Loading config from command line options
Apr 15 09:54:29 raspberrypi mopidy[450]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
Apr 15 09:54:30 raspberrypi mopidy[450]: INFO     [MainThread] mopidy.__main__ Enabled extensions: stream, mpd, softwaremixer, file, iris, http, local, spotify, m3u
Apr 15 09:54:30 raspberrypi mopidy[450]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Apr 15 09:54:32 raspberrypi mopidy[450]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Apr 15 09:54:32 raspberrypi mopidy[450]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Apr 15 09:54:32 raspberrypi mopidy[450]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Apr 15 09:54:32 raspberrypi mopidy[450]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, SpotifyBackend, LocalBacke
Apr 15 09:54:32 raspberrypi mopidy[450]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Apr 15 09:54:33 raspberrypi mopidy[450]: ERROR    [SpotifyEventLoop] spotify.session Spotify login error: <ErrorType.UNABLE_TO_CONTACT_SERVER: 8>
Apr 15 09:54:35 raspberrypi mopidy[450]: INFO     [SpotifyBackend-6] mopidy_spotify.web Logged into Spotify Web API as kfuzej1pajj038nhge0w1af92
Apr 15 10:20:53 raspberrypi mopidy[450]: INFO     [SpotifyBackend-6] mopidy_spotify.playlists Refreshed 18 Spotify playlists
Apr 15 10:20:53 raspberrypi mopidy[450]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Apr 15 10:20:53 raspberrypi mopidy[450]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: IrisFrontend, HttpFrontend, MpdFrontend
Apr 15 10:20:53 raspberrypi mopidy[450]: INFO     [IrisFrontend-11] mopidy_iris.core Starting Iris 3.45.0
Apr 15 10:20:53 raspberrypi mopidy[450]: INFO     [HttpFrontend-13] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Apr 15 10:20:53 raspberrypi mopidy[450]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Apr 15 10:20:53 raspberrypi mopidy[450]: INFO     [MainThread] mopidy.commands Starting GLib mainloop

```

Chapuisat, 2020-04-15:
and output of `mopidy local scan`

```
pi@raspberrypi:~ $ mopidy local scan
INFO     2020-04-15 10:23:49,422 [1082:MainThread] mopidy.__main__
  Starting Mopidy 3.0.2
INFO     2020-04-15 10:23:49,576 [1082:MainThread] mopidy.config
  Loading config from builtin defaults
INFO     2020-04-15 10:23:49,583 [1082:MainThread] mopidy.config
  Loading config from file:///home/pi/.config/mopidy/mopidy.conf
INFO     2020-04-15 10:23:49,590 [1082:MainThread] mopidy.config
  Loading config from command line options
WARNING  2020-04-15 10:23:49,606 [1082:MainThread] mopidy.config
  Ignoring config section 'local-images' because no matching extension was found
INFO     2020-04-15 10:23:50,136 [1082:MainThread] mopidy.__main__
  Enabled extensions: local, softwaremixer, stream, file, m3u, iris, mpd, spotify, http
INFO     2020-04-15 10:23:50,136 [1082:MainThread] mopidy.__main__
  Disabled extensions: none
INFO     2020-04-15 10:23:50,742 [1082:MainThread] mopidy_local.commands
  Finding files in file:///home/pi/RPi-Jukebox-RFID/shared/audiofolders ...
INFO     2020-04-15 10:23:50,774 [1082:MainThread] mopidy_local.commands
  Found 6 files in file:///home/pi/RPi-Jukebox-RFID/shared/audiofolders
INFO     2020-04-15 10:23:50,799 [1082:MainThread] mopidy_local.schema
  Creating SQLite database schema v6
INFO     2020-04-15 10:23:50,839 [1082:MainThread] mopidy_local.commands
  Checking 0 tracks from library
INFO     2020-04-15 10:23:50,841 [1082:MainThread] mopidy_local.commands
  Removing 0 missing tracks
INFO     2020-04-15 10:23:50,844 [1082:MainThread] mopidy_local.commands
  Found 1 tracks which need to be updated
INFO     2020-04-15 10:23:50,844 [1082:MainThread] mopidy_local.commands
  Scanning...
WARNING  2020-04-15 10:23:50,862 [1082:MainThread] mopidy_local.commands
  Failed scanning file:///home/pi/RPi-Jukebox-RFID/shared/audiofolders/placeholder: gst-stream-error-quark: Stream contains no data. (4)
INFO     2020-04-15 10:23:50,863 [1082:MainThread] mopidy_local.commands
  Scanned 1 of 1 files in 0.018s.
INFO     2020-04-15 10:23:50,864 [1082:MainThread] mopidy_local.commands
  Done scanning
INFO     2020-04-15 10:23:50,883 [1082:MainThread] mopidy_local.storage
  Cleaning up image directory
```

Chapuisat, 2020-04-15:
OK, restarted mopidy again, and executed `journalctl -u mopidy` again

```
Apr 15 10:27:14 raspberrypi mopidy[450]: INFO     [MainThread] mopidy.commands GLib mainloop got SIGTERM. Exiting...
Apr 15 10:27:14 raspberrypi systemd[1]: Stopping Mopidy music server...
Apr 15 10:27:14 raspberrypi mopidy[450]: INFO     [MainThread] mopidy.commands Stopping Mopidy frontends
Apr 15 10:27:14 raspberrypi mopidy[450]: INFO     [IrisFrontend-11] mopidy_iris.core Stopping Iris
Apr 15 10:27:14 raspberrypi mopidy[450]: INFO     [MainThread] mopidy.commands Stopping Mopidy core
Apr 15 10:27:14 raspberrypi mopidy[450]: INFO     [MainThread] mopidy.commands Stopping Mopidy backends
Apr 15 10:27:14 raspberrypi mopidy[450]: INFO     [MainThread] mopidy.commands Stopping Mopidy audio
Apr 15 10:27:14 raspberrypi mopidy[450]: INFO     [MainThread] mopidy.commands Stopping Mopidy mixer
Apr 15 10:27:15 raspberrypi systemd[1]: mopidy.service: Succeeded.
Apr 15 10:27:15 raspberrypi systemd[1]: Stopped Mopidy music server.
Apr 15 10:27:19 raspberrypi systemd[1]: Starting Mopidy music server...
Apr 15 10:27:19 raspberrypi systemd[1]: Started Mopidy music server.
Apr 15 10:27:20 raspberrypi mopidy[1570]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
Apr 15 10:27:20 raspberrypi mopidy[1570]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Apr 15 10:27:20 raspberrypi mopidy[1570]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Apr 15 10:27:20 raspberrypi mopidy[1570]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Apr 15 10:27:20 raspberrypi mopidy[1570]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Apr 15 10:27:20 raspberrypi mopidy[1570]: INFO     [MainThread] mopidy.config Loading config from command line options
Apr 15 10:27:20 raspberrypi mopidy[1570]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
Apr 15 10:27:21 raspberrypi mopidy[1570]: INFO     [MainThread] mopidy.__main__ Enabled extensions: softwaremixer, file, stream, mpd, spotify, iris, m3u, local, http
Apr 15 10:27:21 raspberrypi mopidy[1570]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Apr 15 10:27:21 raspberrypi mopidy[1570]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Apr 15 10:27:21 raspberrypi mopidy[1570]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Apr 15 10:27:21 raspberrypi mopidy[1570]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Apr 15 10:27:21 raspberrypi mopidy[1570]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, SpotifyBackend, LocalBack
Apr 15 10:27:22 raspberrypi mopidy[1570]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Apr 15 10:27:22 raspberrypi mopidy[1570]: INFO     [SpotifyBackend-6] mopidy_spotify.web Logged into Spotify Web API as kfuzej1pajj038nhge0w1af92
Apr 15 10:27:22 raspberrypi mopidy[1570]: INFO     [SpotifyEventLoop] mopidy_spotify.backend Logged in to Spotify in online mode
Apr 15 10:27:56 raspberrypi mopidy[1570]: INFO     [SpotifyBackend-6] mopidy_spotify.playlists Refreshed 18 Spotify playlists
Apr 15 10:27:56 raspberrypi mopidy[1570]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Apr 15 10:27:56 raspberrypi mopidy[1570]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: IrisFrontend, HttpFrontend, MpdFrontend
Apr 15 10:27:56 raspberrypi mopidy[1570]: INFO     [IrisFrontend-11] mopidy_iris.core Starting Iris 3.45.0
Apr 15 10:27:56 raspberrypi mopidy[1570]: INFO     [HttpFrontend-13] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Apr 15 10:27:56 raspberrypi mopidy[1570]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Apr 15 10:27:56 raspberrypi mopidy[1570]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
```

What I can see is that mopidy was now able to login to Spotify:

**before:**
`Apr 15 09:54:33 raspberrypi mopidy[450]: ERROR    [SpotifyEventLoop] spotify.session Spotify login error: <ErrorType.UNABLE_TO_CONTACT_SERVER: 8>`

**after restart of mopid:**
`Apr 15 10:27:22 raspberrypi mopidy[1570]: INFO     [SpotifyBackend-6] mopidy_spotify.web Logged into Spotify Web API as kfuzej1pajj038nhge0w1af92`

Why does it not login directly after the booting procedure? To early maybe, before WiFi connection?

s-martin, 2020-04-15:
Ok, seems to be the same as #893.

I agree, that the problem could be that a internet connection is not available yet, but that doesn't explain why this suddenly doesn't work anymore.

s-martin, 2020-04-15:
Closing as duplicate, please see #893

closed 2020-04-15 by s-martin