MiczFlor/RPi-Jukebox-RFID #908: GMusic Installation

comdoxx, 2020-04-17:
## Bug

### What I did

I installed the software by the one line installer and wanted to add GMusic functionality according to the document [https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Enable-Google-Play-Music-GMusic].

### What happened

When entering the line `sudo pip install pyasn1==0.4.1`, the error `pyasn1-modules 0.2.8 has requirement pyasn1<0.5.0,>=0.4.6, but you'll have pyasn1 0.4.1 which is incompatible.` raised. 

### I expected this to happen

Have a working documentation with working version when a specific version is specified. 

### Further information that might help

The following command `sudo easy_install pyasn1==0.4.1` also raised an error: easy_install could not be found. That might be a separate defect.

## Software

### Base image and version

Todays one line installer (stable)

## Hardware

Pi 4B/2GB

s-martin, 2020-04-17:
Probably the docs aren’t up to date.

But if you are on master you need to update Mopidy to version 3 first.
Please see #852 for necessary steps.

In addition you need to install Gmusic extension with `sudo python3 -m pip install Mopidy-Gmusic`.

comdoxx, 2020-04-17:
Thanks Martin,

I now installed pyasn1 0.4.6 even without easy_install. That tool seems to be a legacy tool. When running the `mopidy local scan`, it now prompts: ` Enabled extensions: iris, softwaremixer, file, m3u, http, local, stream, mpd, gmusic` - so I expect gmusic to be active. Spotify is disabled as I didn't fill out any credentials or stuff - not my choice of streaming service...

But where should I find it? When I start Iris, I don't find GMusic in the Services section. If I open [http://pi-keno:6680/iris/library/browse](the library), I see GMusic but as the other two folders (Files and Local Media), it's empty. The latter are empty despite the scan seems to have found files in `~/Rpi-Jukebox-RFID/shared/audiofolders`?

Oh: I now started mopidy on command line and needed to login manually and put the `refresh_token` to the mopidy.conf by hand. That isn't documented?

MiczFlor, 2020-04-18:
Hi @comdoxx 
could you write here a few lines about the steps that made it work for you? Ideally actually into the wiki so we can have one place where the documentation sits. But I can also take your notes and write a wiki page.
Thing is that with Gmusic, mopidy and Spotify we are depending on external software and APIs. And it's been creating troubles again and again. But is outside of our control.

s-martin, 2020-04-20:
I don't have a Gmusic account, but I think at least the following steps are necessary:

1. Install Phoniebox Spotify edition
2. Install GMusic `sudo python3 -m pip install -r requirements-gmusic.txt`
3. Configure GMusic login `mopidy gmusic login`

Currently this works only on develop branch, because Mopidy 3 is necessary.

s-martin, 2020-05-01:
@comdoxx: I have updated the wiki, see https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Enable-Google-Play-Music-GMusic

Could you please check, if that's correct?

s-martin, 2020-05-24:
2.0 release should fix this on master.

closed 2020-05-24 by s-martin