MiczFlor/RPi-Jukebox-RFID #910: 🐛 | Card left on reader is interpreted as second swipe

jo-me, 2020-04-19:
## Bug

### What I did

Put a card on RFID RC522 reader and leave it there

### What happened

Playlist starts playing. Then the card is apparently read again and the "Second swipe" setting is used to determine what happens... e.g. restart playlist.
This is triggered every 2-3seconds in my setup so I cannot just leave the card on the reader.

### I expected this to happen

Between two swipes there needs to be a time where no card is detected otherwise it is not a _second_ swipe, right?

If the card is left on the reader as it could easily be in many of the Phoniebox cases, the only option to prevent this issue is to configure the **second swipe to be only active for system commands**.
But by doing this, **you lose the entire functionality of the second swipe in the context of the current playlist** which is very uncool. You are basically forced to have separate cards or buttons to pause the playback then.

While system commands should simply be retriggered, second swiping while the playlist is running should still e.g. pause or advance the playlist.

### Further information that might help

## Software

### Base image and version

2.0 rc8 

### Branch / Release

master

### Installscript

Buster

## Hardware

### RaspberryPi version

Rpi 2

### RFID Reader

RC522

jo-me, 2020-04-21:
Possible solution approach here: 

https://github.com/ondryaso/pi-rc522/issues/44#issue-342940757

Give the wait_for_tag function a timeout. If that timeout is reached without detecting a card we could call the rfid-trigger_play script from the daemon with an empty card id or remove the files used by that script directly (e.g. Latest_Playlist_Played,  Latest_Folder_Played).

ozone89, 2020-04-24:
problem is, so far only the one or two readers provide an out-of-the-box tag presence API, none of them being supported due to their admittedly high price ($60), and because using them requires dedicated code

s-martin, 2020-04-25:
> Possible solution approach here:
> 
> [ondryaso/pi-rc522#44 (comment)](https://github.com/ondryaso/pi-rc522/issues/44#issue-342940757)
>
Unfortunately there seems to be not much development on pi-rc522. 
 
> Give the wait_for_tag function a timeout. If that timeout is reached without detecting a card we could call the rfid-trigger_play script from the daemon with an empty card id or remove the files used by that script directly (e.g. Latest_Playlist_Played, Latest_Folder_Played).

Could you start a PR?

jo-me, 2020-06-16:
With the new way of including the library in 2.0 the fix can't be patched in the local version anymore since it is somewhere in the pip module directory.

A PR on the pirc522 project seems pointless since it appears to be dead. 
So, I assume we could fork it, fix it and use that instead..

s-martin, 2020-06-16:
I'm a little reluctant to fork pi-rc522, because we then need to maintain it.

But I agree the current situation is unsatisfactory.

jo-me, 2020-06-16:
Yes, I agree.

For testing purposes, I've just created a patched version of pi-rc522 that includes that timeout function and modified the rfid daemon so that it pauses and resumes when you remove and reswipe the same card. It works quite nice, actually. 
I tried timeouts as low as 0.5s and it seemed to work. This could be a solution for RC522 owners to mimic Toniebox behavior.

I'll check next whether my solution approach for the actual issue from above is viable as well .

s-martin, 2020-06-16:
You probably forked the pi-rc522 repo to patch this and provided a PR to the upstream repo?

One idea could be that we use in Phoniebox your forked repo including the patch as long as pi-rc522 repo is not active. If pi-rc522 ever becomes active again and then includes the modifications, we could switch back to pi-rc522.

jo-me, 2020-06-16:
Yeah that could work. I have not prepared a PR yet as I wanted to test it first. 

For the original "card left on reader is interpreted as second swipe" problem, I think there is already an almost complete solution implemented that doesn't even need the patched library..

In the rfid daemon there is a "same_id_delay" that is set to 0 for some reason. I set it to 2 and additionally set `previous_time = time.time()` more generously at the end of the  `if cardid is not None:` block to always update it if a cardId is detected.
This prevents that anything is happening further down the line as long as the card remains on the reader.

At least for the rc522, the card id is almost constantly detected by the reader over and over again. Not sure about the other supported rfid readers, though.

What do you think about that?

s-martin, 2020-06-16:
I don’t use the rc522 reader, so I don’t have any experience. 

If the modifications work in your setup please provide a PR and then it can be integrated in the source code.

jo-me, 2020-06-17:
I've created the PR in the upstream project to add the timeout functionality for RC522 libary.
https://github.com/ondryaso/pi-rc522/pull/70 

Might not be needed for this issue after all but is handy to have if toniebox behavior is desired.

closed 2020-07-16 by MiczFlor